luc-github/ESP3D-WEBUI #112: [Question] How to use embebed camera in ESP32-CAM

tiwanacote, 2020-09-03:
Hi, I have compiled and uploaded ESP3D 3.0 on ESP32-CAM board but I do not find the camera on UI. As I understand this version support ESP32 hardware, but what about its camera?
Isn't it implemented yet?

Thanks!

luc-github, 2020-09-03:
not yet https://github.com/luc-github/ESP3D-WEBUI/issues/94#issuecomment-683278945

luc-github, 2020-09-05:
I guess question is answered

closed 2020-09-05 by luc-github

tiwanacote, 2020-09-10:
On version 3.0, when compiling without a camera (CAMERA_MODEL_AI_THINKER), we could access the UI.
However, after enabling it, no TCP port could be open (80, 23, 8282, 9600). 
Briefly, in both cases AP connection succeeded, but on the second scenario (camera enabled) the project did not show the web.
Camera disabled

;Starting Network>>udp_sendto result 0                                          
M117 Starting WiFi                                                              
M117 192.168.0.1                                                                
M117 AP SSID: 'ESP3D' is started and protected by password                      
M117 Captive Portal started                                                     
M117 HTTP server started port 80                                                
M117 Telnet server started port 23                                              
M117 Ftp server started ports: 21,20,55600                                      
M117 Websocket server started port 8282                                         
M117 192.168.0.1                                                                
M117 New client    

Camera enabled

luc-github, 2020-09-10:
be sure you did a `[ESP444]RESET` when moving from 2.1 to 3.0 may be 
If you have issue open it on https://github.com/luc-github/ESP3D and fill the template with all informations