luc-github/ESP3D-WEBUI #113: how i can compile this new webui for version  3?

labotecno, 2020-09-04:
do you have a tutorial to install the new version V3 ? thank you if not can you help me to do it

luc-github, 2020-09-04:
everything is in readme and the compiled version is in dist directory

labotecno, 2020-09-04:
can i use windows 10 ?

labotecno, 2020-09-04:
C:\esp3\ESP3D-WEBUI-3.0>npm install
npm WARN saveError ENOENT: no such file or directory, open 'C:\esp3\ESP3D-WEBUI-3.0\package.json'
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN enoent ENOENT: no such file or directory, open 'C:\esp3\ESP3D-WEBUI-3.0\package.json'
npm WARN ESP3D-WEBUI-3.0 No description
npm WARN ESP3D-WEBUI-3.0 No repository field.
npm WARN ESP3D-WEBUI-3.0 No README data
npm WARN ESP3D-WEBUI-3.0 No license field.

up to date in 1.043s
found 0 vulnerabilities
 i get this

luc-github, 2020-09-04:
Yes I use windows 10
your version of npm is too old (should be 6.4.1) what is your version of node?

labotecno, 2020-09-04:
6.14.6

luc-github, 2020-09-05:
this is your version of npm, type `node -v`

labotecno, 2020-09-05:
v12.18.3

luc-github, 2020-09-05:
as mentioned in read me I use v10.15.3 - I will chek why latest LTS v12.18.3 show such messages

labotecno, 2020-09-05:
> 
> 
> as mentioned in read me I use v10.15.3 - I will chek why latest LTS v12.18.3 show such messages

thank you i hope its working i want to try version 3

luc-github, 2020-09-05:
I have did a fresh install on one windows 10 and an update on my Programming PC also win10, installation is pretty long but ok
```
Microsoft Windows [Version 10.0.18362.1016]
(c) 2019 Microsoft Corporation. All rights reserved.

C:\Users\luc>npm -v
6.14.6

C:\Users\luc>node -v
v12.18.3

C:\Users\luc>cd Documents

C:\Users\luc\Documents>cd GitHub

C:\Users\luc\Documents\GitHub>cd ESP3D-WEBUI

C:\Users\luc\Documents\GitHub\ESP3D-WEBUI>npm install

> node-sass@4.14.1 install C:\Users\luc\Documents\GitHub\ESP3D-WEBUI\node_modules\node-sass
> node scripts/install.js

Downloading binary from https://github.com/sass/node-sass/releases/download/v4.14.1/win32-x64-72_binding.node
Download complete  ] - :
Binary saved to C:\Users\luc\Documents\GitHub\ESP3D-WEBUI\node_modules\node-sass\vendor\win32-x64-72\binding.node
Caching binary to C:\Users\luc\AppData\Roaming\npm-cache\node-sass\4.14.1\win32-x64-72_binding.node

> node-sass@4.14.1 postinstall C:\Users\luc\Documents\GitHub\ESP3D-WEBUI\node_modules\node-sass
> node scripts/build.js

Binary found at C:\Users\luc\Documents\GitHub\ESP3D-WEBUI\node_modules\node-sass\vendor\win32-x64-72\binding.node
Testing binary
Binary is fine

> nodemon@2.0.4 postinstall C:\Users\luc\Documents\GitHub\ESP3D-WEBUI\node_modules\nodemon
> node bin/postinstall || exit 0

npm WARN ESP3D-WebUI@3.0.0 scripts['server'] should probably be scripts['start'].
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.13 (node_modules\webpack-dev-server\node_modules\fsevents):npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.13: wanted {"os":"darwin","arch":"any"} (current: {"os":"win32","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.13 (node_modules\watchpack-chokidar2\node_modules\fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.13: wanted {"os":"darwin","arch":"any"} (current: {"os":"win32","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@2.1.3 (node_modules\fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.1.3: wanted {"os":"darwin","arch":"any"} (current: {"os":"win32","arch":"x64"})

added 1294 packages from 609 contributors and audited 1297 packages in 34.14s

46 packages are looking for funding
  run `npm fund` for details

found 0 vulnerabilities
```
Then build  :
```
C:\Users\luc\Documents\GitHub\ESP3D-WEBUI>npm run build

> ESP3D-WebUI@3.0.0 build C:\Users\luc\Documents\GitHub\ESP3D-WEBUI
> npm run build-printer & npm run build-grbl & npm run build-lang

> ESP3D-WebUI@3.0.0 build-printer C:\Users\luc\Documents\GitHub\ESP3D-WEBUI
> cross-env NODE_ENV=production TARGET_ENV=printer webpack --mode=production

remove-files-plugin@1.4.3:
Skipped, because not exists – "C:\Users\luc\Documents\GitHub\ESP3D-WEBUI\printer"
Could not find any items to permanently remove

remove-files-plugin@1.4.3:
The following items have been permanently removed:
  folders:
    dist\printer\languages
  files:
[... 145 lines cut ...]
gzipper: GZIP ->
gzipper: 1 file has been compressed.
Compressed files:  [
  'C:\\Users\\luc\\Documents\\GitHub\\ESP3D-WEBUI\\dist\\printer\\languages\\fr.json'
]
gzipper: 1 file has been compressed.
Compressed files:  [
  'C:\\Users\\luc\\Documents\\GitHub\\ESP3D-WEBUI\\dist\\grbl\\languages\\fr.json'
]
gzipper: 1 file has been compressed.
Compressed files:  [
  'C:\\Users\\luc\\Documents\\GitHub\\ESP3D-WEBUI\\dist\\printer\\languages\\zh_cn.json'
]
gzipper: 1 file has been compressed.
Compressed files:  [
  'C:\\Users\\luc\\Documents\\GitHub\\ESP3D-WEBUI\\dist\\grbl\\languages\\zh_cn.json'
]

C:\Users\luc\Documents\GitHub\ESP3D-WEBUI>
```
So I do not have any issue -  I guess your installation of node has issue

You better redo full installation and may be also refresh the esp3d-webui sources as I have updated some package versions to remove security warnings recently

Also be noted that as mentioned before : **you do not need to build anything if you do not change anything** - the build **index.html.gz  is already done in `dist/` directory** (https://github.com/luc-github/ESP3D-WEBUI/blob/3.0/dist/printer/index.html.gz) only printer is currently usable - grbl version will come after

labotecno, 2020-09-05:
seem its working ! with http://localhost:3000/
now how i can to put the last version on the esp32 with cam ? arduino or vscode? 
thank you 
[image]

luc-github, 2020-09-05:
1 - flash your board with ESP3.0 FW https://github.com/luc-github/ESP3D/tree/3.0
2 - upload webUI 3.0 (https://github.com/luc-github/ESP3D-WEBUI/blob/3.0/dist/printer/index.html.gz) when requested

Note: as explained here : https://github.com/luc-github/ESP3D-WEBUI/issues/112 camera UI is not ready - only test page is available

labotecno, 2020-09-05:
i need to config the configaration.h ?

luc-github, 2020-09-06:
yes you need to configure FW according your needs- the configuration.h has comments to explain each settings

royfocker19, 2020-09-07:
Hi Luc, do you think it could be possible to have some base configuration files (like in marlin) for most popular devices (esp32 cam diy more, si thinker, m5stack, etc)?
I cannot make my camera work using the sample file on a esp32 cam from DIY More, and the problem maybe is related to pins, but also some sensors are enabled by default on esp32 firmware, causing reboot if not disabled.

luc-github, 2020-09-07:
@royfocker19  the current 3.0 is in alpha stage - so configuration .h follow my status of features implementation 
final one will be basic most feature will be disabled by default

Sample of several configurations.h is not planed in short term because too many user case - but your problem may be the platformIO.ini if you use or the flash settings this will be addressed https://github.com/luc-github/ESP3D/issues/478
but this is in esp3d FW not esp3d-webUI

labotecno, 2020-09-07:
> 
> 
> yes you need to configure FW according your needs- the configuration.h has comments to explain each settings

thank you for all step to make it working this implementation ! merci

luc-github, 2020-09-07:
I plan to do some videos and docs, but this will be done end of beta when everything will be stable and defined clearly

luc-github, 2020-09-07:
I guess issue can be closed now

closed 2020-09-07 by luc-github