luc-github/ESP3D-WEBUI #114: [BUG] Cannot connect to SSIDs with spaces

royfocker19, 2020-09-06:
Using latest firmware and webui versions, I cannot connect to my WiFi “the cat house”
I can select the said from the list, but once saved and rebooted, it only shows “the” and ESP AP is still available.
I’m running it on an esp32cam from diy more.

luc-github, 2020-09-06:
what version of fw?

royfocker19, 2020-09-06:
FW: 3.0.0.a52
Interface: 3.0.0.61

luc-github, 2020-09-07:
Ok I reproduced issue - looks like the info is well sent from webUI but FW do not save it properly 
I will look at it

luc-github, 2020-09-07:
Ok the bug is definitly due to new version of `const char * Commands::get_param (const char * cmd_params, const char * label)`
it manage space differently and I forget it in web-ui orz... 

I need to add a space formating before sending the command

luc-github, 2020-09-07:
fixed in https://github.com/luc-github/ESP3D-WEBUI/commit/c8c1830e405eaa4fbb2a032ba1f39631645a0b95 v3.0.0.62

closed 2020-09-07 by luc-github