luc-github/ESP3D-WEBUI #116: [Question]ESP3d WebUi stuck loading

Unkowntheifer, 2020-09-16:
I uploaded the index file and got into the configurator but it has just stayed on the loading screen for about an hour
 
[image]
 this is what it looks like

luc-github, 2020-09-16:
I guess the file index.html.gz is corrupted - be also sure the size of partition did not changed between 2 flash

Unkowntheifer, 2020-09-16:
is there a way to fix it or is it dead? or do i need one of the writers?

luc-github, 2020-09-16:
go to maintenance page `http://<your IP>?forcefallback=yes`  and delete corrupted file
then upload again and be sure there is enough space on flash for the file

Unkowntheifer, 2020-09-16:
I never got to the configure page will I still be able to delete the corrupted file from the website you linked?

luc-github, 2020-09-16:
yes that is purpose of maintenance page - it is in FW not in index.html.gz

luc-github, 2020-09-18:
@Unkowntheifer did you solved your issue ?

closed 2020-09-19 by luc-github