luc-github/ESP3D-WEBUI #117: [Question]<Making edits and recompiling idex.html.gz>

Mboss32, 2020-09-18:
HI, 

I have been using the ESP3D WebUI for a while on my GRBL machine.  I am interested in making a few custom edits to help my workflow.  The main questions I have: 

1) what is the best software package to use for making edits? (Visual studio, Netbeans, etc) 
2) What is the process to build the Index.html.gz file?  

Sorry if this is a silly question, I am very familiar with C and C++ but this will be my jumping point into Javascript and html.  

Thanks for your time!

luc-github, 2020-09-18:
1 - it is up to you  there is nothing mandatory, I personnaly use GEdit
2 - it is in Wiki for 2.1https://github.com/luc-github/ESP3D-WEBUI/wiki/Compilation---Development

be noted 2.1 is only for bug fix now, new feature will go to 3.0 which use Preact instead of pure javascript and customized libraries/code
2.1 grbl UI share 90% ui,  with 3.0 UI is splitted giving more flexibility, but I am still on 3D printer one.
I will work on GRB  specific UI soon, I am building one laser system and one spindle system.

Mboss32, 2020-09-18:
wow, thanks for the quick response!  Sorry I missed the wiki compilation.  

The 3.0 UI looks amazing and many of the features are what I want to put into the 2.1 - anyway to encourage 3.0 GRBL and GRBL_ESP32 3.0 progress? I would be happy to donate to the cause!

luc-github, 2020-09-18:
Thank you ^_^ support is always welcome ([here](https://www.paypal.com/donate/?cmd=_s-xclick&hosted_button_id=Y8FFE7NA4LJWQ) or/and [here](https://github.com/sponsors/luc-github)) rewrtite is huge task and time consuming.

Right now, GRBL ui in 2.1 is pretty basic due to :
1 - I had almost 0 knowledge in CNC / GRBL when I did it, it was to support [GRBL_ESP32 FW] (https://github.com/bdring/Grbl_Esp32) but I have no system just a board
2 - UI in 2.1 is 3D printer oriented so introducing grbl was more an ugly patch and not easy to maintain due to monolic architecture

Now, I am still a noob in GRB...L, but I know more, I got some feedback that are I try to implement in 3.0, and want UI to be better, richer, more interactive, etc...

While ESP3D 3.0 FW is pretty frozen : https://github.com/luc-github/ESP3D/blob/3.0/ESP3D-features.xls
  
The Web UI 3.0 definition is still under [definition](https://github.com/luc-github/ESP3D-WEBUI/issues?q=is%3Aissue+is%3Aopen+label%3A3.0)  and implementation - and GRBL is not defined at all actually - only common parts with 3D printer are done,
@bdring did not gave me any feedback/request yet because he is also very busy doing huge improvement in GRBL_ESP32 FW] (https://github.com/bdring/Grbl_Esp32)  so feel free to submit any detailed feature request you need/suggestions/ ideas , it will definitly help a lot to define the new UI and so speed up the coding.

luc-github, 2020-09-21:
I guess issue can be closed

closed 2020-09-21 by luc-github

Mboss32, 2020-09-21:
Thank you for the start and thoughtful response on 3.0 development.  I will definitely submit feedback and requests/ideas.  Where is the best place to do that?

luc-github, 2020-09-21:
In github or in discord