luc-github/ESP3D-WEBUI #119: [Question] Native Smoothieboard support

arthurwolf, 2020-09-21:
Hello!

I was just looking into this: https://github.com/luc-github/smoothieware-webui-for-ESP3D which is supposed to allow running designed-for-Ethernet Smoothie web interfaces, through an ESP3D/Wifi setup.

Was this actually functional? Did it actually do it's job? If so, I really need to 1 get it working myself, and 2 write documentation so Smoothie users can start using it.

Can anyone fill me in on the status here? 

Sorry I post about it here, the repo itself is archived it seems, can't open issues there.

Thanks a lot in advance for any information you might have!

Cheers!

luc-github, 2020-09-21:
Hi Arthur,
ESP3D Webui is the successor of smoothieware-webui-for-ESP3D and has same feature : support smoothieware connected by esp board using esp3d Firmware- the smoothieware ethenet was never on scope, because it do not go thrue esp board.and rely on esp3d webserver not smoothieware.

I hope it answer your question - if not let me know

arthurwolf, 2020-09-21:
So just to be clear, you are saying right now I can get an ESP board,
connect it to my Smoothieboard, and start using the Smoothieboard's web
interface over Wifi?

Could you give me very basic instructions on how to do that ( so I can test
it, and from that test write documentation for Smoothie users ), or point
me at a link/documentation that would explain to me how to do that?

Thank you so much for doing this in the first place, and for helping me now!

arthurwolf, 2020-09-21:
Wow I didn't realize the project was this advanced, I am looking at the
documentation for ESP3D-WEBUI and it is blowing my mind !!! How did I miss
this !!!

luc-github, 2020-09-21:
Arthur to install ESP3D 2.1 follow read me https://github.com/luc-github/ESP3D#installation

For 3.0 webUI is still under devt but it is more rich than 2.1

luc-github, 2020-09-21:
@arthurwolf happy you like

luc-github, 2020-09-23:
@arthurwolf did you succeed to test?

arthurwolf, 2020-09-23:
I'm very busy with Covid+Smoothiev2, but I'll test as soon as possible yes!

luc-github, 2020-09-28:
@arthurwolf I close issue then - feel free to open new issue if needed if this one is locked by inactivity

closed 2020-09-28 by luc-github