luc-github/ESP3D-WEBUI #120: Printing from Web Gui is only possible with the command line [BUG]

BTH-KIN, 2020-10-05:
Dear Madam or Sir

I have your very nice software on the ESP-015 to communicate with my ender 3 with SKR V2.0 and Marlin 2.0.
Your software is from the latest commit (04.10.2020 15:00 European Time.)
My client is a Windows 10 with firefox, the 81.0 (64-Bit) for connection with the printer.

I can communicate in both directions so I can see the temperature, position, SDcard file. I can move the printer and set the temperature and load files to the SD card.

But the problem is, I cannot start the print with the print button, but only with the M32 P !/models/lgbust.gco#  derective. Do you have a solution for this?
 
Thanks for your great work, is a really nice software :)

luc-github, 2020-10-05:
[image]
 I cannot duplicate

as the bot wrote please fill issue template because I need these informations to be able to help

luc-github, 2020-10-12:
I guess no more issue

closed 2020-10-12 by luc-github