luc-github/ESP3D-WEBUI #124: [FEATURE REQUEST]Camera integration discussion

luc-github, 2020-10-14:
This is to discuss about camera integration in 3.0
- there are 2 types of possible output
 * as image, this is for mpeg camera like esp32cam
this one can be resized but need command line authentication support 
it can fit a panel or a full page
 * as object / html page 
this one usually cannot be resized , and so can not easily  fit in a panel but should be ok for a full page using  iframe

additionally 2 sources are possible:
- external IP cam
- embedded IP cam like esp32-cam 

The second one may lead to performace issue doing queries and streaming videos on same device, but this is another issue that I won't address here but as limitation usage when UI will be defined, the only difference is that the UI should allow to control the camera settings as well as displaying video stream.

this is a typical activx output example that will need an iframe and could not be really handled :
[image]

here another example of this time an image that can fit a panel or even be streched as full screen:
[image]

Here an example as image set as background:
in mobile view but sized based on screen width to keep proportion ( not fully fitted)
[image]

in pc view :
[image]

the overlay is basic and need to be completly reviewed to be properly readable

I did not tested the panel solution yet to see what constraint would appear 

I am thinking about a button on top menu to put the camera as main or background or off (no camera output)

So many parameters and so many user cases... orz - which means need to simplify configuration 

Any idea/suggestion is welcome

DivingDuck, 2020-10-14:
Do you know Calibre? Kovid had a nice feature implemented where you can hide / enable parts of the GUI via clickable icons in the bottom of the window frame. I find this kind of control very useful. 
[image]

luc-github, 2020-10-29:
panel part for camera is implemented :
https://github.com/luc-github/ESP3D-WEBUI/issues/123#issuecomment-718693756
if camera content is not an image - the panel add scrollbars

luc-github, 2020-10-30:
embedded camera with flow control is now supported in webUI in panel as well as page, same for external 

the full screen is not really urgent and will be easier to implement after the code refactoring / simplification so it is on old now and will work back on it in beta stage after first refactoring

LuisGeissler, 2022-01-10:
hey,
ESP3D is really great, but I wanted to ask if with the new version it is possible to use an ESP32 cam for control and camera transmission at the same time.

luc-github, 2022-01-10:
yes ESP3D 3.0 + webui 3.0 support ESP32Cam https://github.com/luc-github/ESP3D-WEBUI/issues/123

luc-github, 2022-03-09:
Now I found a method to control camera flow and use any webserver without overload I think current integration is ok 
time to close topic

closed 2022-03-09 by luc-github