luc-github/ESP3D-WEBUI #143: All is working but some features not

Jean-Michel-90, 2020-11-15:
Hello,

First of all thanks for your great job.
Unfortunatly I'm newsby in code and it's difficult for me to have a good understanding on "how to" get a well  done firmware for the module and an adapted WEB UI....

Here is what I've done and the result I've got:
I bought ESP 01S on Aliexpress (https://fr.aliexpress.com/item/33044399651.html?spm=a2g0s.9042311.0.0.2ab26c379xMmjW) and I plugged it on my TFT35 V2 (my printer motherboard is a SKR mini E3 V1.2). 
As soon as I powered the printer, a new wifi (ESP3D) appeared .
As I was affraid to destroy my WIFI module by a wrong code or by a bad programming, I've tried to connect to this net without change.
After uploading the index.html.gz I've got on your Github (https://github.com/luc-github/ESP3D-WEBUI), Web user interface is working and I'm able to send Gcode, move motors access to booth SD cards (TFT one as well as SKR mini E3 V1.2 one) but I have a lot of limitations...
No SD card content (but sometime I can see the directory...)
[image]
Impossible to upload files to SD cards (both) due to 8.3 file format restriction
[image]
Neither BED temperature control nor temperature graph:
[image]
It's impossible to acceed to ESP3D files:
[image]

I'm quite sure it's a problem between original software and Web UI but I need help to manage this issue (in French or English)

Thanks by advance.

luc-github, 2020-11-15:
there is no information on your configuration - what setting do you use to compile - what version of webUI  / esp3d fw do you use, what version of Marliin do you use etc...
 
 if you can not list SD content you won't be able to upload - SD just send M20

the 8.3 sd file limitation is due to fact you do not use DOS file name  8 char for name 3 for extension - it is not  a bug it is Marlin feature

luc-github, 2020-11-15:
Also no worry for your wifi module - it is very hard to brick it - and using ESP3D even with wrong settings - there no risks (in theory)

luc-github, 2020-11-15:
also you are using through TFT - check your TFT brand name and FW version some  TFT need FW update to work with ESP3D

The SD from SKR mini E3 should be accessible from TFT and ESP but again this will depend of TFT capabilities : FW and brand

luc-github, 2020-11-15:
also as it is question not a bug  feel free to join discord channel, it may be more convenient for you

Jean-Michel-90, 2020-11-15:
Hello Luc,
Thanks for your quick answers.
I'm going to reply step to step:
what setting do you use to compile? noone because I don't compile but use the ESP 01S as is
what version of webUI / esp3d fw do you use? 
     *Web UI 2.1 (only index.html.gz have been uploaded from initial web interface)
     * ESP3D: I don't know. it's the one coming with ESP 01S bought on Aliexpress and I cannot find how get it from web UI (I told you I'm a newby...)
Marlin version: 2.0.5.1
If you can not list SD content you won't be able to upload - SD just send M20 . Yes, I know but sometimes I can read SD and I already succeed to print via WEB-UI
But even when I'm able to read the content, sending a file show me only waiting icon...

The 8.3 sd file limitation is due to fact you do not use DOS file name (8 char for name, 3 for extension) - it is not a bug it is Marlin feature. OK, but Marlin is OK to print file with long names.

you are using through TFT - check your TFT brand name and FW version some TFT need FW update to work with ESP3D
TFT is the TFT35 V2 from Bigtreetech . I have updated TFT firmware to Vx.x.26 without any changes.

As it is question not a bug feel free to join discord channel, it may be more convenient for you: I'll try

Jean-Michel-90, 2020-11-15:
Je me suis inscrit sur Discord et je vois bien nos échanges dans #github-esp3d-webui mais je n'ai pas le droit de poster des messages.
En attendant voici ma config complète:
[image]

luc-github, 2020-11-15:
version informations are visible on top of webUI menu 
[image]

luc-github, 2020-11-15:
ok just see your message
update your ESP3D Fw it is very old  latest 2.1 version in github is : 2.1.1.b8

luc-github, 2020-11-15:
oui pour discord il y a une securite pour les spam il faut attendre 10min apres inscriptions pour pouvoir poster

Jean-Michel-90, 2020-11-15:
OK. Do I need to compile it?

luc-github, 2020-11-15:
yep - I do not provide binary as too many esp configurations possible - sorry

luc-github, 2020-11-15:
also does TFT always see printer SD content when esp does not ?

Jean-Michel-90, 2020-11-15:
TFT always see printer SD
I'll try to compile but I'm afraid to need your help....

About Discord, it's not a time limitation, see picture hereunder:
[image]

luc-github, 2020-11-15:
on ne peut pas poster dans le canal du bot github - faut utiliser un autre canal

Jean-Michel-90, 2020-11-15:
OK, je poste direct dans support request au milieu des autres posts?

luc-github, 2020-11-15:
oui le canal de discussion general comme tu as fait est nickel

Télécharger Outlook pour Android<https://aka.ms/ghei36>

luc-github, 2020-11-17:
@Jean-Michel-90 do you have your issue fixed with latest FW ?

Jean-Michel-90, 2020-11-17:
Hi Luc,
Unfortunately not (yet). Following your king advices, i've suceeded to compile but I'm not able to upload the firmwarebecause ESP-01S hasn't enough memory to be updated with Web UI and I haven't a FTDI USB-TTL in hand. I've ordered one and I expect to received it next week.
Anyway, I'll keep you posted.
Thanks again to spend hours to help me.

luc-github, 2020-11-17:
Ok - with the ftdi it should be easy to flash it - good luck 😸

luc-github, 2020-11-30:
I close issue according discord discussion

closed 2020-11-30 by luc-github