luc-github/ESP3D-WEBUI #144: [FEATURE REQUEST]GRBL UI definition

luc-github, 2020-11-17:
I am not experienced in CNC so I need feedback of user who have habits on it

WebUI 3.0 for printer and common is almost done - still have to wiork on winzard - so now I really need input on grbl UI to implement it 

in 2.1 there waa=s a very limited UI -due to existence of 3D printer UI 
In webUI 3.0 everything is splited so UI can be absolutly different without problem

e.g: jog UI for GRBL is based on buttons instead of pronterface inspiration UI for 3D printer

there is only one request for the moment (https://github.com/luc-github/ESP3D-WEBUI/labels/GRBL) so it is time to speak up 😺 

in notifcation area it will have :
* positions of the machine and the job ones -  that will open the jog panel
* status based on `?` request
* job status for GRBL_ESP32 
* error message if any based on alarm code / error code
* reset button wil be the emergency command ? or any other is more suitable

the panels :
* jog panel
* sd panel for grbl_esp32
* overide controls panel but with status according `$G`  answer, so we  should  have something in notification area for this 
* probe panel  - the probe status also should be in notification area (green / red / unknow)

Any other feature would be interesting to be present ?

settings  / macro / user panels will be of course supported

This web UI should cover grbl FW used  with esp3d as daughter board and grbl_esp32 FW with esp32 as main board

Now guys - I am waiting for your requests / suggestions / ideas, discussion is open

eglrd, 2021-01-21:
Hi ! I might be able to help on the UI "look" a bit. Would this thread be the appropriate place to suggest a much desired UI ?
Ideal would be to have in a single screen with tabs:

Screen:
> Position : Machine Position, Work Position (+Work Offset selection, i.e. G54, G55, etc), Zeroing of current position
> Jogging: Buttons with repeat on Mousedown (or on pressed for touch interfaces) and homings
> Spindle and Coolant on/off
> Console feedback

+ "tabs" that can be selected:
> tab Probe: X, Y Z & parameters (type of probe, feedrate, touchplate thickness, retraction, 
> tab Gcode: list, upload, select job (> selecting opens Job tab)
> tab Job: graphical preview (like in cncjs... we can always dream ?) or text highlight of current move / current line in Gcode, a panel with distance to next move (important while running cycle for the first time), Cycle Start, Freehold, and F, S, R overrides
> tab macros: for people extending further their machine.
> tab status: Modal groups & Status report
> tab system: to control GRBL ESP32 system values

I might be able to come up with UI elements to have all this fitting in a 16/9 aspect ratio (horizontal most likely): shall I submit something or is it too late ?

luc-github, 2021-01-21:
Hi thank you,
Here is the right place - just some comments
UI need to be also tablet / phone compatible, I am afraind limit UI to 16/9 is little bit reductive but should be ok if fit phone interface, several comments from 2.1 UI of GRBL was phone support and no scrolling so that why 3.0 use panel with moving position.
That said what are the benefis of tabs vs panels ?

Tab grbl is already in settings page to put all settings in one place instead of spread them everywhere.
About preview, webUI is not streaming origin - I really do not see an easy way to load a preview - cncjs is on pc and already has access to all gcode file without affecting grbl -  webUI is a remote tool which would need to load the complete file to generate the preview - this will bring performance issue on grbl itself - but better hope than stay doing nothing
The jog Panel is currently  like this :
[image]
it support up to 6 axis
the positions (M and W)  should be displayed in notification area like every display information status / message /  etc ...
console is one panel 
file job is one panel

>  text highlight of current move / current line in Gcode, a panel with distance to next move (important while running cycle for the first time), 

How to get such info ? remember webUI is remote  tool not the streaming host or is there new feature in grbl_esp32 that I am not aware of ?
Macros are in dashboard because we need to have quick access to them 
Console panel is already done 
You can see how printer display handle UI using the offline version `npm run dev`
I just share my original plan - a panel for each necessary feature probe / overide / Spindle / Coolant / anything needed to leverage api already developed for 3D printer, as features are different from 3D printer
No limit for panels numbers all what it need.

That said I am open to any suggestion as UI is splited for GRBL and not really started for GRBL specific features
so feel free to share mockups here - everything is open to discussion and very welcome

eglrd, 2021-01-22:
Cool, 'will give it a go and submit mockups later here.

Key argument for tab is not to ever have to scroll, and keep context of main screen at all time and be able to abort/control as much as possible (much more critical than for 3D print as you can break your machine badly)

I realised there is not the info needed to generate a "next move / distance to go" feedback... that is a bit unfortunate because when you run a cycle for the first time, the "practice" is:
- slow rapids
- cycle start
- check machine moves as it goes
- freehold when approaching the piece
- check current position, and check "distance to go"... this can be done also by looking at gcode and seeing want the next move are luckily... and if everything seems ok
- cycle start again to "free" the machine until the next critical move
- Feedrate or speed can be adjusted on the fly if "noise" of the machine is not good (once again free hold might save the day multiple times here)
- repeat until the end
...
only once everything went well do you "free" the machine for the next cycle of cam execution... (rapids 100%, feed and speed validated).

I'm trying to come up with screenshot of mockup for what I would suggest as screen/tabs.

(I also realized I might not have 2.1 given my jog menu in not looking like the one you sent - I'm coming from GRBL_ESP32 so he might have branched earlier)

luc-github, 2021-01-22:
2.1 is the current one it looks like pronterface and CNC user seems not liking it but prefer the button like UI
So it is what I did in 3.0 - it is the screenshot I shared above based on CNC users feedback

eglrd, 2021-01-23:
a quick one just a mockup wireframe with minimal styling... time flies... here is the 1st part I had in mind : main part of the screen that needs to remain visible at all time to allow actions on the machine. 
i'll try and get back later today to try and explain the philosophy.
[image]

luc-github, 2021-01-23:
Thank you - also FYI this the current approach for the UI https://github.com/luc-github/ESP3D-WEBUI/discussions/94

eglrd, 2021-01-23:
Okay, I see: quite a departure from current approach... so I guess I'm a bit too disruptive then. 
here is a mockup of a full layout with the elements that are important in my mind, maybe this can help anyway ?
[image]

1) BIG action button, placed in specific region of the screen. More often than not, I'm looking at the part/stock to machine when jogging so I need to be able to "locate" my hand on the control screen : here, as a suggestion for a right-handed person (this needs to be adapted for a left handed...) you would put your fingers behind the screen on the jog buttons side and push the correct jog actions with your thumb... that way you can "feel" X is close to the top, Y under, Z under, etc.
Same goes for cycle start & hold, that can be located without looking at then again.
1.1) critical action button are not mixed to avoid mis-use
1.2) grouped action buttons (jog, home, spindle, coolant, job control)
1.3) reset & Unlock are positioned so that reset is easy to hit... and unlock needs to be looked for (to make sure you are not hitting it by accident...)
1.4) job a DRO side quite appart : to allow to touch one side controling the other (not hiding it with your hand)

2) BIG "DRO" : to allow some "hand machining" potentially

3) Buttons with "active" feedback (the white square in limits, spindle or coolant mode) to show visually what is active (the expectation would be to have them filled with a color/light when active)

4) Square button must be seen as "dialog box opener" showing the list of possible options to set: ex G54 can be touched, showing other WCS (G55,... G57) and changed to appropriate value. same goes for G21, G09, G1, G0, jogging step, etc.

... but I understand the departure from your current work... that being said maybe there are some ideas you could consider when building something for specific CNC use ? I'd say the most important aspect is that CNC is much more "on hand" interface, actually potentally operating the maching thru macros (oups forgot to include it there, but it would be on left pane list) or jogging with the right parameter facing stock for instance...

let me know if you'd be interested to dig, I'd love to contribute usefully...

luc-github, 2021-01-23:
yes anything is good to share - I will look at it - as there are several command sI am not aware of

I may as a lot of questions sorry in advance:
What are `set x` `set y`  and `set z` buttons?
What about B and C axis?
What is `150 G1 work` button?
What is `1200 G0 rapid` button?
What the difference between console part and the text under TestGrbl ?

May I know where these icons come from ?

eglrd, 2021-01-23:
- Set X and Y are to enter a given value (you might want to start a feature in a position that is known not to be 0). ex G10 L2 P1 X-10 (to set current position at -10... and not 0, that zeroing does). you might want this to start from a given feature in your work
- B and C would be accessibe scrolling if enabled in prefs (scrollbar in the void on the left of the DRO)
- G1 is "F value" when in "Run" mode (Feedrate). Depending on the controller options you might enter into G1 mode to do some "fly-by-wire" machining: i.e. jogging manually, limiting your feedrate to G1  "speed" and not G0 "Rapids" to machine feature without programming gcode !
- G0 is "F value" in "rapid" mode. i.e.
... essentially a CNC moves either in G0 (rapids when the head is is the air) or G1(work, when the mill engage with the material) and limit speed is set by F value for each via gcode - its a modal state
- console is current status and allow direct interaction whenever, as well as activating verbose mode to catch all event as desired. for ex on GRBL_ESP32 you can monitor state, change current values, etc.
- the text under testGrbl is the file loaded to be streamed / executed. the triangle next to the (TestGRBL) would ideally follow the buffer state of GRBL allowing to know where the execution is at (scrolling the text), so that if for some reason you need to "free hold", i.e. pause streaming, you know where you are at... and where you need to set things back to if you had to interrupt things (and change a broken mill for instance...)

icons are from "http://thenounproject.com/", with some quick modification from me. the only constraint in an open source project would be to include reference to their creators in the commentary / ressources section to use them.

eglrd, 2021-01-24:
mmm looking a the GRBL in more details (I'm coming from smoothieware using CNC.js) the "G1 150" could become "Jog 150", i.e. setting up the jog speed to a desired "working" feedrate... (having to be repeated for each jog move instructions as the F value is not modal when joggin). Again the philosophy here is to a allow "machine-by-wire" capabilities (=operate the machine as if it was a manual one, if this notion is not clear). I'm digging more into GRBL as I'm going so sorry if my past experience creates confusion.

luc-github, 2021-01-24:
Ho you are using GRBL emulation in smoothieware - interresting

DivingDuck, 2021-01-24:
It's available for Smoothieware for quite some time. You can compile it with option CNC or load the CNC version bin file.

luc-github, 2021-01-24:
yes I know - there is also lpc-grbl, I was thinking about difference between grbl emulation and grbl itself

eglrd, 2021-01-24:
> Ho you are using GRBL emulation in smoothieware - interresting

I'm using GRBL mode on Smoothieboard for one of my machine, but building a GRBL/ESP32 second one now based on Bart's 6-Pack_CNC_Controller, so actual GRBL I believe ? There are a few new seductive new options with this setup.

luc-github, 2021-01-24:
Yes GRBL_ESP32 is sticking to GRBL FW but improved,

eglrd, 2021-01-24:
> It's available for Smoothieware for quite some time. You can compile it with option CNC or load the CNC version bin file.

Yup, working great with a 4 Axis machine for me... with a few limitations though ? I don't think the jogging interrupts of GRBL1.1 and above work as well on Smoothieware, if I'm not mistaking ? (I mean, for a Smothieboard v2, as the v3 is not yet out... !)

luc-github, 2021-01-24:
yes my understanding is that real time commands are not handled in real time in emulation unlike in GRBL

DivingDuck, 2021-01-24:
If I remember correct there was some code rewriting for jogging in the past tow or three month.
(But you are right regarding availability and communication too I guess)

eglrd, 2021-01-29:
@DivingDuck abandoned or am I wrong ? Temptation is high to go LPV_GRBL on my Smoothieboard now... pff.

@luc-github : should I give a try to try and style the wireframe I share into a more "material design" style ? (it was only a wireframe in my prec. posts)

luc-github, 2021-01-29:
@eglrd yes your UI is really disruptive (but that is startup Nation ^_^)

but this UI consider all control are enabled - what about if no Mist - should button still show up?
This UI looks CNC only oriented as you mentioned . What about laser?

The set x is actually the full button ? or just the small part of the button ?
The home buttons look really small , same for texts - but may be because it is just mockup ?
Is `G21` a key button that need to be so big / visible and accessible ?
I understand the white part as a state renderer (ON/OFF)  on some button  but on limit switch I do not get what the button should do 
>the text under testGrbl is the file loaded to be streamed / executed   

But this is not available in webUI or is there a kind of  `echo` of each command when they are executed (coming from SD or strem app) ?

can you also  also clarify to me what commands are :
Stop
Reset
Cycle Start
Free Hold
Step

also I am curious how you see the probe UI 

I understand the tab UI you describe but where the `emergency stop` button shoud be displayed in such UI ? it should be always visible - right ? 
You said UI is for right handed ? so this is UI for specific resolution ? what about tablet and PC ?

Sorry many questions but this is unclear for me, because before going to material design all should be defined in wireframe to be sure it is right way - no ?

jeffeb3, 2021-03-13:
I have used CNC.js to control my CNC machine a lot, and I like the job area there. It is worth looking at closely. 

https://cnc.js.org/

There is also a "tinyweb" template for very small screens:

https://github.com/cncjs/cncjs-pendant-tinyweb

Coming from a 3D printer world, and entering the CNC world, there are a few things that are different, but not obvious:
1. In 3DP, the coordinates are set, and you use the bed, and the distance from the 0,0 for printing. In CNC, the workpiece may be a piece of drift wood, or a trapezoid or parallelogram of MDF. The key coordinate in CNC is the workpiece origin. I usually use the top of the workpiece for Z=0, or and either the lower left corner of the work, or the center if I am trying to center some art. Grbl also keeps track of "machine coordinates" and "workspace coordinates" separately, so you can easily home the machine to set the machine coords, and then explicitly set the coordinates on your workpiece. If you need to take a lunch break, you can return and home the machine again, and the workspace coordinates will still be at the same place relative to your chunk of driftwood.
2. The machines can be BIG. Mine has a work area of 36"x48" and there are similar machines that are a whole 4'x8' that are using DIY electronics. A button to return to the zero (in x, y, or Z, but not all at the same time) are really helpful.
3. They are more dangerous to leave alone. It may be counter intuitive, because they don't have intentional heaters, but they can cause fires more frequently, and there are no safety systems. Not sure exactly how this applies to a UI, but it's worth pointing out.
4. The feedrate and spindle speed should be easy to tweak. Maybe because I'm just always standing there, I change them a lot.
5. For a laser, you want to be able to quickly turn it off, and a preset for 2% is nice for lining things up. In the little TFT35 bigtreetech software, we have 0,2%,100% buttons, and then +/- 5% buttons.
6. A 4 axis machine is rare. A 6 axis machine is very very rare. And I think a lot of the bigger machines are using things like Mach4, or LinuxCNC. If you focus on 3 axis, and maybe 4 axis, I think you would be fine.
7. grbl can do inches, and a lot of mechanics and woodworkers in North America prefer it.
8. A lot like the duet settings, everything is done in $ settings. In the version I have, there is a quick (<80 characters) note about what the setting is. That is really helpful to remind me which one is which.
9. I spend a lot of time helping in the V1Engineering forums, and my CNC is a low rider CNC from there. The common firmware is Marlin, with EXTRUDERS=0. We configure it for users for a few boards (skrs, rambos, ramps). The common controllers are the TFTs, LCDs, a pi running cnc.js, or repetier host (which is my least favorite). To support users like that, you would need to have a version that conversed with Marlin, but had the grbl UI. I am not saying this is a great idea, because there are already a lot of options there. But it is worth knowing about.

I personally like the cnc.js DRO area a lot. I don't really need to see 7 sigfigs on each axis. But don't come from a bridgeport mill background. I think hiding things like coolant and spindle controls are a good idea, because a machine will either have it, or not. There's no good reason to keep it there on a machine that doesn't. The machine and workspace coordinates are very useful for every job. Macros are great, but some shortcut buttons like 2% laser, go to X=0, etc. would be useful too.

I don't really agree that I will be pushing many buttons from muscle memory. I would probably position my finger, then look at the bit and then push the button a dozen times.

DeeEmm, 2021-07-17:
I think that the current GRBL implementation is really not too far from where things need to be it is already very usable. I agree with the comments relating to CNCjs as I too use this, it's really good and very usable. There's not too much difference between the two from a usability perspective. At least when used on a computer. Tablets are a different matter however.

I like the bold easy-to-use-on-a-tablet interface that @eglrd eglrd posted above. Being able to use a tablet is a big plus for a CNC IMHO. Take a look at any professional CNC and they will have a similar interface. (Typically a HMI)

It is worth noting that the operating environments for CNC's are very different to a 3D printer, often they are in a dirty / messy environment like a workshop, so many users will not want to use a computer or laptop. For this reason a UI that can be operated via touchscreen is a big bonus as I think that many people will be like myself who have specifically switched to ESP32 for the WEBUI capabilities and being able to use it with a tablet.

I have run tethered machines with both Mach3 and Linux CNC and specifically moved away from that kind of setup as the workshop is no place for a computer, plus I find them to be either feature bloated (mach3) or unnecessarily difficult to setup and maintain (LinuxCNC). I currently have machines running from both dedicated offline CNC controllers as well as Arduino + RasPi using CNCjs. So for me ESP3D-WEBUI (specifically on GRBL_ESP32) is a really good fit for an all-in-one solution that can be used with a cheap tablet. The tablet can be utilised like a HMI - mounted to the machine OR removed from its mount and used as a remote pendant for job setup. Plus they can be mounted in protective cases which make them more suitable for the aggressive environment that these machines are often used in.

Another consideration and really the main usability difference between 3D printers and CNC's is the workflow of how you set a job up. For a 3D printer you have homing limits. Remembering that you always start from an empty bed, so it is always the same. You home the machine and you are good to go.

With a CNC, assuming that you are using a CAD > CAM setup like Fusion360. You input your stock dimension into your CAM software and it generates the necessary Gcode. You mount your stock to the table and then set the machine to the stock. So to set the job up you do not (generally) use homing limits, instead you zero your work coordinate system (WCS) on the stock  (just as you did in your CAM software when you generated your Gcode.) 

So the ability to control / zero the WCS from the interface is an important distinction. Of course you can add this via macro commands, but having the ability to do it natively from the UI is normal with a CNC (just as shown in @eglrd's rendering).

You can then augment this with macro routines. Generally you use a touch probe and touch off on all 3 axis. before moving to home

Another CNC only feature associated with the WCS is move to WCS Zero - Each axis individually or all together.

Again mosts CNC's do not need / use homing. In fact the homing limits most hobby machines use are so inaccurate that they cannot be used reliably.  So the WCS and the ability to move around the WCS is much more important on a CNC than the ability to home or move to home. That said. Some machines do use homing limits and for these cases the homing functionality should be retained as well. 

With regards to @jeffeb3's comments on multi axis machines, I both agree and disagree. GRBL_ESP32 supports up to 6 axis and also adopts ESP32_WEBUI so I think inclusion for up to 6 axis should be considered. But, with that said, I am building a 5 axis machine and would use a dedicated controller for it.

I think the ability to enable / disable elements is a good idea, whether this is from settings within the UI, done at compile time or perhaps using uploadable files for each feature - a bit like CNCjs's ability to handle plugins. I know that there are memory constraints so I know that a full on plugin system may not be achievable but being able to customise the interface with features that are relevant to your machine would be good. 

As you are aware I'm working on integrating automatic bed levelling so the ability to integrate this functionality (or any other functionality) by simply including the relevant files makes extensibility much easier.

Which leads on to separation of code, style and function. Essentially just using good W3C code segregation practices. By keeping these items separate addition of extra features becomes more manageable.

It would also be easier for the end user to design new UI's that retain the same JS functions and code hooks, but are visually different. 

To a degree I think that the code is already structured like this (no inline style). 

/DM

DeeEmm, 2021-07-17:
> [image]

I love this. It is heaps better than the standard layout for use with a tablet. 

For comparison here is the CNCjs 'shopfloor-tablet' layout

[image]

DeeEmm, 2021-07-17:
> * They are more dangerous to leave alone. It may be counter intuitive, because they don't have intentional heaters, but they can cause fires more frequently, and there are no safety systems. Not sure exactly how this applies to a UI, but it's worth pointing out.
> * For a laser, you want to be able to quickly turn it off, and a preset for 2% is nice for lining things up. In the little TFT35 bigtreetech software, we have 0,2%,100% buttons, and then +/- 5% buttons.

From a safety perspective devices need hard wired protection. i.e. Laser power needs to be controlled via a N/C contact on the Estop NOT be controlled from software. Same for motion safety. The power to the steppers also needs to be run through a N/C contact on the Estop.

Generally the correct way to wire this is to switch a MCR (master control relay) that cuts power to devices such as the laser supply, the stepper drives and spindle power. When you hit that Estop all power is shut down instantly.

If you have a machine that is capable of physical harm, or used in a public environment then your MCR needs to meet category 3 or category 4 safety requirements. This governs the ability of the relay to be used in the case of a fault with the relay. These are generally dedicated safety relays like a Pilz. 

The determination of whether a category 3 or cat 4 system is required is usually determined by a risk assessment, but generally if it has the ability to cause death or permanent harm a cat 4 system is required.

In no cases should software be relied upon to control safety critical features. 

Safety is a wiring problem. NOT a software problem.

SnowBeaver, 2021-07-17:
I agree with @DeeEmm  on opting for a tablet instead of a laptop to control the CNC. I am building a wood cnc and I specifically chose Grbl_Esp32 for its ability to control the cnc from a tablet. The other convenience is to be able to monitor the cnc remotely. One thing I would appreciate for remote monitoring, is the see the percentage to the job execution and when is next tool change and which tool it is.

Is it possible to monitor the current or power drawn from each steeper motor? This is to monitor if too much burden is applied to the tool.

luc-github, 2021-07-17:
the job status is a available in ? feedback but how to know the tool change?
same question for power draw, how to get the value?

SnowBeaver, 2021-07-17:
Control tab for machine calibration. In my setup I do not have a local PC, all commands are done on a tablet. 

I have yet to gain experience with a cnc, but actions that comes to mind:
step motor pulses per unit of displacement. Displacement should be as large as possible for accuracy. Have a repeatability function to check how small and large displacement are accurate.
XY Squaring
Limit switches settings according to squaring.
There are probably more calibration items for 3D printer, laser, etc. to include.   

Squaring

MitchBradley, 2022-02-01:
I just now saw this issue and would like to comment on a few items.  I am a FluidNC developer who is working on refining and implementing the interface between FluidNC and WebUI 3.0-rewrite.

> I agree with @DeeEmm on opting for a tablet instead of a laptop to control the CNC.

The current version of FluidNC uses a modified version of WebUI 2 that has a tablet-specific UI (in addition to the normal WebUI dashboard) under a separate tab.  The tablet UI has large buttons that are easy to hit with fat fingers, a fixed non-scrolling layout so everything stays on the screen without tricky multi-finger gestures to bring a control back onto the screen, a simple GCode visualizer, and other features to make it easy to use and safe in a shop environment controlling a dangerous machine.  That tablet UI currently only works with WebUI 2 but I intend to port it to WebUI 3 soon.

> Control tab for machine calibration ... step motor pulses per unit of displacement ... XY Squaring ... Limit switches settings according to squaring

Those specific settings are part of the controller firmware. WebUI (all versions) has a screen where you adjust them.  WebUI asks the firmware which settings are available and lets you view and change their values.  So this wish list item is pretty much a "done deal".

> Spindle RPM

Spindle RPM control is complicated
* The GCode program sets the desired spindle "speed" by issuing an S word like Snnnn, where nnnn is the desired speed.  nnnn is often interpreted as RPM but in reality it is just a number in arbitrary units.  For some "spindles" like lasers or plasma torches, RPM is not a meaningful unit.
* Firmware setting establish the correspondence between S values and the actual spindle RPM or power. This can be fairly complex because of the many different hardware ways to control spindle speeds - PWM, 0-10V analog, Modbus serial, etc.  Another complexity is the fact that spindles have a maximum speed and often have a minimum speed below which they will not work, and the mapping from hardware control values to RPM is often nonlinear.  FluidNC has a very flexible method for handling all of those factors and even supports multiple spindles on the same machine, with different settings for each one.
* Spindles take time to change speed.  Motion must account for that because you will break a bit if you start moving before the spindle has "spun up".  Controller firmware has ways to configure spin-up and spin-down times; FluidNC automatically adjusts the times based on how much the speed is changing.
* Once the GCode program has set the desired speed, most control firmware like GRBL and FluidNC has a way to "override" it, by boosting or reducing the speed in, say, 10% increments.  This is typically done by clicking on a UI element that then sends an override command to the firmware, modifying the speed that was established in the GCode.
* For laser "spindles", the firmware can adjust the power automatically depending on the speed of motion, to avoid burn spots when the motion has to slow down to turn a corner.
* Plasma "spindles" need another layer of automatic control to ensure a consistent cut.
* The firmware can report the speed setting (GCode S value as modified by overrides) so the UI can display it, but that is not necessarily the actual speed.  Depending on load and other factors, the spindle might not be running at the speed that you asked for.  High end spindles sometimes have sensors to measure the actual speed and report it, but that is uncommon on hobbyist spindles.

MitchBradley, 2022-02-01:
> Is it possible to monitor the current or power drawn from each steeper motor?

Some stepper drivers like Trinamics have such monitoring, but getting the information back to the controller is very tricky and highly dependent on the particular driver.  Some closed-loop stepper drivers have an alarm output than will shut down the stepper if it overloads too badly.  That too can be tricky to handle in firmware.

DeeEmm, 2022-02-01:
> Plasma "spindles" need another layer of automatic control to ensure a consistent cut.

I can shed some light here. I run a plasma and have used it with various controllers (Linux CNC / Mach3 / etc) The additional requirements relate to torch height control  (THC) and 'touchoff'.  There is no spindle 'speed' control as such, the torch is either on or off. 

Touchoff is very similar to a probe routine. The head is usually floating and the torch tip is driven down to contact the work piece to determine where the top Z height is. This action operates a limit switch (which is why the head 'floats'). The touchoff routine is generally contained in a macro and leverages the built in probe input and M101 / M102 Gcode commands. 

Touchoff code is added at the post process stage and so in this respect is transparent to the controller as it is simply part of the program - much like the Torch On and Torch Off commands. 

I'm not sure of the value of trying to bring touchoff into the GUI or if it even has a real-world use, as each arc start move in the program really needs to proceed a touchoff as the pierce height and cut heights are critical to good cut quality. A manually triggered routine could be used for initial setup but I personally just note the Z axis DRO values and manually drive the head until the limit operates. This is something that you only ever need to do once. You do not need to do it every tool change like with a probe routine.

The big difference between the standard probe routine and a touchoff routine is that with a standard probe routine the measured probe height is basically Z=0 whereas with the touchoff routine the trigger height is below Z=0 due to the additional distance that the Z axis needs to move to trigger the limit once the head contacts the workpiece. For this reason an offset is required to be subtracted from the triggered height to get Z=0. In this regards it is not possible to utilise the existing probe routine without modification.

Regarding torch height control (THC) this is an external controller that measures the arc voltage and adjusts the torch height  (Z axis) to suit. AFAIK there are no GCode processors (GRML / Marlin / etc) that implement THC or provide a provision for external THC inputs (Z adjust +/-). Generally for this kind of controller setup the THC controller injects steps directly into the stepper driver control lines in between the controller output and stepper driver input. The PROMA THC controller is one of the few THC controllers that can be used with this type of hardware setup (GRBL / MARLIN / etc) as it provides step +/- outputs

So there is no need to specifically write a 'plasma widget' to be able to use the ESP3D / GRBL / Marlin with a plasma, but if you did want to, then you could create an updated version of the probe widget that included provision for a Z height offset. Perhaps this is actually a valid change for the existing probe widget as it adds flexibility for any machine using a limit switch or capacitive / inductive sensor instead of using the actual tool as a probe, as all of these setups have a z offset.

SnowBeaver, 2022-02-03:
I suggest to add a project management panel. The objective is to layout the necessary steps to complete a manufacturing job job.

Multi-axis controlled machine are becoming polyvalent as they can milled, add (printer) material, scrubbed (laser) write etc.. Tool can coexist side by side, such as a router and laser, and could be run automatically within the same job. In that context, even for a simple cnc job, there are so many details and particularities that it is not practical to remember everything and potentially leads to low quality product. For that reason, we need a tool to organize the steps to manufacture an object.It can be seen as a sequencer where a single action is executed for each step.

I do not want to look too far ahead and kill the idea by adding too many features. So I suggest to focus with present needs and add features as required. At minimum, the project panel would layout the steps to complete a job and stop where tools need to be change. In the long term, it could embrace several processes (cnc, printer, etc..) and manage an elaborate manufacturing project.

To the operator, the management panel is organized as a spread sheet. Each row is an action to be executed. I believe IFTTT is already implemented in the webui, so notifications can be send from the webui to the operator. While browsing the Project file, The Project Management panel could display the content of the zip file first row to provide information to the operator of the zip content. Once the zip is loaded, the operator presses Start to launch the sequencer. The sequencer pause when the operator keyword is reached. Sequence resume when operator presses Resume.

The project could be stored in a zip file or any suitable file container. The zip  includes all the nc files and the project file. 

First row is job description
Second row is the file version
Third row Column Titles (in case user want to edit outside the Webui)

First column: indicate where the actions goes: CNC, Laser, Printer, Operator. When operator is selected, activities are halted for a manual intervention.
Second column: nc file to send to the machine
Third column; required notification via IFTTT or other
Fourth column: Tool required. To instruct the automatic tool changer.
Fourth column: Description

See attached a picture of  a project example to manufacture a sign on wood.
[image]

DeeEmm, 2022-02-03:
This functionality is already present in the post processor stage, It is transparent to the machine. If the machine has multi operation capabilities then it should be part of the CAM process not a function of the machine.

Essentially fist operation > tool change > second operation > tool change > etc.

SnowBeaver, 2022-02-04:
@DeeEmm  Which post processor are you referring to?

DeeEmm, 2022-02-04:
The post processor should be part of whatever CAD / CAM program you use. If you use a workflow that integrates several separate programs then your post processor aspect may not have such capability as it does not directly integrate with the entire process.

I use Fusion360 myself, it's a pretty common CAD / CAM solution. The post processor can be custom written to match the machine type. So if you have a machine with multiple functions and a job that requires several operations with several different tools, the program is generated with all of the necessary tool changes and machine reconfigurations and a single program files is generated. So there is no need to sequence a bunch of sperate programs.

For a multi purpose machine that has automatic tool change functionality it is a set and go program. For something like a 3D printer that has interchangeable heads or a router that does not have auto tool change, the program will wait for user interaction such that the user changes the head or tool at the appropriate time and then takes the machine off of hold so that the process continues on with the next operation.

This is pretty standard stuff and nothing new but for extreme use cases or non-standard machines would require that you custom write the post processor. In Fusion360 this is in a basic language and can be written by the end user. Here's the one I wrote for my plasma as an example if you are interested - https://github.com/DeeEmm/DDCSV11-Plasma For GRBL there's already a post processor that works out of the box.

The case for combining 3D printing and traditional machining on the same machine is a special use-case as this requires a Marlin based controller as AFAIK this is the only firmware currently capable of both NC and 3D FDM functionality. I have an MPCNC that I have set up like this for both 3D printing and CNC work.  Fusion does use an external slicer to convert models for 3D printing and so whilst I'm pretty sure that Fusion360 could integrate the slicer output to produce a combined output, I have not actually tried this myself as I only ever use one or the other process on that particular machine.

Most 2.5D programs should have auto tool change functionality but you obviously need a tool change unit to automate that. They should however have the ability to queue up manual tool changes so that the machine goes into hold to both indicate to the user and allow the user to undertake the tool change. This is where a machine hold indicator lamp is useful. This functionality does however come down to whatever program you use. Some are unfortunately little more than toys and getting multi tool functionality requires stepping up to a better CAD > CAM solution.

Trying to get multi head capability from most 2.5D programs is not possible as programs such as those have a very narrow functionality. So to do something like mix a profile cut part with a drilling routine or multiple operations on a 3D part (cutting / surfacing / facing / pocketing / drilling / etc) means you need to use several different programs and manually stitch the code together. In this case it is better to use something like Fusion360 which has the capability to cater for all aspects of the job (unless you don't mind manually joining the programs). Using the machine itself to schedule the jobs to stitch together separate program in my opinion is not the right thing to do, it's more a case of using the right tool for the job in the first place if you are trying to automate this process.

Simply cutting and pasting all of your seperate programs into one file will also work, but be sure to add hold points in between each section to allow you to change the tool. You can easily write a macro to automate this.

I've never regretted the time to learn Fusion360. I moved away from using Sheetcam on my plasma to Fusion360. I lost a few features such as automatically ganging up multiple parts but this is available using plugins. Then later when I built a 3 axis CNC I was able to jump straight in and create jobs using the same basic CAD skills on the same program, it just uses a different post processor. Similarly with using a laser, it is just a case of choosing (or creating / editing ) a post processor to work with your machine. Most standard machines will work with the included post processors.

The may be one caveat to this in that Fusion will not do raster conversion which some of the 2.5D programs do quite well. This is a specific use case and one that I know a lot of 2.5D router and laser users use to convert images to machine paths. However it is still possible to do but does require some image pre-processing, either to convert to SVG or to manually trace. In the case of manually tracing this can actually be done directly within Fusion360 using the sketch tool, by importing the image as a layer within a sketch and then tracing it using bezier curves - essentially exactly the same as you would using inkscape or whatever other SVG program you use. The real benefit here is when you want to make edits you can easily do so without needing to modify the path in an external program and re-import.

For the case of converting a raster image to greyscale for laser etching then there's really no other option but to use something like lightburn.

HTH

SnowBeaver, 2022-02-05:
@DeeEmm 
Wow, that is a very detailed reply and you speak from experience. I appreciate that.

I am relatively new to cnc, so I have less mileage in this area. I build my cnc and I am using FluidNC as the controller. The program is still in beta but for what I am doing is it a robust platform developed by highly skilled experienced peoples. The development team are responsive to users. FluidNC offers configuration flexibility without recompile. 

I use the controller (6-pack) that has been developed by the FluidNC team. It has key features I was looking for, wifi,  onboard SDcard, onboard stepstick and sockets to plug different accessories. For my cnc, 6-pack was all I needed, no exterior device required. Wifi allow me remote control and monitoring. Wifi allow me to use a tablet to jog and control my machine, and eliminate the use of a wired control station. As a hobbyist it is a simple and robust solution.

To access the controller, I use the web interface that was developed by Luc. It is a simple interface for configuration, setup and job control and monitoring. The later is a efficient tablet interface developed to be used with fingers. I looked at other controller interface, and I did not find any that had wifi and uses the onboard SDcard to store the nc file. Streaming command and data over wifi is not 100% reliable in my view. The way to go is a controller hosting a webui and onboard sdcard.

This ecosystem requires to upload the nc to the controller.  To generate the nc files, I use different programs. I tend to favor freely available solution. Since nc files comes from different directions, I cannot automate or document a cutting job using a single resource, such as fusion360 or other similar software. My only solution resides in the 6-pack/FluidNC webui. I could store everything in a zip file and write a document about how the cutting should be organized and executed, but that can easily be integrated in the webui panel and on top of that the nc files can be send in sequenced as per given instructions.

My proposal is simple to put in one place the nice files to do the cutting and document that process. I think it is a relatively simple addition to the webui where files and instructions will be consolidated in on place/container (zip for example). The benefit is the cut/laser job becomes documented and organized.

MitchBradley, 2022-02-05:
(I am a FluidNC developer and the author of the tablet UI). The way I do this for my production jobs is as follows.  I put all the gcode programs for a given part in a separate folder named for that part.  The file names are prefixed with the step number, like 0StockPrep.gc, 1TopRoundover.gc, 2BottomProfile.gc, 3Cutout.gc.  Each file begins with a comment line that tells the tool and the origin position, like  (3_16 EM, stock URHC).

DeeEmm, 2022-02-05:
@SnowBeaver Yes I too use FluidNC on one of my machines, although I'm still currently on the older version (GRBL-ESP32) which alsos uses Luc's interface. It is indeed a great project. Hats off to Mitch, Bart and Luc for all of their hard work.

Without wanting to take this too far off topic but just to share - Fusion has a 'maker' license which is free although there are some limitations but these generally all have a workaround. For many it is overkill and has a steep learning curve, especially if you've never dabbled in parametric modelling or CAD before, but as I said I think it's a worthwhile challenge. There's plenty of youtube tutorials to help you get to grips with the CAD > CAM workflow. It's quite satisfying not to have to use three or four programs to create a job.

Mitch's comments just reminded me that you can also use a Gcode program to call other Gcode programs, it's not something I've tried myself however and it has different implementations on different controllers. The code is M98. I'm not sure if / how this is supported in FluidNc / GRBL but it allows you to manually manage programs like Mitch has suggested and then create a master program that calls the individual programs. 

This master program essentially becomes a very simple program to create once you have learned the syntax. You can also easily add hold commands and maybe even a macro for the manual tool change to move the machine to a tool change location and wait for user intervention.

Subsequent jobs just become a case of cutting and pasting the new sub program details into the master program sequence.

MitchBradley, 2022-02-06:
FluidNC does not have M98

DeeEmm, 2022-02-06:
Ahh well, there's your answer.

This is where stepping up to a dedicated NC controller can have some advantages. GRBL does have a somewhat reduced function set, Marlin even more so. I've ended up moving to a dedicated NC controller on a couple of my machines but that's a completely different kettle of fish. There's advantages and disadvantages to all setups. 

Perhaps there's value in UI based sequencer but I'm not sure how it would handle multi-threading (it needs to jump out and then back to the main routine) I guess the same limitation also applies to FluidNC @MitchBradley ?

luc-github, 2022-07-14:
GRBL UI for Spindle and laser is now implemented :
[image]

More features may come later but at least the basic are there

Thank you for all suggestions, they were very useful

closed 2022-07-14 by luc-github