luc-github/ESP3D-WEBUI #146: [FEATURE REQUEST] Add titles to buttons

skorokithakis, 2020-12-03:
The buttons don't have HTML titles (tooltips), and it's hard to tell what many do. Adding title attributes would help greatly.

luc-github, 2020-12-03:
this is planed in 3.0, all text, UI and translations are rewriten

skorokithakis, 2020-12-03:
Fantastic, thanks! Would it be possible to add a "go to zero" button on axes (for times when there's no homing).

luc-github, 2020-12-03:
there is such function in 3D printer UI 3.0 - you can set any position actually 
[image]
In 2.1 you can use macro button to do this 
please check 3.0 status: https://github.com/luc-github/ESP3D-WEBUI/issues/94

for GRBL, discussion is still open - I do not have a clear UI defined yet : https://github.com/luc-github/ESP3D-WEBUI/issues/144

related features may also apear here : https://github.com/luc-github/ESP3D/blob/3.0/ESP3D-features.xls?raw=true

skorokithakis, 2020-12-03:
Oh that looks fantastic, thank you! Yes, I'm doing it with a macro currently (in ESP32-GRBL).

luc-github, 2020-12-03:
feel free to join/discuss/submit any suggestion in above ticket for grblui, it is not defined and can be different from 3d printer one, I have splitted the code to give more flexibility
you can also join discord chanel to discuss new wished features

skorokithakis, 2020-12-03:
I will, thank you!

luc-github, 2020-12-04:
so I guess issue can be closed as it is already in 3.0

closed 2020-12-04 by luc-github