luc-github/ESP3D-WEBUI #147: Temperature panel problem

Mdcompositi, 2020-12-06:
Hello
I have a problem with temperature panel
I only displayHEATER T0 and when I start the heating, the heating of the bed starts (not the hotend!)
in a nutshell I have  in the panel only HEATER T0 ( which however heats the bed) and instead the second heater is missing which should be BED

FW version 2.1.1.b8
WEBUI version 2.1.b68

Best regards
Maurizio

luc-github, 2020-12-06:
you need to enable these control in preferences
[image]

luc-github, 2020-12-06:
@Mdcompositi did this answered your question ?

Mdcompositi, 2020-12-06:
Many thanks
Problem solved!
Now all works fine!
Thanks for sharing......

closed 2020-12-06 by luc-github