luc-github/ESP3D-WEBUI #154: New login forces the current user off[BUG]

Hagar01, 2021-01-16:
**Describe the bug**
While setting up and logged in as admin, my son in his curiosity tried to log in as a user.
This forced me off without asking.

**To Reproduce**
Log in from several sources at once.

**Expected behavior**
The UI should ask the current user if they have finished and are ready to let go.
If they are in the middle of doing something then it could upset things.

**ESP3D Firmware:  (please complete the following information)**
- Version: 2.1.1
- Web UI Version: 2.1
- WiFi mode: Client
- Vanilla

**Target Firmware:  (please complete the following information)**
 - Name: Marlin
 - Version: 2.0.7.2

**Board used (please complete the following information):**
 - MCU: ESP8266
 - Name: ATmega2560+ESP8266 R3
 - Flash size: 4M

**Browser used (please complete the following information):**
- Name: Chrome, Firefox
- Version: Latest
- Operating system: Windows 10, Linux, Android

**Additional context**
It would be nice to allow up to, maybe, 5 or 6 logins at once (configurable).
With the first being Read/Write, and the rest being read only.
The subsequent users should then request write access with either a timeout or a popup on the primary login.

luc-github, 2021-01-16:
yes it works like this if you try to connect and failed you are disconnected

using more login is not on roadmap, currently system allow one login/connexion at once so all other pen connections are closed when new connection happen
I do not understand the user case of several read only users, can you elaborate?

Hagar01, 2021-01-16:
So any new user automatically forces the original user off?

A read only login would allow others to see if the printer is busy and then ask for write access.

Currently the next user forces the original one off.

luc-github, 2021-01-16:
yes as only one user can control printer and webserver handle only one query at once

I still do not understand the user case - why a printer should have several users to access it at once?

Hagar01, 2021-01-16:
It is to NOT force the original user off.
This is bad behavior.
The original user should have the priority, NOT the second user.

Hagar01, 2021-01-16:
And why cant the admin log on to check the printer occasionaly?

They dont need to have write access to do this, just view the current temps etc.

luc-github, 2021-01-16:
>This is bad behavior.
>The original user should have the priority, NOT the second user.
Ok this is your point of view but this is current design, what happen if page is open but no one in front to let access ? others connections wait for current user to release the page ?
I do not plan to handle complex situation - one printer = one user 

>And why cant the admin log on to check the printer occasionaly?
why is there a need for this ? 

I could imagine a classroom / Fablab management but this is not implemented yet

luc-github, 2021-01-18:
I think question has been answered, so I close issue then

closed 2021-01-18 by luc-github