luc-github/ESP3D-WEBUI #155: Split Pages - No single monolithic page

Hagar01, 2021-01-16:
**Is your feature request related to a problem? Please describe.**
The monolithic page can take ages to load. Separating out the pages may help keep the load time and memory usage down.

**Describe the solution you'd like**
I would like to see a more relaxed list of files that can be accessed as needed, rather than everything at once.

luc-github, 2021-01-16:
using monolitic file save flash size actually
split file in different page wilk increase the flash need which is already critical on sole systems

the 3.0 has smaller size and use language pack to limit loading resources,
the time to load should only take a little the first time, after file should be in cache, but even it should not take ages, additionally webserver allow one connection at once so spiitting html css js will actually not decrease loading time

what board seetting do you have? may be something is not correct to be so slow
can you share the `[ESP420]plain` output?

Hagar01, 2021-01-16:
I have 1.7 Mb of flash memory.

Currently index.html.gz uses around 170Kb.

I understand that those with smaller flash need this - but what about those with space to spare?

I am using an ATmega2560+ESP8266 R3 all in one.

It has 4Mb for the ESP8266 formatted for 2MB Flash, 1MB Code.

luc-github, 2021-01-16:
well I handle all configurations so the minimal is the reference, 170KB is with all languages - WebUI 3.0 is currently under 100KB
nothing prevent you to do you own webUI  
and again the loading time should not be long as file is in cache

luc-github, 2021-01-18:
ok if no update on this I close issue

closed 2021-01-18 by luc-github