luc-github/ESP3D-WEBUI #156: "Show when starting" for more panels?

Legsmaniac, 2021-01-17:
While using ESP3D 2.1.x on my printer, I'm currently testing 3.0 on a spare Wemos D1.

**Is your feature request related to a problem? Please describe.**
Not a problem, more an annoyance.  I realise version 3.0 is only just starting and under development but....
Will you be adding more "Show when starting" options in Interface Settings for the other panels?

**Describe the solution you'd like**
I'd like to be able to click Dashboard and see all my required panels without having to set them up every time the page is refreshed.

**Describe alternatives you've considered**
Um.....  I just considered checking with you so here I am. 

**Additional context**
Loving 3 so far, keep up the good work!
Love the "Add you own panels" section, I've created and uploaded some HTML to add a panel to control my lighting on the printer which runs WLED on a Wemos D1.  Very handy.

luc-github, 2021-01-17:
It is was not planned but I may consider it as nice to have

Legsmaniac, 2021-01-17:
That would be great.  Hope you do.  Looking forward to it.  Thank you.

closed 2021-01-17 by Legsmaniac