luc-github/ESP3D-WEBUI #157: [REFACTORING]3.0 Preact refactoring->3.1

luc-github, 2021-01-26:
ESP3D-WBUI  3.0 is using Preact and bootstrap 4.0
Current preact implementation is to be honnest,not really well done - that is my first Preact project and I have a lack of Preact knowledge - so I should consider it as a draft even it is currently workable.

So a rewrite is definitly necessary:
@alxblog did a great start on [POC](https://github.com/luc-github/ESP3D-WEBUI/tree/3.0-spectre-poc)
It is based on spectre which seems lighter than bootstrap but meet all needs of ESP3D-WEBUI and even more.

The POC approach was to start a full rewrite to not be limited, but that is a huge task and unfortunatly @alxblog seems busy with others things - when on another side my preact knowledge is still pretty low  😊 
But several approaches are really the way the webui need to gp:
 - use full native state/ hook/context Preact API  (e.g: no storeon lib,  no `document.getElementById` )
 - remove sliders when input is already available (e. g temperature controls)
 - use panel mode even for settings 
 -  separate code even for 3D printer FW and not only with GRBL
 - use real ESP3D emulator and even allow to connect to local Ramps / printer board to use it to simulate Printer / esp3d serial interactions
 - use modals and toasts instead of current dialog implementation  
 - no specifics buttons for Terminal and Files but a drop down menu for all panels
 - define in one place UI components for multiple use
 - simplify current UI as much as possible but keep fast access  to features
 - I really need to understand Preact Coding Mindset 👿 even reading current POC I still miss lot of concepts 😓 
 
 I think a full rewrite is too agressive (at least for me) and a step by step changes is the way to go  
 
  If anyone has suggestions/ideas / want to be my preact mentor(s) to achieve this I am ready ^_^

luc-github, 2021-01-29:
OK I have started several online trainings for web development - so this month I will focus on them

luc-github, 2021-02-17:
quick update as ESP3D activity is slowed down recently

I am still following  webDev training - I have ordered 30 days for all trainings, and actually it is like pulling infos and realizing at each new training, I need more training   orz...^_^

I have already followed CSS/HTML (110 lessons) then Javascript (standard and advanced, 198 lessons)  and definitly I was outdated 😓 

I am now on NodeJS training part  (standard/advanced, 224 lessons), then I will follow react training (standard/advanced, 195)  and may be will also follow the typescript training (58 lessons) TBC

To pratice I am rewriting the embedded page which is simple but use several bases of code for webUI I am trying to fit spectre POC UI for consistency :
[image]
I will include the esp3d emulator and offline test 
currently I am around 2KB by disabling the transpiling as the target will be modern browser and I do not use any new javascript feature 

I do not think I can finish all training within the registered month (now 10 days left and still a lot to learn) so I will register for additional month.

I hope results will be visible soon ^_^ , so thank you for your patience

luc-github, 2021-03-06:
Update - busy but still ongoing on rewrite
here current embedded vs new embedded webUI :
[image]
ersion
Final size should be almost same as current but build process using webpack and E06 allows test version, I am trying to simplify and keep a consistent UI with coming WebUI 3.0

luc-github, 2021-03-10:
The new embedded page is now part of ESP3D, the build process use webpack and so it will be easier to keep build libraries up to date unlike today, the UI is inpired from spectre css and I have tried to simplified a much as I could.

It has same features a latest embdedded page  but refactored code.
 
The offline test is now available with simple ESP3D emulator, not yet the one done by @alxblog because it is still under devt
the gain in size is 10% 6,55 KB => 5,94 KB
The webpack build process cover :
* build to index.html.gz from sources
* conversion from index.html.gz to embedded.h
* copy new embedded.h to ESP3D sources file

So in embedded directory:
* `nmp install` to setup environment
* `npm run dev` to start server and test UI
* `npm run build` to generate and update the file embedded.h

Note: the tool bin2c is still necessary to convert the binary file to a text file to include in sources

Feel free to test and comment, there is always area of improvement for sure  😸 

Now back to react / preact training

luc-github, 2021-03-24:
Some update:
The rewrite using proper Preact coding method is ongoing , most of work is based on @alxblog POC that I am studying to apply properly - thank to his patience for all my questions  and for completing my training on PreactJs 😅 
The UI is based on PreactJs / Spectre.css - we keep general flow of (3.0) but simplify several parts which make it lighter 

I am still on WebUi API itself implementation (settings management / websocket support / error rendering / modal dialogs implementation / code splitting,  etc...)- not yet on real pages rendering

Here some ongoing test with menu implemented and http call test for the about page FYR:
  
[image]

luc-github, 2021-03-27:
Now connection / disconnection pages will look like this

[image]

luc-github, 2021-04-05:
Update :
[image]

Current size of 3.0 rewrite is 30.6KB at same state 3.0 was 35.7KB, 

UI update		: 3.0.0.a4 : 30.6KB	   | 0.0.0.6  : 35.7KB

but rewrite has already authentication support as well as all translations, connection status, mobile view, so benefit of rewrite is already noticeable even some refactoring could still be done but I am limited with with my current Preact knowledge
This looks definitely very promising 😺

ETE-Design, 2021-04-05:
@luc-github This is so great, and seems like you have also added Update FW Support... Is this only working for ESP3D or also 3D Printer Mainboard?

luc-github, 2021-04-05:
Update ESP3D FW update is supported in all webUI version
about printer FW update support: only ESP32 based boards are supported (GRBL_ESP32, Marlin with esp3dLib)
Anyway, on recent 32bits boards, usually it just need to be present on SD and restart printer board, so this should be covered when https://github.com/luc-github/ESP3D/issues/575 will be implemented, but it will also need printer FW code modification, so not for short term I think.

ETE-Design, 2021-04-05:
> Update ESP3D FW update is supported in all webUI version
> about printer FW update support: only ESP32 based boards are supported (GRBL_ESP32, Marlin with esp3dLib)
> Anyway, on recent 32bits boards, usually it just need to be present on SD and restart printer board, so this should be covered when [luc-github/ESP3D#575](https://github.com/luc-github/ESP3D/issues/575) will be implemented, but it will also need printer FW code modification, so not for short term I think.

Okey, have an MKS Robin Lite v1.1... So Wifi is possible with a addon module... So if you need help testing the FW Uodate for Mainboard I can help on that part... Also know Marlin, so what need yo be added for it to work?

luc-github, 2021-04-05:
on MKS when theoricaly it is possible - in reality :
1 -  MKS marlin coderename the uploaded file to dos filenname with .GCO extension
2 - MKS bootloader use Robin_nano35.bin only for FW update 

So need fix in bootloader to support dos name as Marlin only support Dos name to write on SD, and prevent Marlin code to rename file extension if .bin

Both are out of ESP3D Scope sorry

luc-github, 2021-04-25:
I was busy with some IT issues ( my Nas died, Travis become slow, etc..) and some personnal things kept me busy
Here the first part of Features settings UI implementation

[image]

Still need to implement the save / validation / import and export / restart but rendering is now more consistent .
Some features will also need so code update in `[ESP400]` output format, but currently I have only modified the test server part 

I hope this part to be done next week

luc-github, 2021-05-01:
the features tab is done - it has some new features that need an update of ESP3D FW 
it allow to do all changes then save once, FW also now send information about necessary restart or not 
[image]

Next step is  settings interface tab - which will follow features tab style

ETE-Design, 2021-05-01:
@luc-github Seem's really nice, is it also possible to add SSID / Security Options when network is hidden?

luc-github, 2021-05-01:
what do you mean ? when Network is hidden you must type SSID and password - what is different ?

luc-github, 2021-05-22:
Update, the web UI settings page is not almost done, it still miss:
1 - proper language selection, based on what language pack is available
2 - proper theme selection, based on selected pack
3 - Macro editor
4 - extra panel / page editor

luc-github, 2021-05-26:
The theme and language support by external pack is now implemented, for the language packs, UI  still need to be finished to validate the list of string to translate - so current language packs are just for POC but API is ready
For theme support - it is currently limited to css, a POC of Dark mode is done as well as another purple just as example, I suggest to wait the UI is finalized before doing a final theme but feel free to play

luc-github, 2021-07-04:
Some update :
I got busy and  some family things got my attention, which also result of a big lack of motivation for a couple of month.
So the development was really slow down, I also found a nasty bug that fried my brain but it is now fixed, so next week I should be back on track and focus on webUI - the macro / extra panels configuration code should be implemented soon, code preparation and base is ready - now need to do the implementatio. :crossed_fingers:

luc-github, 2021-07-18:
Macros and Panels configuration are now implemented in settings, I have removed the color button customization because it is actually useless for panel / page, and it simplify the customization UI and the preferences.json file, also it may complexify the code at it would need also a push state color and hover state color that should be calculated, using space that could be used for better usage, and finally it may be weird with theme management as not take in account at all.

[image]
[image]

The error conditions in extrapanels / macros settings are not yet implemented because not yet well defined, TBD...

Next steps are more functionnals because they will use settings:
1- implement polling commands
2 - implement the extra pages display
3 - implement terminal panel in dashboard page
4 - implement jog panel in dashboard page

luc-github, 2021-07-18:
Update of UI with some conditions added 
[image]

DeeEmm, 2021-07-24:
Luc, do you have any view on a size comparison between V3 and 2.1? 

How does the size of deployed code for the configuration interface compare to 2.1?

luc-github, 2021-07-24:
No I did not do size tracker when doing 2..x , but 3.x refactoring work in more generic way than 2.x and bring more features

 Refering  final package size:
2.1 english only  is 106kb, 3.0 english only is 95.6kb 3.0 refactored should be 10kb smaller Ithan initial  3.0 I hope, when 3.0 has more features than 2.1 and refactored version has also few more features than 3.0

in all case the target is to have smaller size between 2.0 vs 3.0

DeeEmm, 2021-07-24:
Thanks Luc.

I recall some discussions about the 3.0 interface being modular, does this mean that for a CNC application we will not need to include the 3D printing components like temperature graphing and extruder control? (and vice versa) which saves space for other CNC specific features, or is there just a single deployment for all machines?

Just trying to get my head around the structure of the new code and potential available space for writing new features.

luc-github, 2021-07-24:
yes cnc and 3Dprinting are now splited avoiding to increase size  for not used features and also allowing different UI between cnc and 3Dprinting, that said cnc ui is not yet done in 3.0

luc-github, 2021-08-03:
2 - implement the extra pages display (finally done :+1:) got some nasty bugs,  which drive me crazy, but finally I understood my mistakes and fixed them :sweat: 
Next:
3 - implement terminal panel in dashboard page, and panel menu as sugested by @alxblog to have more generic UI
Clicking on menu item will make appear the panel if it was closed, and also put in top position just under menu, this will avoid any scrolling when want quick access to panel like in current 3.0 - the change is the drop down menu 

4 - implement jog panel in dashboard page

ETE-Design, 2021-08-04:
@luc-github Is ESP32-Cam now implemented, with all features?

luc-github, 2021-08-04:
what you mean by all features ?

ETE-Design, 2021-08-04:
@luc-github Like send files direct to it, and when done parse it over to the sd-card in the printer... This should speed up file transfer right?
Also Timelapse Screenshot mabye?

luc-github, 2021-08-04:
what is done in webui rewrite of 3.0 is what it is described in this thread , so no
as you see terminal is not yet implemented, other webui features will follow as described, if not described yet, it means not done yet in rewrite

about snapshot it can be added but I do see how to manage timelapse for the moment, streaming is managed by webui not esp3d.
being gcode host is not done yet in ESP3D and not related to web UI

luc-github, 2021-08-05:
I have just fixed several glitches when doing user case tests and also implemented the hold /stream button as well as the snapshot button 
[image]
Back to

luc-github, 2021-08-20:
I have almost done the information flow to filter  / render data flow and dispatch accordingly
The Terminal is not yet finished but send command and receive answer is working well, the extension support is also working well so new feature added 😸 
I did some test on OSX/Safari and when I have no issue on firefox / chrome /opera - well safari need some tuning, several css need to be added - I have issue with SVG crashing the drop menu and generating errors....
I did test on my 2011 / Sierra mac air, so let say: he is slow like hell and I cannot debug in such conditions.
I will try to setup a virtual Machine because better to clean up the rendering problem before going further and redo lot ot things.
So originally I expected to finish the terminal this week but as in Safari I cannot even drop down the panel menu..., I think It will be postponed to next week. in 2.1 / 3.0 I did some ugly hack for safari that I would like to avoid from now 
Just hope can do a proper virtual machine as I cannot afford to buy a new Mac 😢 
Let see what can be done - and finger crossed no hard issues will come 🤞

luc-github, 2021-08-22:
Ok finally setup the OSX virtual machine took longer than fix the css issues, I did not do anything about SVG as the error actually are not seen but just catched by console so...
the terminal panel is now done, I still have some work on data processing and filtering but it is working,
[image]

The extensions feature is also implemented (https://github.com/luc-github/ESP3D-WEBUI/issues/185 )- it should allow to add additionnal features easily  :[image]

DeeEmm, 2021-08-22:
Great work Luc,

I'm guessing that at this stage there's no documentation on how to create an extension, but is there an existing extension that would serve as a good example to take a look at?

/DM

luc-github, 2021-08-22:
Check from here https://github.com/luc-github/ESP3D-WEBUI/issues/185#issuecomment-899082944
sample code is  also in server directory

DeeEmm, 2021-08-22:
> Check from here #185 (comment) <https://github.com/luc-github/ESP3D-WEBUI/issues/185#issuecomment-899082944>
> sample code is also in server directory

Looks good Luc,

I'll take a more in depth look this week.

Thanks for all your hard work :)

/DM

luc-github, 2021-08-23:
Extra panels are not part of dashboard
[image]
I got some random panel corruption on with kri:moto as panel when closing then opening panel, but was solved by leaving / going back to dashboard - not sure what the issue as it does not happen on test server - so I am guessing some race issue...TBC

Next steps are the macro panel then the Filesystem panel

luc-github, 2021-09-07:
Quick update,
* Redraw and corruptions issues are now solved - all were related to context behavior - now I got it I handled it (I finally got some epiphany... learning curve is now in good shape) - this lead me to think the translation context is now not necessary because there is not `hot change` of language so I will remove it soon and so simplify the code,  in same spirit modals calls will be simplified as the previous calls were using context and it can now be simplified...
The code simplification should not affect the code size a lot but will improve code readibility a lot - TBD and TBC - need to rewrite a little the rewrite ... (never ending task...)

* Extensions and local images are now cached to avoid to query  webserver when not necessarily.

 * The files Panel is almost done - Flash support is implemented, the additionnal FS need some additionnal code but most of API is ready - thanks @alxblog for suggestion to use objects the code is easy to add new FS - the missing part is the Websocket answer management for `Begin files list... End files list` and corresponding formating.

*  I have added Drag and Drop feature for file upload with a minimal code addition 

Conclusion it is so far so good,😸      

[image]      

Next chalenge is to catch and display the files list from serial / websocket (e.g. `M20`) in a smooth way - it should be done for this week end 🤞

luc-github, 2021-09-11:
Stream API catching is now implemented  for Serial SD and should work with any others sources, TFT SD / TFT USB, etc..
Adding any additional FS : Shared SD,  TFT SD / TFT USB (BTT and MKS) just need one definition file per new FS, but as all my devices are now packed for my moving and in luggages - I will add them next month when  in new location.

[image]

After some code refactoring the size is now 53KB in rewrite for a equivalent of 57.1KB in 3.0 but with more feature in rewrite and better browser rendering - so still on good path.

Current code is now 99% Preact oriented and I did not saw any issue, all previous issues are now solved 🎉 
One random behaviour I recently noticed is: Modal randomly take time to show the first time when select a large size file, I am not sure  why this happen - TBA 

next step are:
* Machine settings tab
* More Machine specific code preparation
As most of my hardware are packed - real situation test will be limite until next month and I may only use simulator server : TBC

luc-github, 2021-09-14:
Machine settings is now done for Marlin as POC
[image]

Next steps:
- prepare splitted code for:
  * Repetier (3D printer)
  * Smoothieware V1 (3D printer)
  * Reprap (3D printer)
  * Marlin_embedded  (3D printer) (need new version of ESP3DLib)
  * Marlin Kimbra (3D printer) ? (TBC)
  * GRBL (CNC)
  * GRBL_ESP32 (CND) =>FluidNC ? (TBC new new version of entries points)
  * GRBL (Sand table)
  * Marlin  (Sand table)

-  specific Dashboard Panel for 3D printer :
    *  Jog panel and position display as POC
    *  Temperatures panel and temperatures/sensor display as POC

luc-github, 2021-09-15:
Devt and Build scripts have been added, as well as backbone code based on Marlin and need to be updated accordingly per target and sub target accordingly:
It is based on 3 main applications CNC / 3DPrinter / Sand Table, CNC may be splitted to laser / milling if necessary but as UI is not defined yet : https://github.com/luc-github/ESP3D-WEBUI/issues/144 - TBD

[image]

GRBL_ESP32 won't be supported but the new evolution will be : FluidNC
Reprap connected to serial and reprap connected so spi (embedded) have been added as futur support

I removed MarlinKimbra (https://github.com/MKFirmware/MK4duo) because sadly it seems project is now no more sustained 😞, I did not followed what's happened, it was a nice Fw with unique features.

So it will give more time for Reprap support addition.

If any target is missing - feel free to highlight it 😸

luc-github, 2021-09-22:
3D printer settings are on going:
Smoothieware:
* config.txt
[image]

* override
[image]

Marlin:
[image]

Repetier:
[image]

So far so good, but I have forgotten some features present in current 3.0 😞 

* one save button for all settings - currently  for each setting there is one button
* import /export settings : this one will be changed to use original FW format output instead of JSON one 
Once these feature are implemented on 3D printer settings - I will do the CNC and Sand Table ones

luc-github, 2021-11-08:
[image]
Jog for Marlin is now implemented for Marlin printer3d
I still need to add the positions control to information page but should be done today

Now POC is almost done for Jog, I will do it for all Printer3D and do the UI for CNC / Sandtable which is slightly different:
no Pronterface UI = just buttons, support up to 6 axis, WCO and MCO, Zero buttons for each axis, no disable motor, no parking command seems requested in https://github.com/luc-github/ESP3D-WEBUI/issues/144, so it won't be added neither.

luc-github, 2021-11-17:
Jog Panel for GRBL is now done :
[image]

The number of axis detection is dynamic and the axis homing command is set in settings, disable and stop command are configurable in settings, real time commands are usable in this configurable commands and are also show in terminal 
```
0x18 (ctrl-x) : Soft-Reset : #SOFTRESET#
0x84 : Safety Door #SAFETYDOOR#
0x85 : Jog Cancel #JOGCANCEL#

Feed Overrides
0x90 : Set 100% of programmed rate. #FO100#
0x91 : Increase 10%	#FO+10#
0x92 : Decrease 10%	#FO-10#
0x93 : Increase 1%	#FO+1#
0x94 : Decrease 1%	#FO-1#

Rapid Overrides
0x95 : Set to 100% full rapid rate. #RO100#
0x96 : Set to 50% of rapid rate.    #RO50#	
0x97 : Set to 25% of rapid rate.    #RO25#

Spindle Speed Overrides 10%->200%
0x99 : Set 100% of programmed spindle speed #SSO100#
0x9A : Increase 10%	#SSO+10#
0x9B : Decrease 10%	#SSO-10#
0x9C : Increase 1%	#SSO+1#
0x9D : Decrease 1%	#SSO-1#

0x9E : Toggle Spindle Stop #T-SPINDLESTOP#
0xA0 : Toggle Flood Coolant #T-FLOODCOOLANT#
0xA1 : Toggle Mist Coolant #T-MISTCOOLANT#
```
Next steps:
* add the real time commands support in Macros
* complete FluidNC Jog Panel and Sand table jog panel based on GRBL one
* try to add a dependency parmeter to show hide some settings like if X axis is set to hide - do not show settings relatated to it like feedrate definition to simplify UI as POC

luc-github, 2021-11-19:
I have added some control dependency API so now in settings to show/hide a control depending another control value is possible, by just adding the id of the reference control in UI json
I have implemented only for input control based on boolean value of reference control and applied the filter only to feedrate of GRBL
```
{
        "id": "feedrate",
        "type": "group",
        "label": "CN9",
        "value": [
      {
        "id": "xyfeedrate",
        "depend":[
            {"id":"showx",
            "value": true},
            {"id":"showy",
              "value": true}
          ],
        "type": "number",
        "label": "XY",
        "min": "1",
        "append": "CN1",
        "value": "1000"
      },
      {
        "id": "zfeedrate",
        "depend":[
          {"id":"showz",
            "value": true}
          ],
        "type": "number",
        "label": "Z",
        "append": "CN1",
        "min": "1",
        "value": "1000"
      },
``` 
This will open possibility to more contextual UI in settings
[image]

Next:
* add the real time commands support in Macros
* complete FluidNC Jog Panel and Sand table jog panel based on GRBL one
* show ESP3D sensors value in one panel
* start specific panels implementation
   - temperatures, extruders (3D printer)
   - status, override, probe (CNC)
   - Sand Table -TBD,  feel free to raise suggestion

luc-github, 2021-11-20:
3D printers  / CNC / Sand Table are now at same level  of code, variable POC in macro is ready - need to define the needed variables, for CNC / and Sand table the real time commands are part of variable List already
next step is

add sensors display and  add specifics panels:
* 3D printer       
   - Extruders controls
   - Fans controls
   - Flow controls
   - Temperatures control
   - Print status
   - TBD
* CNC   
   - Status
   - Override
   - Probe
   - TBD
* Sand Table
   - TBD

luc-github, 2022-01-08:
Project was not updated since end of November due to personal issue and travelling back to France, now I am back to Taiwan and ramp up to recover the delay. But as I am in quarantine hotel with limited test environment so please be patient.

As raised in discord, the direct SD (Marlin+ESP3DLIB /FluidNC) and Shared SD (all FW with proper HW ) support is not yet implemented I will work on it.
I do not plan to support M28/M29 file upload as it is only a source of complain and disapointment due to extremly slow speed

then I will continue on specific panel implementation

ETE-Design, 2022-01-19:
@luc-github If you use the "MKS Protocol" and of cause have an MKS Mainboard then the Transfer speed would be faster... Could you mabye just implement M28/M29 for us with that board - already use it for uploading and works fine here... There is a lot of MKS Mainboards around, just saying...

luc-github, 2022-01-19:
supporting M28/M29 is a big challenge now with all tft sending queries and new auto report features in Firmware, for a result which is in best case hazardous, slow and that will be used by very few people I prefer to spend my time  in something   more usable. if really need this protocol then better use the 2.1 ESP3D.

MitchBradley, 2022-01-20:
All that training that you did has really paid off. The structure of the new code is very clean.

ETE-Design, 2022-01-20:
@luc-github Was just thinking if it could be possible as you already can define in configuration.h so you only activate it if you have an MKS Board like here, would be a great feature for all the MKS People, mabye a FR you can look at later on?:
`//COMMUNICATION_PROTOCOL: to communicate with printer or TFT
//RAW_SERIAL standard serial
//MKS_SERIAL Mks protocol
#define COMMUNICATION_PROTOCOL RAW_SERIAL`

luc-github, 2022-01-21:
@ETE-Design sorry I am not sure to understand your meaning:
on MKS boards which already support faster SD transfer protocol it would be a great addition to add support of M28/M29 (slow like hell and buggy protocol )? is that your meaning ?

ETE-Design, 2022-01-21:
@luc-github Sorry thought you would remove it all, so if "Fast" SD Transfer is already supported them I'm glad :-) Thanks for your great work...

luc-github, 2022-01-24:
@ETE-Design yes it is in ESP3D 3.0  and WebUI 3.0 - but not yet in refactoring version of WebUI,  actually your message reminded me that I missed to add it in TODO list as well as MKS TFTSD and MKS TFT USB

nielsnl68, 2022-03-19:
hello @luc-github , i'm very new to FluidNc and ESP3d-WEBUI. Sadly i was not so fund of the WebUI that is used currently. I have done a lot of work with Javascript/html5 in the past and opted to help to make it a little bit nicer.
@MitchBradley (a dev on FluidNC) told me they where working on your new WEBUI3 refactoring project so it would work within FluidNc, and that it would be much better when i direct my energy to the new layout. 

At this point of time i have walked trough your code and as other said it looks very promising. There are still some thinks i do not understand well but that will come in time and i hope you could clear some of them. 
In return i offer my time to you in return. 

PS. I registered myself on your discord server and i found out that i need to validate myself using a mobile phone (SMS?), sadly i do not have a mobile phone number, i use VOIP to my land-line for my mobile. Could you allow me access to your discord an other way?

luc-github, 2022-03-19:
Yes I have set discord moderation to this level due to spamming, I have lowered now.

nielsnl68, 2022-03-19:
Thanks, i have already started reading that issue it is a lot.

luc-github, 2022-03-20:
As explained here : https://github.com/luc-github/ESP3D/discussions/540#discussioncomment-2329607
the  rewrite is the next version of the draft 3.0
In process of porting ESP3DLib with ESP3D 3.0 I did several update / fixes that will soon be back ported to ESP3D 3.0, one major update is the ESPXXX command output format that have been standardized :
always plain text by default and add `json=yes` or `json` (only ESP400 does not do plain text but it has no meaning in plain text)
this is to have consistent output
when human type command , it get human readable output, for http / telnet / websocket commands to facilitate response handling the output can be json following this format 
```
{
      cmd: <the requested command id>,
      status: <the response status : ok / error>
      data: <the command's response which can be text or json if more than one response, this can also be the error detail if status is error>,
}
```
e.g:
`[ESP410]`
```
echo: Start Scan
echo: luc-ext1	52%	Secure
echo: luc-ext2	44%	Secure
echo: End Scan
ok
```
With json output:
`[ESP410]json`
```
{
      cmd: "410",
      status: "ok",
      data: [{ SSID: "luc-ext1", SIGNAL: "52", IS_PROTECTED: "1" },
             { SSID: "luc-ext2", SIGNAL: "44", IS_PROTECTED: "1" }
            ],
}
```
this will make sanity check easier when handling commands's response by getting original command requester / status / answer in clear output

So it will impact the webui rewrite, as it is  still in heavy development, it is fine but to avoid to have to update the current webui 3.0, and because these changes are currently limited to ESP3DLib V2/3, from now ,the rewrite can only be used with ESP3DLib v2/3 until all code is merged
I do not plan to do an special ESP3D branch, you can do it manually if want to test - ping me if need some help

I will  merge ESP3DLib code to ESP3D 3.0 when ok and WebUI rewrite to 3.0 at once  to avoid too much confusion.

Now all ESPXXX commands output have been updated in ESP3DLib V2/V3, I am now in process to update rewrite according new format, including the simulator - I hope to end it today.

luc-github, 2022-03-24:
WebUI notification using [ESP600] is now implemented
[image]
I need to do https://github.com/luc-github/ESP3D/issues/712 for https://github.com/luc-github/ESP3DLib/tree/2-0 and I will start on Notifications panel

nielsnl68, 2022-03-25:
Hello @luc-github , I have made myself available for some work on the CNC side of the webui-3. For that i have been looked at the webui-3 as it is now. And i love how you have done the main configuration screens (although the tab names need some tweaking). 
With one exception; the buttons at the bottom of those screens, would it be technical possible to show them in the right corner of the header bar. The same also applies for the panel selection button on the dashboard, i love to see that moved inside the header at the right side as well. That way we can always find them. And that is also the way that lot of other web-dasboards work. When this is possible, are you willing to change that?

As last item, I have a hard time believing in the panel based dashboard with the fixed width, i'm still try to think of a better way. but i'm blank, so i will comeback about this later.

luc-github, 2022-03-25:
@I do not understand - if you have screen capture with your modification will be easier to understand

[22 more replies]