luc-github/ESP3D-WEBUI #158: [BUG] SD Upload fails with BTT TFT connected ESP01s

DaStivi, 2021-02-02:
**Describe the bug**
Uploading to (Mainboard) SD fails after a couple of seconds, webui displays error1, BTT TFT starts beeping and display "Error: No Checksum with line number, Last Line: 1"

gco file gets created on sd card but with 0byte

ESP3D-WEBui isn't able to communicate anylonger with TFT/Mainboard, i managed to send M29 once via TFT Terminal that stopped the beeping and constant error poping up on tft but else i've to reset tft to stop and bring back the connection

**To Reproduce**
Steps to reproduce the behavior:
upload gco file to mainboard sd

**Expected behavior**
gco file uploaded successfully to sdcard

**Screenshots**

**ESP3D Firmware:  (please complete the following information)**
- Version: 2.1.1.b10
- Web UI Version: 2.1b68

**Target Firmware:  (please complete the following information)**
 - Name: Marlin
 - Version 2.0.7
 - SKR1.4 (Biqu B1 Printer)
 - BTT TFT35v3 (varriant of it)

**Board used (please complete the following information):**
 - MCU: (BTT) ESP01s 1M 

**Browser used (please complete the following information):**
- Name: [e.g. Chrome]
- Version  [e.g. 2.0]
- Operating system [e.g. Windows 10]

**Additional context**

set #define TX_BUFFER_SIZE 128 and #define RX_BUFFER_SIZE 1024 in marlin firmware, as i tried other solutions from past sd upload issues... 

sending commands, getting temperatures works as expected

luc-github, 2021-02-02:
do you have auto temperature send enabled ?
or any auto command enabled ?
if esp connected to BTT TFT ?

luc-github, 2021-02-02:
to check if any autosend command is enabled just type  in webUI
`M28 test.gco`

if anything appear please share

Once done type `M29` to end upload mode

DaStivi, 2021-02-02:
> do you have auto temperature send enabled ?
> or any auto command enabled ?
> if esp connected to BTT TFT ?

not that i'm aware of.... BUT, as you mention auto command... i do see ah constant M114 query gooing on in esp terminal and even in tft terminal..... but  i don't know from where's this command is comming... at least on esp terminal i don't see the M114 command it self, but on tft terminal i see the M114 command sending...

luc-github, 2021-02-02:
what is  `M115 TFT` output ?

DaStivi, 2021-02-02:
M115 TFT
FIRMWARE_NAME: BIGTREE_TFT35_V3.0_B1.26.x SOURCE_CODE_URL:https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware
Cap:HOTEND_NUM:1
Cap:EXTRUDER_NUM:1
Cap:FAN_NUM:1
Cap:FAN_CTRL_NUM:0
ok

DaStivi, 2021-02-02:
for the sake of completeness, its ah biqu B1 printer with an skr1.4 and ah special version of the btt tft35 ... and interesstingly the ESP port on the tft pcb  has ah wrong pin layout... i did find this also happend on the older versions of btt tft35... and has been fixed with newer versions... interestingly in the b1 facebook group there is also ah user that reported on my post that his board has different print on it.... where mine shows esp01s his has only esp printed and on his tft he could just drop in the esp01s module... i had to pin it differently as the headers are "mirrored"

[image]
[image]
i did soldered the pins now out and put in onthe other side of the esp module to get rid of the cables... 
[image]

why do i mention this? i've read in another sd upload issue, it could also be an voltage supply problem, just not sure if the messed up anything else on the pcb maybe?! 🤔

luc-github, 2021-02-02:
In theory the `M28 test.gcode` should make the tft stop polling which not the case - right ?
Does auto check position  uncheck in webUI still show the M114 feedback - just to be sure poilling come from TFT and not webUI

luc-github, 2021-02-02:
no in this case it is not voltage issue as it is `error 1 ` and check sum error - it is some polling that corrupt communication

DaStivi, 2021-02-02:
> In theory the `M28 test.gcode` should make the tft stop polling which not the case - right ?

i would say so yes!

> Does auto check position uncheck in webUI still show the M114 feedback - just to be sure poilling come from TFT and not webUI

i am pretty sure its not webUI... auto check position is off yes... and even without the esp01 i did noticed this behavior in tft terminal... 

> no in this case it is not voltage issue as it is `error 1 ` and check sum error - it is some polling that corrupt communication
i am not 100% sure now but i think i'd saw, webui does display error 1

luc-github, 2021-02-02:
weird Issue should be in TFT FW  not in Marlin - when I did the [PR to support upload](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/648/files#) I have added the [stop polling feature](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/648/files#diff-181f2ff068fe481967bad8899b7330dba9592f9136fbbbb3b69447ab8c591751R288-R290)
It seems the TFT ignore it for M114 - so something changed in TFT FW after my PR and break the upload feature

luc-github, 2021-02-02:
the poling consist in Temperature / Positions / Feedrate / Flow rate

do only positions answer appear after `M28 test.gco` or all answers  ?

DaStivi, 2021-02-02:
> It seems the TFT ignore it for M114 - so something changed in TFT FW after my PR and break the upload feature

as i'm a noob with github i did checked the files manually and it differs from your code ... :(

luc-github, 2021-02-02:
you cannot do upload when printing - Marlin will not handle that - no idea how he will behave to be honest....

Better wait your print is done to do the upload test I mentioned to be sure you get proper behavior

DaStivi, 2021-02-02:
sending M28 test.gco "freeze" the webui and tft starts putting the error out... and i'm sorry, i got the error on tft wrong... its  "no checksum with line number, last line: 0"
[image]

sending M29 with tft terminal states file saved... 

i tested it now again it behaves a little different, as i have just waited a longer time... and also tried to sent ah M29 with esp3d-webui, tft stilll beeping, but web-ui does react again and shows this now in terminal: 

echo:E0 Flow: 100%
Count E:0
X:-3.0000 Y:-2.0000 Z:0.0000 E:0.0000 Count X:-240 Y:-160 Z:50
FR:100%
echo:E0 Flow: 100%
 T:29.32 /0.0000 B:31.15 /0.0000 @:0 B@:0
Count E:0
X:-3.0000 Y:-2.0000 Z:0.0000 E:0.0000 Count X:-240 Y:-160 Z:50
FR:100%
echo:E0 Flow: 100%
Count E:0
 T:29.32 /0.0000 B:31.04 /0.0000 @:0 B@:0
M28 test.gco
M29
echo:Now fresh file: test.gco
Writing to file: test.gco
//action:notification test.gcook
Error:No Checksum with line number, Last Line: 0
Resend: 1
Error:No Checksum with line number, Last Line: 0
Resend: 1
Error:No Checksum with line number, Last Line: 0
Resend: 1
Error:No Checksum with line number, Last Line: 0
Resend: 1
 T:29.32 /0.0000 B:31.04 /0.0000 @:0 B@:0
Error:No Checksum with line number, Last Line: 0
Resend: 1
Error:No Checksum with line number, Last Line: 0
Resend: 1
 T:29.20 /0.0000 B:31.15 /0.0000 @:0 B@:0

interesstingly there is also a difference in the M114 output before and after trying to save ah file... before there is full x,y,z output... after m28/m29 i only see the temps...

DaStivi, 2021-02-02:
another thing i noticed now... after the steps i can't even read from the tft sd... or sending any other command, it just puts the command in the esp-webui terminal but no answer to that command..

after restarting the esp3d from webui same behaviour... need to reset tft, so i really guess its a tft firmware issue :/

luc-github, 2021-02-02:
>M28 test.gco "freeze" the webui and tft starts putting the error out

It means TFT send commands when it should not I can see it received `T:29.32 /0.0000 B:31.04 /0.0000 @:0 B@:0` so it means `M105` was sent by TFT 
because after `M28 test.gco` polling should stop and any command sent but M29 are supposed to be writen in SD file
also any command sent after `M28 test.gco` should  follow `N<number> <command + parameter>*<Checksum>`
and so any raw commands sent by TFT will generate error 

It is definitly a TFT FW issue

luc-github, 2021-02-02:
to Disable autotemp type `M155 S0` 

I see you did `M29` just after `M28` and it open File after so I guess the `M29`  was ignored need to send only after it said file is open or has error
You can try to enable Marlin DEBUG to echo each command, I am not sure it will echo anything when M28 is used but worth a try:
`M111 S1`
about TFT FW issue may be it is due to : `BTT TFT35v3 (variant of it)` which may affect FW somehow but as `M115 TFT` is working I do not know - need to dig in code and I am busy right now due to personnal training I need to finish quickly -  so need time

DaStivi, 2021-02-02:
> to Disable autotemp type `M155 S0`

i try this later... indeed there is the option to have the "status" (temps) on tft screen all the time... but i don't believe thats the issue... i will raise an issue on btt tft git later too... need some time to recap everything there 😐

thx anyway for your massiv help sofar on this topic!

luc-github, 2021-02-02:
yes it is not issue but clear a little the output to focus on unwished commands 😸 

Thank you please update the status if any - I am curious what did broken the process 😎

emklap, 2021-02-02:
> to check if any autosend command is enabled just type in webUI
> `M28 test.gco`
> 
> if anything appear please share
> 
> Once done type `M29` to end upload mode
Same problem here on a stock BIQU B1 printer,  SKR 1.4 main board, Marlin 2.0.6,  TFT35_B1_V30  0.26.x 
After sending M28 test.gco the empty file is created on the SD card with the following messages in the web UI
========================================
M28 test.gco
echo:Now fresh file: test.gco
Writing to file: test.gco
//action:notification test.gco
ok
Error:No Checksum with line number, Last Line: 0
Resend: 1
Error:No Checksum with line number, Last Line: 0
Resend: 1
========================================

emklap, 2021-02-02:
I did a test by executing the M28 test2.gco command on the TFT of the printer itself, and it gives the same erroneous behavior.
An empty file is created and the printer show the error message "Error:No Checksum with line number, Last Line: 0 Resend: 1"

Could this maybe have a different root cause ?

DaStivi, 2021-02-02:
> I did a test by executing the M28 test2.gco command on the TFT of the printer itself, and it gives the same erroneous behavior.
> An empty file is created and the printer show the error message "Error:No Checksum with line number, Last Line: 0 Resend: 1"
> 
> This looks more like a TFT FW issue to me?

what printer and especially tft  do you have?

emklap, 2021-02-02:
I have a BIQU B1 printer, SKR 1.4 main board, Marlin 2.0.6, TFT35_B1_V30 0.26.x. 
The Marlin and TFT FW is the one that came with the printer.
[image]
[image]

luc-github, 2021-02-02:
yes same behavior - after M28 command nothing should happen as no command should be sent, but obviously Marlin receive several commands

DaStivi, 2021-02-02:
i am pretty sure with using one of lukes ABL Marlin firmware in the past i hadn't theissue of constant commands in the terminal... i've noticed this earlier but didn't had issue sofar... so i was wondering each update... but i'm not sure how this is all related to each other... because another issue with the UBL Firmware is thati don't receive the eeprom save notifications anymore... that was alsoworking with an older ABL version... i think to remember that this also not worked with the stockfirmware that came with the printer (where i think to rember have noticed the constat terminal stream too..)

luc-github, 2021-02-02:
Ok just some feedback 
I have installed Marlin 2.0.7.2 on my SKR 1.3, I have installed latest ESP3D / latest WebUI - today git binary for my BTT TFT 35-E3

I can reproduce issue 
I have used Telnet to be sure no webUI related issue
when I send M28 test.gco
* polling is not stopped => numbering/CRC error pop up
* M29 cannot not be sent to Marlin to stop the writing to SD=> reset TFT allow to send M29 command

definitly a TFT issue as this was not like that 

I found the bug : 
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/TFT/src/User/API/interfaceCmd.c#L208

This block every commands from WebUI but let all commands from TFT when stop polling is enabled which is wrong
```
  if (!ispolling && !fromTFT)
  { //ignore any query from TFT
    purgeLastCmd();
    return;
  }
```

change the condition to correct one:
```if (!ispolling && fromTFT)```
Stop polling and allow upload like it should be

DaStivi, 2021-02-02:
you're a genius... i did read exactly that if loop, but i had no idea how it supposed to be or if this is not correct :(

uploading looks like to work now!

luc-github, 2021-02-02:
great thank you for the confirmation

DaStivi, 2021-02-02:
i need to test tomorrow again, but one odd thing, after starting the print from esp webui (maiboard SD), its starts heating up... then i've aborted the print from the TFT and instead of aborting it starts printing, even nozzle&bed are not heated, also the tft then display busy processing and i can't do anything just reseting the tft to stop this!

output from webui terminal for that job
> ile opened: KFACTOR.GCO Size: 7037
File selected
//action:notification KFACTOR.GCO
ok
//action:resume//action:prompt_end//action:prompt_begin Resuming SD//action:prompt_button Dismiss//action:prompt_show//action:notification KFACTOR.GCO
ok
M876 Responding PROMPT_GCODE_INFO
 T:26.14 /0.0000 B:26.82 /0.0000 @:0 B@:0
 T:26.14 /0.0000 B:26.70 /0.0000 @:0 B@:0
 T:26.14 /0.0000 B:26.70 /0.0000 @:0 B@:0
 T:26.25 /0.0000 B:26.82 /0.0000 @:0 B@:0
 T:26.14 /0.0000 B:26.70 /0.0000 @:0 B@:0
 T:26.14 /0.0000 B:26.70 /0.0000 @:0 B@:0
//action:notification Bed Heating... T:26.14 /245.0000 B:26.70 /70.0000 @:127 B@:127
 T:27.61 /245.0000 B:27.16 /70.0000 @:127 B@:127
 T:32.19 /245.0000 B:27.84 /70.0000 @:127 B@:127
 T:38.50 /245.0000 B:28.86 /70.0000 @:127 B@:127
 T:45.99 /245.0000 B:30.00 /70.0000 @:127 B@:127
 T:53.81 /245.0000 B:30.94 /70.0000 @:127 B@:127
 T:61.67 /245.0000 B:31.98 /70.0000 @:127 B@:127
 T:69.57 /245.0000 B:33.13 /70.0000 @:127 B@:127
 T:77.39 /245.0000 B:34.38 /70.0000 @:127 B@:127
 T:85.03 /245.0000 B:35.25 /70.0000 @:127 B@:127
 T:92.40 /245.0000 B:36.25 /70.0000 @:127 B@:127
 T:99.73 /245.0000 B:37.25 /70.0000 @:127 B@:127
 T:106.79 /245.0000 B:38.17 /70.0000 @:127 B@:127
 T:113.82 /245.0000 B:39.08 /70.0000 @:127 B@:127
 T:120.77 /245.0000 B:40.07 /70.0000 @:127 B@:127
 T:127.46 /245.0000 B:41.03 /70.0000 @:127 B@:127
 T:133.99 /245.0000 B:41.91 /70.0000 @:127 B@:127
M106 P0 S0
 T:140.56 /245.0000 B:42.87 /70.0000 @:127 B@:127
 T:146.85 /245.0000 B:43.75 /70.0000 @:127 B@:127
 T:153.15 /245.0000 B:44.71 /70.0000 @:127 B@:127
 T:159.35 /245.0000 B:45.66 /70.0000 @:127 B@:127

no sign from the abort command, but not even sure i should see it in esp-terminal?

luc-github, 2021-02-03:
when press abort in web UI it send :
[image]
I do not know what TFT send but in all case if printing from Printer SD and already start heating you cannot send any command until heating is done  unless emergency parser is enabled :
https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/Configuration_adv.h#L1940-L1947

luc-github, 2021-02-03:
I guess  issue can be closed now

closed 2021-02-03 by luc-github

jimhiggs, 2021-02-03:
I was struggling the same problem with SD copy from ESP3D to the onboard SD with it failing with the error message: Error:No Checksum with line number, Last Line: 0 on the TFT on all but extremely short .gco files. I am using the latest ESP3D and TFT3.5 V2 2.0.26.x (loaded in December).

I just manually patched Luc's fix of interfaceCmd.c.  I am now getting successful SD copies from ESP3D to SD but the TFT display is still showing up the   Error:No Checksum with line number, Last Line: 0 error shortly after he copy starts and then when the transfer is complete and I go into 'print' to list files on the SD it is stuck on the black loading  screen and I have to reset the TFT to continue. The copied file then shows up on the print files list.

luc-github, 2021-02-04:
I do not reproduce this issue but looks again TFT FW issue not esp3d neither webUI issue

jimhiggs, 2021-02-04:
Sorry @luc-github I was jumping between github issues. I meant to post this in the BTT TFT firmware issues. I agree not a ESP3D or WebUI issue, will repost in the correct place.

DaStivi, 2021-02-04:
> I was struggling the same problem with SD copy from ESP3D to the onboard SD with it failing with the error message: Error:No Checksum with line number, Last Line: 0 on the TFT on all but extremely short .gco files. I am using the latest ESP3D and TFT3.5 V2 2.0.26.x (loaded in December).
> 
> I just manually patched @luc-github fix of interfaceCmd.c. I am now getting successful SD copies from ESP3D to SD but the TFT display is still showing up the Error:No Checksum with line number, Last Line: 0 error shortly after the copy starts and then when the transfer is complete and I go into 'print' to list files on the SD it is stuck on the black loading screen and I have to reset the TFT to continue. The copied file then shows up on the print files list.
> 
> Edit: I just resynced with the master as I see @luc-github fix is included. Prblem still exists.

and you're sure you applied and complied the fix correctly?? and also made shure you see the actual builddate and time then in the Info display??

jimhiggs, 2021-02-05:
First I just changed the code in interfaceCmd.c from if (!ispolling && !fromTFT) to if (!ispolling && fromTFT). Recompiled and copied .bin to my Pro V1. Results were that the copy was now successful but still getting error on TFT. Then I updated the firmware on TFT to the latest version on git. Result is the same.
 I use git and I updated to the latest master of TFT35. I had to include the fonts, icons , language file and config.ini  so I am pretty sure it is the latest.  But I repeated just to be sure. info reads TFT35_V2.0.26.x Feb 4 2021. Yesterday it read TFT35_V2.0.26.x Feb 3 2021 so is the date the date the firmware is updated rather than a version date?

jimhiggs, 2021-02-05:
I just started a copy of a large model file and it is transferring at a rate of 2% every 3 minutes. If that is normal it will only be useful to transfer files this way for fairly small prints

luc-github, 2021-02-05:
https://github.com/luc-github/ESP3D/discussions/535