luc-github/ESP3D-WEBUI #159: [FEATURE REQUEST]Dark Mode

Legsmaniac, 2021-02-11:
**Is your feature request related to a problem? Please describe.**
Printing at night as most of us do, setting prints off overnight, the screen is so bright.

**Describe the solution you'd like**
Can we have a Dark Mode please?

**Describe alternatives you've considered**
Chrome has the ability to add a dark mode but it's not as good as a built-in dark mode.

**Additional context**
This is Chrome's Dark Mode, not perfect, the X's to close windows for example are barely visible.

[image]

Would be cool to see a proper Dark Mode option in the WebUI.  Everyone else are doing it these days.
Thank you.

luc-github, 2021-03-15:
I think this will be handled like language packs - loading an external css file will allow to theme the UI , so I will not affect the final webUI size - but keeping flexibility to change colors and some UI parts - TBC
The css file could be compressed also to save space on flash

luc-github, 2021-05-26:
Theme support using external css is now supported in rewrite
[image]

closed 2021-05-26 by luc-github