luc-github/ESP3D-WEBUI #160: [BUG]Authentication - Cannot log in

Legsmaniac, 2021-02-13:
**Describe the bug**
I set up my ESP3D v3.0 with `#define AUTHENTICATION_FEATURE` uncommented since I wanted password security.  And all I get is the box with the spinning circle, no log in. 

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. I installed a completely fresh instance of ESP3D v3.0 onto a NodeMCU 8266 and erased all flash contents.
2. Log into ESP3D WiFi network and go to 192.168.0.1, sure enough, you're presented with the log in box.
3. Enter User admin and Password admin, you have the files upload page.  Upload the index.html.gz and set up.
4. I set up Client access, Static IP and my router details as normal.  Then Restart ESP.
5. All I get in the spinning circle constantly, no log in box.
6. I also tried again but leaving it as AP access and still no log in.
7. It doesn't matter whether I leave the PW as admin default or set my own PW, it's still the same problem.

**Expected behavior**
I expect to be prompted to log in.

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

**ESP3D Firmware:**
- ESP core version: 3.0
- Version:3.0.0.a90
- Web UI Version:3.0.0.81 
- Wifi mode:Doesn't matter, I tried both STA (Client mode) and left it as AP, still the same problem. 
- Flash method: Arduino Ide
- Any change done in FW - just uncommented `#define AUTHENTICATION_FEATURE
 
 tips: to get ESP3D informations type `[ESP420]plain`    
       to get WebUI version check your menu header   
**_I can't as I cannot log in._** 

**Target Firmware:**
 - Name: Marlin
 - Version Marlin bugfix-2.0.x 
 
 tip: to get firmware info type `M115` in console    

**Board used (please complete the following information):**
 - MCU: 8266
 - Name:NodeMCU12f
 - Flash size: 4M: 2M/2M

**Additional context**
Nothing more to add, hope it can be sorted. Thanks.

Legsmaniac, 2021-02-13:
Sorry for delay.
OK, I reflashed from scratch, it's working now, I get the log in box.
However.....
What are the Password limitations?
I tried setting a PW which had special characters in.  Whilst it accepted it, I couldn't log in again.  I had to reflash from scratch and start again.
Then I tried a plain PW of just alphanumeric content some 12 characters long.  Again it accepted it but again I couldn't log in.  Reflash and start from scratch.
This time, I set it up but left user and pass default admin.  Fine so far.
Then I went into settings and tried setting a Username, leaving the PW as admin.
Once again, I cannot log in.

Seems it won't let me set my own user and password.  Or actually it will, but won't let me log in with them or anything else, including admin/admin.

luc-github, 2021-02-13:
the lenght is 20 after the webUI says cannot

there is not other constraint - I have just changed to `12345678912345678912` and it worked

setting a username ?
this is not possible unless you change it in code itself  but I did not tested that 
only `admin` and `user` are allowed

Legsmaniac, 2021-02-13:
Ah!!  My Bad!
I was going through the setup so fast I wasn't reading it properly.  Instead, I was "assuming".
Assuming it was standard to most everything else I set up and so I thought that the two boxes were for Username and Password.
This time, I looked carefully and discovered it says Admin password and User password.  So I was setting a "Username" in the Admin password field without realising.  No wonder I couldn't log in again!

Would have preferred a Username option but no worries, it's no biggie.
Right, I guess it's working again then, thanks for the quick fix.
Topic ready for closure.  😃

luc-github, 2021-02-13:
define username when you can only have one connection at once, is not really necessary IMHO

Yes I get used people do read things 

Ok close issue then

closed 2021-02-13 by luc-github

Legsmaniac, 2021-02-13:
> define username when you can only have one connection at once, is not really necessary IMHO

I think you miss the point.  It's for security.
All too many IoT things seem to be fixed with "admin" these days which means a hackers job is already half done for them.  They only have the password left to crack.  Having a username is another 50% of the security, something else for hackers to work out.

And before you ask "Who going to want to hack it on my own network at home?!" 
The reason I wanted to use authentication log in is because I have set my ESP3D up to be accessible from outside my network and created an App for my phone for ESP3D so I can quickly log into my printer from away from home to see how my prints are doing and be able to stop it if something has gone wrong.

luc-github, 2021-02-13:
you missed my point ;) I was talking about allow to change in settings 
as wrote you can change them in code - I just did not tested
https://github.com/luc-github/ESP3D/blob/3.0/esp3d/src/modules/authentication/authentication_service.h#L31-L32