luc-github/ESP3D-WEBUI #162: [FEATURE REQUEST]Terminal Panel Input Box Location

Legsmaniac, 2021-02-20:
**Is your feature request related to a problem? Please describe.**
I note that the Input box for the Terminal Panel in version 3 has moved to the top whereas version 2 placed it at the bottom where I was used to it being.  I found this confusing and find it hard to get used to.  The verbose results in the Terminal window scroll downwards and end at the bottom where I expect the Input box to be to continue with commands.

**Describe the solution you'd like**
I'd like to see the Input box at the bottom of the Terminal window in version 3, as it was in version 2.  Please could you consider a) moving it back to the bottom or b) give users the choice by way of a check box in the Interface Settings page.

**Describe alternatives you've considered**
None as I have no idea how to do it myself.

**Additional context**
Add any other context or screenshots about the feature request here.
None necessary, above description should suffice.  So here I am, politely asking if you would kindly consider my request. :-)

luc-github, 2021-02-20:
it has been put in this location because users feedback they do not want to scroll down window to access input box, and first thing they want to access is input not tegminal output 
I am not sure how it could be confusing as all UI is different

Legsmaniac, 2021-02-20:
Er.... that's a bit silly cuz you have to scroll down anyway to read the full results.
I meant confusing when you're used to it being at the bottom using version 2.  
Most things like this I use, you type at the bottom.
Even Windows and Linux terminals as they scroll down, entry is always still at the bottom of the scroll.

I did wonder if it was user-request and I appreciate that.  But then, there must be some users like myself who prefer it at the bottom.  So surely the solution would be to give them the choice?  Just as it's user-choice whether they prefer dark mode or not?

luc-github, 2021-02-20:
well several people want just send command and do not care the output, some app even do not display the result like MKS plugin :
[image]

or Repetier host put input in different place than output:
https://www.repetier.com/documentation/repetier-host/rhmanual-control/

[image]

UI is definitly subjective and it hard to please everyone

Legsmaniac, 2021-02-20:
> and it is hard to please everyone

Not really.  Adding a checkbox to switch to bottom input would then please everyone?
Best of both worlds.
😁

luc-github, 2021-02-20:
well I try to keep WebUI as small as possible - so you understand such things is not really a top priority in current todo list

closed 2021-06-02 by stale[bot]