luc-github/ESP3D-WEBUI #164: WebUI - axis movement by cursor keys

luc-github, 2021-03-06:
From Grbl_Esp32 created by [JensHauser](https://github.com/JensHauser): bdring/Grbl_Esp32#791  

**Please describe the feature you would like implemented**
Up/Down and Left/Right cursor keys for x/y axis would be very helpful as a solution for making movements.

**What do you need the feature for?**
I use a laptop with WebUI to move the x/y axis to the right location. Hitting the circle field and in parallel watching the cnc mill is not that easy :-) Taking cursor keys, e.g. in 10mm steps, would be better.

**Will this feature appear to a lot of users?**
I hope so.

luc-github, 2021-03-06:
@JensHauser issue has been moved to webUI

luc-github, 2021-03-06:
In 3.0 the plan is to use button UI instead of pronterface for Jog: https://github.com/luc-github/ESP3D-WEBUI/issues/144#issuecomment-764845963

Access Key has been discussed here: https://github.com/luc-github/ESP3D-WEBUI/issues/108, but I did not digged in to solve the problem yet

Feel free to join discussion here: https://github.com/luc-github/ESP3D-WEBUI/issues/144

luc-github, 2021-08-08:
about access-key, this  https://developer.mozilla.org/en-US/docs/Web/HTML/Global_attributes/accesskey explain that they behave differently according OS and browser and may need several keys at once - which is not really great, so solution does not fit dirrections keys 
In addition it seems recommendation is actually :
> it is generally advised not to use accesskeys for most general-purpose websites and web apps.

So I think I will use onkeydown  for up/down/left/right arrow on main container  https://www.w3schools.com/jsref/event_onkeydown.asp  when jog panel is visible

luc-github, 2021-11-07:
POC is now implemented in printer using keyboard eventListener - I meet a problem with removeEventListener that seems not working with preact for some reason, so I have add some hack until fix is found
The Printer UI use Pronterface but for keyboard it switch to mobile UI, for not printer UI that do not use pronterface UI (e.g CNC) the UI will stay the same (button style UI)
[image]

closed 2021-11-07 by luc-github

luc-github, 2021-11-08:
Key mapped helper added
[image]

JensHauser, 2021-11-08:
Thank you, Luc. I guess this enhancement will be part of next ESP32/Fluid distro, so I´m curious to make use of it

luc-github, 2021-11-08:
@JensHauser for what I can see fluidNC use Mitch fork with cncjs UI which has already this feature if I am not wrong