luc-github/ESP3D-WEBUI #170: [FEATURE REQUEST] Firmware upload

Lino77, 2021-03-24:
Is it possible to include firmware upload on the SD card. gcode on SD printer card works fine, but my Firmware.bin is unfortunately changed to gco

luc-github, 2021-03-24:
what is your configuration ?
what firmware version are you using ?

Lino77, 2021-03-26:
I am using a current Marlin on an MKS Nano V3 with ESP3.0 Build a93 Interface build 82, and would like to load the Marlin firmware onto the SD card via webupload

luc-github, 2021-03-26:
so you enabled mks protocol in ESP3D?
if yes the upload of any file should work, the file name is the name used

Lino77, 2021-03-26:
this is my config an it dosnt work 

About ESP3D
Firmware: 3.0.0.a93  
Interface: 3.0.0.82  
Browser version: Chrome 89
Chip ID: 54341
CPU Frequency: 160 Mhz
Free memory: 22.69 KB
SDK: 2.2.2-dev(38a443e)
Flash Size: 4.00 MB
Available Size for update: 1023.00 KB
Filesystem type: LittleFS
Filesystem usage: 120.00 KB/1.98 MB
Baud rate: 115200
Sleep mode: None
WiFi: Enabled
Hostname: esp3d
HTTP port: 80
Telnet port: 23
WebDav port: 8181
Client mode: Enabled
Mac address: 40:F5:20:1D:D4:45
SSID: ########
Signal: 100 %
Phy Mode: 11n
Channel: 6
IP Mode: DHCP
IP: 192.168.178.89
Gateway: 192.168.178.1
Mask: 255.255.255.0
DNS: 192.168.178.1
Access point mode: Disabled
Mac address: 42:F5:20:1D:D4:45
Serial: Enabled
Notifications: Disabled
Serial: MKS
Target firmware: Marlin
FW architecture: ESP82XX

luc-github, 2021-03-26:
if file is renamed it is renamed by MKS Marlin code

Lino77, 2021-03-26:
okay thought through ESp because it turns ROBIN_NANO_V3.BIN into ROBin ~ .GCO.
but I build everything again and bottle everything again and then test it again

luc-github, 2021-03-26:
yes that is done by marlin mks code, on printer fw, you can submit issue on mks repository but Marlin SD library only support dos name  8 chars long

luc-github, 2021-03-26:
I think one solution would be that bootloader accept firmware.bin as file name, and mks do not apply any rename if filename is firmware.bin

luc-github, 2021-03-27:
I close issue as not on esp3d scope, rename happen  in Marlin code

closed 2021-03-27 by luc-github