luc-github/ESP3D-WEBUI #176: [BUG]GRBL_ESP32 with ESP3DUI on top, sdcard no connection through WEBUI

kai1-AAA, 2021-05-09:
Compiled and flashed GRBL_ESP32 per bdring instruction (SDCARD enabled in config.h)

Steps to reproduce the behavior:
1. Get latest GRBL_ESP32 package flash it with Arduino IDE to ESP32 DEV module(esp32 1.0.6).
2. Install UI .gz from GRBL_ESP32 directory package (\data), tried several package fro ESP3D, ESP3D-WEBUI, all th same.
3. Telnet and command line are response to [esp200] and $sd/status, show status, list the card and so on.
4. But in web UI SDcard section - refresh or upload buttons return error message - No connection.

**Expected behavior**
SD section should show and manage card contents and allow to run g
[image]
codes.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**ESP3D Firmware:  (please complete the following information)**

WEB UI: 2.1b68 
GRBL_ESP32 FW: 1.3a (20210424)
Wifi mode: STA

Board used (please complete the following information):**
 - MCU: ESP32WROOM
 - Name: NodeMCU
 - Flash size: 4M: 1.9M OTA/0.19M spiffs]

**Browser used (please complete the following information):**
- Name: Chrome
- Version   90.0.4430.93
- Operating system  Windows 10
[image]

luc-github, 2021-05-09:
which branch of grbl_esp32 are you using? dev or main ?

luc-github, 2021-05-09:
I do not duplicate flashing platformIO

[image]

I see you use esp32 1.0.6, GRBL_ESP32 recommand to use 1.0.4 the later release having some issues:
https://github.com/bdring/Grbl_Esp32/issues/884

Please try using esp32 1.0.4 or platformIO - it works on my side - the webUI did not changed since 6 months so I doubt the issue  is in it.

kai1-AAA, 2021-05-09:
> which branch of grbl_esp32 are you using? dev or main ?

i used main branch, will try version 1.0.4 , look if this helps.

luc-github, 2021-05-09:
my screenshot is done using today downloaded main branch

kai1-AAA, 2021-05-10:
checked its weird) but everything works now, not all files loading properly but mostly its working ok, thanks for help!

kai1-AAA, 2021-05-10:
use 1.0.4 esp32 core for GRBL_esp32

closed 2021-05-10 by kai1-AAA