luc-github/ESP3D-WEBUI #180: [BUG] OK P63 in SD file list

badbod, 2021-06-20:
**Describe the bug**
When viewing the SD files there is always a OK P63 entry in the list

**To Reproduce**
connect ESP01S to TFT35 E3 V3
Insert SD card in controller board 
Click SD button in WEB-UI

**Expected behavior**
OK response should be parsed out.

**Screenshots**
[image]

**ESP3D Firmware:  (please complete the following information)**
- Version: 2.1.1
- Web UI Version:2.1
- Wifi mode: Client Station
- Any change done in FW: Attach config.h? No other changes.

**Target Firmware: 
 - Name: Marlin
 - Version 2.0.9

**Board used (please complete the following information):**
 - MCU:  ARM Cortex-M3 LPC1769
 - Name: BTT_SKR_E3_TURBO
 - Flash size: 1M
 

**Browser used (please complete the following information):**
- Name: Opera 
- Version  LVL2 (core: 76.0.4017.205)
- Operating system: Windows 10

**Additional context**
Add any other context about the problem here.
Developer tools output, error message, etc...

luc-github, 2021-06-20:
what version of webui are you uising ? 2.1 is just the base, open the menu it is displayed on top of it

luc-github, 2021-06-22:
@badbod any update ?

badbod, 2021-06-27:
sorry for delay, here it is
UI: 2.1b70 / FW: 2.1.1.b8

luc-github, 2021-06-28:
I can work around it, this is why I needed the output - so it won't display it - but such output may disturb the file upload if you use it 

Please, give a try to UI: 2.1b71  https://github.com/luc-github/ESP3D-WEBUI/blob/2.1/index.html.gz

badbod, 2021-07-03:
Works perfectly, thank you.

luc-github, 2021-07-03:
great thank you, I close issue then

closed 2021-07-03 by luc-github