luc-github/ESP3D-WEBUI #184: Machine Cordinates are Not Updating in my UI

jjhamb, 2021-08-02:
I Have managed to Flash ESP3D on ESP01, Index,html.tz has peen uploaded. I can see GRBL parameters and output on console. I am able to JOG and Home the Machine.
However I am unable to see Machine Coordinates getting any info. I have tried setting the $10=2 and 3 without any Joy.
Additionally I don't see any user set up. 
Please note that File System on ESP index.html.gz, favicon.ico, 404.html file ....  Kindly advice what needs to change.
My Machine is otherwise working fine thru UGS. 
RX TX lines using ESP01 are working too as I can see and change machine parameters.

luc-github, 2021-08-02:
do you see the machine coordinate in terminal in `?` feedback ?
Can you screenshot the page ?

jjhamb, 2021-08-02:
Hi LUC, thanks for quick reply. Here are the screenshots.
[image]
[image]
[image]
[image]

luc-github, 2021-08-02:
I only see MPos coordinate, no WPos in GRBL polling is that normal ?

jjhamb, 2021-08-02:
I am not sure how it should look...  but position updates fine if I use Universal Gcode Sender.

luc-github, 2021-08-02:
This is what I have on my GRBL_ESP32
[image]

jjhamb, 2021-08-02:
I am hoping that I too can get the same output as you. But I do not see any Idle messages on ESP3D UI. 
I was hoping that someone here may be able to help. I do not even see Machine Coordinates
[image]
.

jjhamb, 2021-08-02:
IF I connect a Bluetooth module where ESP is connected, I can connect my UGS and can see data change as the machine moves/jogs.

luc-github, 2021-08-02:
Please share several `?` output as if you do not have any WCO coordinate it should display at least the Mpos ones

jjhamb, 2021-08-02:
$
[HLP:$$ $# $G $I $N $x=val $Nx=line $J=line $SLP $C $X $H ~ ! ? ctrl-x]
ok

$#
[G54:-249.000,-249.000,0.000]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G28:-264.000,-289.000,0.000]
[G30:-264.000,-289.000,0.000]
[G92:0.000,0.000,0.000]
[TLO:0.000]
[PRB:0.000,0.000,0.000:0]
ok

$G
[GC:G0 G54 G17 G21 G90 G94 M3 M9 T0 F0 S0]
ok

$I
[VER:1.1h.20190825:]
[OPT:VC,15,128]
ok
$N
$N0=
$N1=
ok

$C
ERROR:8
OK
error:8
ok

$
[HLP:$$ $# $G $I $N $x=val $Nx=line $J=line $SLP $C $X $H ~ ! ? ctrl-x]
ok
~
ok
!
ok
?
<Alarm|MPos:0.000,0.000,0.000|Bf:15,126|FS:0,0|WCO:-249.000,-249.000,0.000>
ok

---- USB Serial is not connected.

jjhamb, 2021-08-02:
$$
$0=10
$1=25
$2=0
$3=3
$4=0
$5=0
$6=0
$10=3
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=1
$23=3
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=80.000
$101=80.000
$102=250.000
$110=1000.000
$111=1000.000
$112=200.000
$120=500.000
$121=50.000
$122=10.000
$130=250.000
$131=250.000
$132=200.000

luc-github, 2021-08-02:
I see 
`<Alarm|MPos:0.000,0.000,0.000|Bf:15,126|FS:0,0|WCO:-249.000,-249.000,0.000>` so I need to check why webUI do not update the coordinates as it see them in terminal ,
May I know which version of webUI you use ? it display in menu on right when you click on it

jjhamb, 2021-08-02:
UI: 2.1b57 / FW: 2.1.1.b8

jjhamb, 2021-08-02:
ok

luc-github, 2021-08-02:
Ok I have reproduced your issue - it looks a bug in webUI - I think I have found where - I am testing the fix

jjhamb, 2021-08-02:
I tried the latest version...  No change is seen
[image]

luc-github, 2021-08-02:
Yes hold a sec -  checking the fix

jjhamb, 2021-08-02:
> Yes hold a sec - checking the fix

Thanks...  eagerly waiting.

luc-github, 2021-08-02:
Please give a try to this and let me know - version is a72

jjhamb, 2021-08-02:
Hi...  Installed A72, Power cycled the machine. I am getting Machine position now after homing command but nothing changes on Jogs.
[image]
[image]

luc-github, 2021-08-02:
click the auto check

jjhamb, 2021-08-02:
Auto check is at 3s

jjhamb, 2021-08-02:
I also tried changing $10 parameter on grbl by setting it to 0, 1, 2, 3 .. no change.

jjhamb, 2021-08-02:
BTW I do not see <Idle|........> messages in my console even with verbose on.

luc-github, 2021-08-02:
when you check the auto check every 3s
what is the output in terminal after 20s ?

jjhamb, 2021-08-02:
Hi , Sorry for the delay
I get attached output 80-90s apart...   "ok" or just "k"
Screenshot of about 10min Idle time is attached
[image]

jjhamb, 2021-08-03:
Hi again....
I have checked my controller, It is sending out Work Position. This output was received in UGS by connecting Bluetooth module where ESP is connected.

 

//
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0|WCO:-229.000,-229.000,0.000>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0|Ov:100,100,100|A:S>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0|WCO:-229.000,-229.000,0.000>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0|Ov:100,100,100|A:S>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0|WCO:-229.000,-229.000,0.000>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0|Ov:100,100,100|A:S>
//

Idle statements may not be getting properly parsed in ESP3d
Further note that Selecting or Deselecting Verbose in Web Interface makes no difference at my end. I also do not see <Idle|...> statements on port 8888 to which I connect via RAW mode of putty.
  My GRBL Version is 
//
$I
[VER:1.1h.20190825:]
[OPT:VC,15,128]
ok
//

[image]

luc-github, 2021-08-03:
I do not understand 
when ESP board is connected you cannot connect Bluetooth neither USB
if your ESP board do not get anything it means wiring issue

jjhamb, 2021-08-03:
I removed ESP Board to connect with UGS using Bluetooth module to see if Work Position is being reported and it is. 
Only one board is connected at a time to rx/TX ports...  USB is not in use.

luc-github, 2021-08-03:
GRBL is working I have no doubt as you can use by USB 
I try to understand your issue but I get lost with your report talking about other Apps
I am not at your side so please understand that I need clear feedback, as I do not see what you are doing to help you
1 - please do a screen of all webUI when ESP3D board is connected and poilling - not partial screenshot
2 - you mentioned you used putty and it was working well- was that with ESP3D telnet feature ?
3 - what is the output in webUI when you enter `?` then `$` in webui terminal - I need screenshot of all webUI also

jjhamb, 2021-08-03:
I apologise for the confusion....  
1. I will send you the screenshots.  - Reverting
2. Putty connects to Port 8888 but I get same output as what I get on WEB-UI Console. I did not use telnet but RAW mode instead.
3.  $ & ? are as follows:
$
[HLP:$$ $# $G $I $N $x=val $Nx=line $J=line $SLP $C $X $H ~ ! ? ctrl-x]
ok
~
ok
!
ok
?
<Alarm|MPos:0.000,0.000,0.000|Bf:15,126|FS:0,0|WCO:-249.000,-249.000,0.000>
ok

4. Positions are displayed only in UGS but not in WEB-UI
sending screen shots shortly

jjhamb, 2021-08-03:
Attaching Screenshots with ESP connected
Terminal Screenshot is also attached (using telnet)
Screenshots after that are from WEBUI

After homing Machine reports <Home|MPos:-249.000,-249.000,0.000|FS:0,0|Ov:100,100,100|A:S>
This updates Machine position correctly.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

luc-github, 2021-08-03:
Ok so $$ give feedback and I see some feedback on terminal - I have a guess that some answers do not have \r nor \n 
like `<State:xxxx>`
Can you tell me what version of GRBL you are using ?

jjhamb, 2021-08-03:
$I
[VER:1.1h.20190825:]
[OPT:VC,15,128]
ok

luc-github, 2021-08-03:
You typed in webUI terminal, right ?

jjhamb, 2021-08-03:
Yes on the WEB UI

luc-github, 2021-08-03:
in same time
please flash esp3d using 160Mz for CPU frequency not 80Mz - this is why it may loose some data

luc-github, 2021-08-03:
Seems your FW use the `static void report_util_line_feed() { printPgmString(PSTR("\r\n")); }`
and 
```
serial_write('>');
report_util_line_feed();
```
so the issue is not the sending end line but data corruption we can see with ok 

May I know how do you build flash ESP3D ? arduino IDE? PlatformIO ?

jjhamb, 2021-08-03:
> May I know how do you build flash ESP3D ? arduino IDE? PlatformIO ?

I used Arduino IDE

jjhamb, 2021-08-03:
> in same time
> please flash esp3d using 160Mz for CPU frequency not 80Mz - this is why it may loose some data
> 
> I am checking to see if need to add `>` as line end in the realease of GRBL you are using

I will try to flash using 160Mhz

jjhamb, 2021-08-03:
> in same time
> please flash esp3d using 160Mz for CPU frequency not 80Mz - this is why it may loose some data

Tried flashing with 160Mhz... It did not work, ESP kept resetting..

luc-github, 2021-08-03:
Do a 160Mhz should not make reset 

what esp8266 core version are you using ?
Do you use the libraries present in repository libraries directory or use another version?
What settings do you put in Arduino IDE to flash ?

jjhamb, 2021-08-03:
Executable segment sizes:
IROM   : 459992          - code in flash         (default or ICACHE_FLASH_ATTR) 
IRAM   : 28576   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 
DATA   : 1340  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 5288  ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 27104 )         - zeroed variables      (global, static) in RAM/HEAP 
esptool.py v2.8
Serial port COM13
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 2c:f4:32:44:ec:ef
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 1MB
Compressed 499344 bytes to 356457...
Wrote 499344 bytes (356457 compressed) at 0x00000000 in 31.6 seconds (effective 126.5 kbit/s)...
Hash of data verified.
--- Above is the Output from Arduino IDE  and settings for the board are in the screenshot....   
After flashing at 160Mhz ESP repeatedly Resets

[image]

jjhamb, 2021-08-03:
I have not installed any additional libraries... Board Manager in use is 2.7.4  ...
If I use latest version 3.0.2 I get error at #include "WiFiClientSecureAxTLS.h" in line 38 of notifications_service.cpp

luc-github, 2021-08-03:
3.0 is not supported - they break many thing so must use 2.X as stated in read me
[image]

2.7.4 should be ok 

ESP3D need libraries as stated in readme :
[image]
The one in red is the key one
[image]

Your FS is too big : use 1M (SPIFFS 256KB) not more

jjhamb, 2021-08-03:
All above noted....  Will add these libraries manually now and change the SF size. 
Will get back shortly..

jjhamb, 2021-08-03:
I have re-flashed ESP after copying libraries to Arduino/Libraries and SPIFFS 256KB
there is no change as far as Work position is concerned. Screenshot is attached
[image]

luc-github, 2021-08-03:
can you enter the `?` command in webui terminal ? is there any response from GRBL visible in webUI

jjhamb, 2021-08-03:
?
<Idle|MPos:-249.000,-249.000,0.000|Bf:15,126|FS:0,0>
ok
ok
$J=G91 G21 F1000 X10
$J=G91 G21 F1000 Y10
?
<Idle|MPos:-239.000,-239.000,0.000|Bf:15,126|FS:0,0>
ok
ok

jjhamb, 2021-08-03:
Response is coming...  First ? is after homing and second one after Jogging 10mm on X and Y axis...  but Work position is not updating.

luc-github, 2021-08-03:
yes as mentioned there is not WCO in status - until there are no WCO ins status Work position are not updated
It seems command from terminal show the status, but not when send from poilling command, I am checking

jjhamb, 2021-08-03:
Below is verbose output when UGS was being used [I had put this up earlier]
Idle  outputs are coming in batches of 10, 8 lines are same and two lines have additional data

---UGS Console in Verbose mode without ESP Connected --//
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0|WCO:-229.000,-229.000,0.000>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0|Ov:100,100,100|A:S>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0|WCO:-229.000,-229.000,0.000>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0|Ov:100,100,100|A:S>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0|WCO:-229.000,-229.000,0.000>
[verbose] <Idle|MPos:-149.000,-249.000,0.000|FS:0,0|Ov:100,100,100|A:S>
//

luc-github, 2021-08-03:
yes WCO come time to time 
I think if you do several `?` in terminal it will show up
what is stumble me is that `?`refresh seems not working on you system for some reason, the ok behavior is also very weird

jjhamb, 2021-08-03:
You are right sending Repeated Manual ?;s did update my work position.
But that is not ideal I think.  

[image]

luc-github, 2021-08-03:
this is how grbl works, that is why there is automatic send of `?` once WCO is get - the calculation of working can be done for each new position

luc-github, 2021-08-03:
what is your Browser Name ? I do not recognize it

jjhamb, 2021-08-03:
I am using chrome....  some screenshots are from Mac others from windows. Last one is from my phone.
I am new to GRBL and CNC but what I see on Serial console using UGS with verbose setting on...  grbl reports Idle status very frequently. Perhaps a check filter is needed to trap messages that have WCO in them and ignore others.

luc-github, 2021-08-03:
Do you know how to open the developer tools of your browser and check the console output ?

jjhamb, 2021-08-03:
Im on developer console now

jjhamb, 2021-08-03:
[image]

luc-github, 2021-08-03:
there is definlty something wrong this what I have
[image]

[20 more replies]

closed 2021-08-07 by luc-github