luc-github/ESP3D-WEBUI #186: [FEATURE REQUEST] Add the option to upload gcode to an external SD card

magnificu, 2021-08-24:
It would be great if you can add the option to upload the gcode files to an external SD card that is connected to the main board.
From the printer LCD I can select and print from the SD card, but not from the ESP3D WEBUI.

luc-github, 2021-08-24:
which version of ESP3D /WebUI are you using ?
which main board is connected the SD ? Printer ? TFT? ESP?

magnificu, 2021-08-24:
I am using the latest stable 2.x ESP3D version.
Board: BTT SKR2 (Marlin v2.9.1)
TFT: BTT TFT50 (BIGTREETECH-TouchScreenFirmware v0.27)
ESP: BTT ESP-07S (ESP3D v2.1.1)
3D Printer: Self made (Hypercube)

After refresh of the SD Files in ESP3D WebUI I see: echo: Unknown command: NaN GB
If I attache the SD cart to the board there is no difference.

luc-github, 2021-08-24:
Sorry I still do not understand:

> If I attache the SD cart to the board there is no difference.

Where is the SDCard reader connected ?
Where is the ESP board connected ?

magnificu, 2021-08-24:
The SDCard reader is connected to the TFT
The ESP board is connected to the SKR2

Here you can find the user manuals in case they are needed.
https://github.com/bigtreetech/SKR-2/tree/master/Hardware
https://github.com/bigtreetech/BIGTREETECH-TouchScreenHardware/tree/master/BTT%20TFT50%20V3.0/Hardware

luc-github, 2021-08-24:
So there is not way the ESP can get the content of TFT SDCARD, 
ESP3D can query TFT if it is connected to it not in remote, because even it can send M118 P0 M20 SD:,  Marlin will not route the answer of TFT to ESPboard - the answer will only go to Marlin which do not know what to do with this output 

if you what to control your TFT SD card the ESP board must be connected to it not to printer board

magnificu, 2021-08-24:
Unfortunately I cannot connect the ESP to the TFT.
Ok. Thank you for your answer.

luc-github, 2021-08-24:
In that case you must connect an SDCARD reader to the printer board and enable support in Marlin - like this it will work also

closed 2021-08-24 by luc-github