luc-github/ESP3D-WEBUI #187: [BUG]

zanglerecke, 2021-09-05:
**Describe the bug**
The UI is not getting started. 

**To Reproduce**
Steps to reproduce the behavior:
1. Fresh flashed esp8266ex 
2. uploaded the .gz from ./dist/GRBL/production
3. click on esp3d
4. just a spinning circle

5. goto <IP>/?forcefallback=yes
6. remove .gz
7. uploaded the .gz from ./dist/printer/production
8. click on esp3d 
9. working

**Expected behavior**
As i wanna control my laser (running on grbl) with it, i used the grbl version (but i do not know the differences (of the UI).

**ESP3D Firmware:  **
- Version:[ 3.0]
- Web UI Version:[ 3.0]
- Wifi mode:[STA & AP]
- Any change done in FW: yes: enabled the fixed Station & Password

**Target Firmware:  (please complete the following information)**
 - Name: [grbl)
 - Version [1.1]

**Board used (please complete the following information):**
 - MCU: [ESP8266] 
 - Name:[esp-12f] 
 - Flash size: [4M]

**Browser used (please complete the following information):**
- Name: [firefox, chrome, ...]
- Version  [different ]
- Operating system [android, debian, mint]

luc-github, 2021-09-05:
GRBL ui is not yet done in 3.0

zanglerecke, 2021-09-05:
oh.. i did not see that ;) sry for opening a bug...

luc-github, 2021-09-05:
You can use ESP3D/Webui 2.1 until it is implemented

zanglerecke, 2021-09-05:
yes, but in version 2.1 there is no implementation of telnet? or did i miss something? 

i use laserweb as desktop application, and want to just use the esp as an adapter with benefits ;)

luc-github, 2021-09-05:
yes 2.1 has telnet same as 3.0

zanglerecke, 2021-09-05:
.. then i did some other thing wrong.

i just read the oder threads. maybe there are some informations (or other things i can use/reply ;) ... thanks for your work.

luc-github, 2021-09-05:
in 2.1 telnet port is 8888, called data port

luc-github, 2021-09-05:
I guess issue can be closed then

closed 2021-09-05 by luc-github

zanglerecke, 2021-09-05:
yes, i agree totaly