luc-github/ESP3D-WEBUI #192: [BUG] GRBL V3 does not start

Luterocoruna, 2021-10-02:
**Describe the bug**
When you load GRBL webUI it does not start

**To Reproduce**
load index.html.gz created for GRBL and webui does Not start. the start time circle keeps spinning

**Expected behavior**
webui starts

**Screenshots**
If applicable, add screenshots to help explain your problem.

**ESP3D Firmware:  (please complete the following information)**
- Version:[3]
- Web UI Version:[3]
- Wifi mode:[e.g. STA]
- Any change done in FW --> no changes

**Target Firmware:  (please complete the following information)**
 - Name: [GRBL]
 - Version [3.0]

**Board used (please complete the following information):**
 - MCU: [ESP8266] 
 - Name:[wemos mini v3] 
 - Flash size: [4M: 2M/2M]

**Browser used (please complete the following information):**
- Name: [Chrome]
- Version  [94]
- Operating system [Windows 10]

**Additional context**
in the file
ESP3D-WEBUI/src/components/settings/webui.js in line 48 --> you use initmachine
that function does not exist
you call it in line 199 and the webui does not start
I think the one that you should call is 
initDefaultMachineValues
so in line 48 you substitute initmachine for initDefaultMachineValues
and in line 199 you substitute initmachine for initDefaultMachineValues
After those changes the webui for grbl starts
then there are other problems but this one is fixed, or so i think

luc-github, 2021-10-02:
the webui 3.0 has no real grbl support yet, only Marlin was done as POC with some part for GRBL but not tested
This version of 3,0 was my first project using preact and so it is full of glitches, the rewrite is ongoing but gui is not yet defined https://github.com/luc-github/ESP3D-WEBUI/issues/144
so no webui 3.0 for grbl yet sorry, pls be patient it will be out soon

Luterocoruna, 2021-10-02:
Thank you so much for your great work.

luc-github, 2021-10-03:
I close issue then, thank you

closed 2021-10-03 by luc-github