luc-github/ESP3D-WEBUI #200: Cannot Access IP Address

OwenD30, 2021-11-05:
I connected the module to my home wifi and found its IP address (192.168.1.8) in my router, but I get a timeout error when I try to connect to it on all my devices. It's connected but I can't access it.

luc-github, 2021-11-05:
please fill all informations requested in template https://github.com/luc-github/ESP3D-WEBUI/blob/2.1/.github/ISSUE_TEMPLATE/bug_report.md

OwenD30, 2021-11-05:
I just was thinking about this issue and I'm quite sure it's something with my router. No one else has had this issue and if I can see it on my router device manager it must be a problem with my network, not your firmware. Also, I am using [Knutwurst's firmware](https://github.com/knutwurst/ESP3D/releases) on the ESP-01 so there are just too many variables and I don't want you to waste your time trying to fix a problem that no one else has. You can just resolve the issue, I'll figure it out on my own. 

Great job with the project btw

luc-github, 2021-11-05:
as I see he only provide binary so you do not know the settings used 
it can also be an issue with webui upload, try 
`http://yourip?forcefallback=yes` to access emdedded page

closed 2021-11-05 by luc-github

OwenD30, 2021-11-05:
It ended up being a network error, I fixed it and everything is working properly now. Thanks so much for the awesome features this thing has!