luc-github/ESP3D-WEBUI #208: detecting completion of http request

tmartin007, 2021-11-23:
When issuing a command via WEBUI's terminal (e.g. G28), a response similar to the following is received.

```
G28
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
X:5.00 Y:5.00 Z:5.00 A:29.00 Count X:25600 Y:25600 Z:25600 A:1060572
ok
```

The trailing ```ok``` indicates that Homing has completed.

When issuing (e.g.) G28 via a http request (i.e.: ```http:\\192.xxx.xxx.xxx/command?commantText=**G28**```), responses are very rarely received, which makes it difficult in some cases to issue commands in a controlled sequence. - I have tried various REST clients, web-browsers, curl and NSURLSession (in an iOS app) without success.

Could you provide any hint, how a http request should be assembled to cause a response similar to WEBUI's terminal?

BTW: ```http:\\192.xxx.xxx.xxx/command?commantText=**M114**``` beautifully responses as expected, e.g.:

```
X:5.00 Y:5.00 Z:5.00 A:29.00 Count X:25600 Y:25600 Z:25600 A:1060572
ok
``` 

Thank you very much in advance.

Best regards.

luc-github, 2021-11-23:
the GCODE answers from commands do not go to HTTP but websocket , only [ESPXXX] answers go by http

an HTTP answer need a start and an end - but when sending GCODE to printer ESP3D has no idea when the printer will start answer and when the answer is finished, so all answers from serial go to websocket which is a kind of open pipe, and it  is webUI who do the sorting 
it is what is is displayed in webUI terminal

tmartin007, 2021-11-23:
Hi,

thanks for the hint.

I can indeed receive the GCODE answers with WebSockets now.

Thanks again and best regards.

> On 23. Nov 2021, at 03:33, Luc ***@***.***> wrote:
> 
> the GCODE answers from commands do not go to HTTP but websocket , only [ESPXXX] answers go by http
> 
> an HTTP answer need a start and an end - but when sending GCODE to printer ESP3D has no idea when the printer will start answer and when the answer is finished, so all answers from serial go to websocket which is a kind of open pipe, and it is webUI who do the sorting
> it is what is is displayed in webUI terminal
> 
>

closed 2021-11-24 by luc-github