luc-github/ESP3D-WEBUI #213: How to get the Camera of ESP32 Cam working?

Erik262, 2021-12-15:
I've got the ESP32-Cam flashed with ESP 3.0 and WebUI 3.0 and didn't get the settings like [here](https://github.com/luc-github/ESP3D-WEBUI/issues/123#issuecomment-718693756) in the pictures. What has to be done to get the camera feed from the cam module on the ESP32 itself?

luc-github, 2021-12-16:
did you configured the camera in configuration.h?

Erik262, 2021-12-16:
Well, this wasn't mentioned anywhere 😅. I reflashed again and now it works. Shouldn't be the cam (or at least the thinker one) be the default?

Another question:
Is it possible to get the cam running on firmware 3.0, but with WEBUI 2.1? Since 3.0 is way to buggy (I know alpha status)

luc-github, 2021-12-16:
no they are not compatible, the webui 3.0 is under rewrite and rewrite is more stable but it is not finished yet