luc-github/ESP3D-WEBUI #217: Telnet question

JensInternal, 2021-12-29:
My setup is FluidNC3.2.8 with current WebUI.

I have another ESP32, connected via WiFi/Telnet to FluidNC, it´s programmed via Windows/PlatformIO/Arduino platform via serial. In parallel, the WebUI of FluidNC is open.

The issue is unreliable connections after reset. Once it´s established, it works perfect.

My simple question is, how can I establish a -reliable- socket connection to port 53 of the FluidNC WbUI (esp. by often interrupting/re-uploading a program)?

- Connect to WiFi with Wifi.connect() - ok
- Connect via WiFi/client.connect() - sometimes ok
- Many times I don't get a Grbl welcome message, even if isConnected()==true.
- Pressing "Reset" in the WebUI is sometimes of help, but not reliable
- Connecting via Putty/raw mode leads to rejection of a connection. Feels like Putty is rejected in comparison to my code below?

My program runs in a loop, I do CTRl-C and client.stop() after X sec. of inactivity from serial control console.

To better understand my potential mistake, I wrote a test program with many log messages (-> Serial.print).
Hope this gives an indicator how to establish a reliable connection better - thanks for any suggestions.

```
bool checkSocket(bool keepConnectionOpen)
{
  int ret = c.connect(WLAN_HOST, WLAN_TELNETPORT);
  log("CheckSocket. Client connected=", true, false);
  log(ret, false, true);

  if (c.connected())
  {
    log("CheckSocket. Connected to server port", true, true);

    c.println();
    log("CheckSocket. Sent \"CR\", expect reply to determine keep alive", true, true);
    delay(1000);

    if (c.available())
    {
      log("CheckSocket. Response received=", true, false);
      Serial.print("\"");
      while (c.available())
      {
        char ch = c.read();
        if ((ch != 13) && (ch != 10))
          Serial.print(ch);
      }
      Serial.println("\"");
      if (!keepConnectionOpen)
        c.stop();
      return true;
    }
    else
    {
      log("CheckSocket. Connected, but server not responding.", true, true);
      return false;
    }
  }
  else
  {
    log("CheckSocket. Server not connected.", true, true);
    return false;
  }

  log("CheckSocket. No connection possible.", true, true);
  return false;
}
```

luc-github, 2021-12-29:
Using putty should work - unless FluidNC does not give answer because FW is busy, ESP has limited resource do you use WebUI in  same time as telnet ? or serial in same time as telnet ?

JensInternal, 2021-12-29:
Thanks, Luc, for your advice..
I set up a test case:
- Switched on FluidNC, Homed XYZ, idle mode, no other external connections
- 5xPutty connect, disconnect. Works perfectly
- Started visual studio code, connected ESP32/USB, started serial monitor with the program above. 5x no connection
- From this point of time, Putty doesn't connect anymore

So, in my thinking, something within VSC/PlatformIO/Serial monitor/my programm seems to be an issue.

luc-github, 2021-12-29:
I think if you do permanent pollling on serial, the telnet has no time to respond on time because FW is busy with serial.
remember ESP board is little chip with good capabilities but not supporting really multithreading like a PC

JensInternal, 2021-12-29:
Ok, test case adjusted to avoid interference with

- WebUI
- serial connection
- VSC/PlatformIO/Serial monitor

- Any open connections are closed at the beginning
- WiFi is close at the beginning, then opened again
- Connection is closed after successful interaction with FluidNC
- My program now switches on the internal LED, if a connection is established.
- Switched on FluidNC, Homed XYZ, idle mode, no other external connections
- Connected ESP32/USB
- Once I hit the reset button in WebUI, I get a connect immediately (LED on)
- Pressing reset button delivers repeatable results

Later, I switched serial on again, proper results.

So, in my understanding I didn't close the client connection before reset when leaving serial monitoring with CTRL-C, the program was still running with an open connection to FluidNC. Seems any timeout or open connections survived the reset.

Let´s see :-)

JensInternal, 2022-01-01:
Issue probably found.

Adding a client.stop() after doing Grbl processing and before the while loop looks good. The while loop is left suddenly and the ESP32 is blinking. The argument with VSC/serial monitoring was not the right way, although it looked like this at first.

Learning:
* only telnet [server](https://forum.arduino.cc/t/what-does-client-stop-do-and-how-does-if-client-work-from-the-server-side/239211) can disconnect the session (in full featured servers something like the command 'close' to close the session)
* FluidNC has a small brother of Telnet with Grbl focused commands and doesn't seem to have a one for closing a session
* Adding a client.stop() to the source code is mandatory to let server know, it can close the session
* Luc pointed out that client.stop() also takes time, although the programm goes on. So I added a delay of 5 seconds

```
<reading and writing to Grbl>

client.stop();
delay(5000);

Log.verbose("Before while loop");
while (client.connected()) {
   while (client.available()) {
      Serial.write(client.read());
   }
}

<blinkSeveralTimes>
```

The last code line can be reached only by a preliminary client.stop();

Current test loop (opening, sending, receiving, closing, ESP.restart) runs for couple of minutes, so this looks promising.

closed 2022-01-01 by JensInternal