luc-github/ESP3D-WEBUI #221: [FEATURE REQUEST] Support for long file name in files pannel in 3.0

bg-master, 2022-01-23:
It would be very useful to list the long file name in the files panel, available since Marlin 2.0.9 (M20 L). At the moment to identify a file I manually execute the M20 L command from the terminal panel.

bg-master, 2022-01-23:
I can confirm that with Marlin 2.0.9.3 it works.

```
>>> M20
SENDING:M20
Begin file list
CAFA2B~1.GCO 17494
CA394B~1.GCO 60198
CA3603~1.GCO 900266
CALIBR~2.GCO 218439
CALIBR~4.GCO 1847167
End file list
ok P0 B0

>>> M20 L
SENDING:M20 L
Begin file list
CAFA2B~1.GCO 17494 Calibration_FirstLayerCenter.gcode
CA394B~1.GCO 60198 Calibration_ThinWallCube.gcode
CA3603~1.GCO 900266 Calibration_Retraction.gcode
CALIBR~2.GCO 218439 Calibration_StepMmCube.gcode
CALIBR~4.GCO 1847167 Calibration_Jerk.gcode
End file list
ok P0 B0
```

luc-github, 2022-01-24:
no space in long name are supported ?

bg-master, 2022-01-29:
Also spaces are supported.
```gcode
>>> M20 L
SENDING:M20 L
Begin file list
CA0211~1.GCO 1248145 Calibration Retraction Pyramid Multiple.gcode
CA4FEF~1.GCO 1847167 Calibration Jerk.gcode
End file list
ok P0 B0
```

luc-github, 2022-03-01:
@bg-master I am testing the M20 L
```
M20 L
Begin file list
CUBE.GCO 210240 CUBE.GCO
CUBE01.GCO 2089832 CUBE01.GCO
SUPPORT2.GCO 4613256 SUPPORT2.GCO
ARCHIVE/CUBE-C~1.GCO 210240 Archive/CUBE - Copy.GCO
NEWFOL~1/SUPPORT2.GCO 4613256 New folder/SUPPORT2.GCO
End file list
ok
M23 Archive/CUBE - Copy.GCO
echo:Now fresh file: Archive/CUBE
open failed, File: CUBE.
ok
```
files in sub folder are listed but cannot be printed - is that normal ? 
Seems consistent with comment : https://github.com/MarlinFirmware/Marlin/pull/23526#issuecomment-1015616650
But then I do not understand why to list sub folder
I am pretty sure it was working before - am I wrong ? Do I miss something ?

luc-github, 2022-03-03:
After more test
1 - long name path cannot work with M23
```
M23  '/New folder/SUPPORT2.GCO'
echo:Now fresh file: '/New
open failed, File: '.
ok
```
=> need to use short path for commands

2 - utF8 files cannot be printed in long name
 /盧克test.GCO
```
TEST~1.GCO 4613256 __test.GCO
```
=> cannot use UTF name even the option is enabled 

3 - long file name can be printed if on root if not UTF8

4 - utf8 long file name cannot be printed if on root
```
echo:Now fresh file: /盧克test.GCO
open failed, File: 盧克test.GCO.
ok
echo:Now fresh file: /__test.GCO
open failed, File: __test.GCO.
ok
```
=>Need to use shortname for UTF8 filename for commands

So long name is still only partially supported - and UTF8 is not

luc-github, 2022-04-24:
I am working on long name support and again new bug, any subdirectories are wipped in long name
if directory is DATA at root it disapear, if it is a subdirectory of subdirectory it disapear also

```
Begin file list
RESOUR~1/GCODE/APPLE~1.GCO 8140 resources//Apple.gcode
RESOUR~1/GCODE/BANANA~1.GCO 7554 resources//Banana.gcode
RESOUR~1/GCODE/CHERRY~1.GCO 6465 resources//Cherry.gcode
RESOUR~1/GCODE/PEACH~1.GCO 8467 resources//Peach.gcode
RESOUR~1/GCODE/PEAR~1.GCO 6010 resources//Pear.gcode
RESOUR~1/DATA/APPLE~1.GCO 8140 resources//Apple.gcode
RESOUR~1/DATA/BANANA~1.GCO 7554 resources//Banana.gcode
RESOUR~1/DATA/CHERRY~1.GCO 6465 resources//Cherry.gcode
RESOUR~1/DATA/PEACH~1.GCO 8467 resources//Peach.gcode
RESOUR~1/DATA/PEAR~1.GCO 6010 resources//Pear.gcode
TESTCUBE.GCO 353194 TESTCUBE.GCO
TEST1.GCO 1143935 TEST1.GCO
GCODE-~1/TESTLO~1.GCO 25 gcode - Copy/testlongname.gcode
GCODE-~1/TESTCUBE.GCO 353194 gcode - Copy/TESTCUBE.GCO
DATA/TESTLO~1.GCO 25 /testlongname.gcode
DATA/TESTCUBE.GCO 353194 /TESTCUBE.GCO
End file list
ok
```

luc-github, 2022-04-24:
Implemented - but do not handle Marlin giving wrong output

[image]

closed 2022-04-24 by luc-github

luc-github, 2022-08-26:
I tested latest release and now `M20 L` seems correct
```
20:43:41.223 > PID-PI~1.GCO 1490254 pID-pika2.gco
20:43:41.229 > IGX-PI~1.GCO 1573255 iGX-pika2.gcode
20:43:41.231 > TINYBE~1.GCO 2014025 tinybeev1-Body.gcode
20:43:41.234 > MKSMIN~1.GCO 575334 mksmini12864v3.gco
20:43:41.237 > SUBFOL~1/TINYBE~1.GCO 2014025 SubFolder/tinybeev1-Body.gcode
20:43:41.243 > SUBFOL~1/TINYBE~2.GCO 2014025 SubFolder/tinybee Body.gcode
20:43:41.249 > NEWFOL~1/TINYBE~1.GCO 2014025 New folder/tinybee Body.gcode
20:43:41.254 > NEWFOL~1/TINYBE~2.GCO 2014025 New folder/tinybeev1-Body.gcode
20:43:41.260 > End file list
20:43:41.260 > ok
20:43:42.197 > echo:192.168.2.178
M23 SubFolder/tinybeev1-Body.gcode
20:44:03.392 > echo:Now fresh file: SubFolder/tinybeev1-Body.gcode
20:44:03.400 > File opened: tinybeev1-Body.gcode Size: 2014025
20:44:03.404 > File selected
20:44:03.404 > ok
M23 'SubFolder/tinybee Body.gcode'
20:44:36.966 > echo:Now fresh file: 'SubFolder/tinybee
20:44:36.970 > open failed, File: 'SubFolder.
20:44:36.972 > ok
M23 "SubFolder/tinybee Body.gcode"
20:44:56.087 > echo:Now fresh file: "SubFolder/tinybee
20:44:56.090 > open failed, File: "SubFolder.
20:44:56.092 > ok
```
But space are not allowed - I have check Marlin `M23` code and indeed the space are used to cut the path
so as workaround I will do https://github.com/luc-github/ESP3D-WEBUI/issues/259