luc-github/ESP3D-WEBUI #225: [BUG] Printer Configuration is empty

dagarwal82, 2022-01-27:
**Describe the bug**
No customization. Downloaded Marlin 2.0.9.3, board is BOARD_RAMPS_14_EFB. Just enabled EEPROM_SETTINGS and flashed Arduino Mega 2560. Downloaded ESP3D-2.1.1 and flashed the ESP32-CAM board.
Uploaded latest index.html.gz (tried both from ESP3d folder and esp3d-webui github repo). 
But Printer Configuration is empty

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**ESP3D Firmware:  (please complete the following information)**
- Version:2.1.1
- Web UI Version:2.1.1
- Wifi mode:Both
- Any change done in FW - None

**Target Firmware:  (please complete the following information)**
 - Name: Marlin
 - Version 2.0.9.3

**Board used (please complete the following information):**
 - MCU: Ai Thinker ESP32-CAM
 - Name:[e.g. NodeMCU2s] 
 - Flash size: [e.g. 4M: 2M/2M]

**Browser used (please complete the following information):**
- Name: Chrome
- Version  [e.g. 2.0]
- Operating system [e.g. Windows 10]

**Additional context**
Add any other context about the problem here.
Developer tools output, error message, etc...

luc-github, 2022-01-27:
Only printer config does not show?
What typing M115 and M503 show in webui terminal ?

dagarwal82, 2022-01-27:
Controls doesn't work either. 

[image]
[image]
[image]
[image]
[image]
[image]

luc-github, 2022-01-27:
so issue is communication problem
How do you wire esp to mega?
what baud rate on each serial?

dagarwal82, 2022-01-28:
Yes, I am definitely something wrong.
Using AUX 1 of RAMPS 1.4 
5v, ground, Rx, Tx (thats the config on AUX 1) . I am connecting Rx to Tx and vice versa between RAMPS and ESP32 drive.
baud rate as in the screenshot is 115200
[image]

RAMPS 1.4 is connected to Mega and ESP is connected to RAMPS 1.4

luc-github, 2022-01-28:
ok I think the issue is you directly wire a 5V board' s pin to a 3.3v board pin's - esp32 is really not 5V tolerant - you need to decrease the voltage on esp32 RX  pins - hoping you did not damaged your board
check https://github.com/luc-github/ESP3D/wiki/ESP-32-CAM or https://github.com/luc-github/ESP3D/wiki/D1-mini

dagarwal82, 2022-01-28:
Ok going to try. rest all the things works. I can control everything by connecting to computer.

dagarwal82, 2022-01-28:
Just to confirm its ESP32-CAM (not just ESP32,  If that makes sense I am novice). still 3.3v ?

luc-github, 2022-01-28:
yes when esp8266 can handle such way (but I still do not recommend it),  all esp32 based mcu are using 3.3v on rx/tx and any GPIO pins so model does not matter and they are really not 5V tolerant at all

dagarwal82, 2022-01-28:
Question. shoud I follow the links that you shared for the connection or change it to use 3.3v . both links that you shared are using 5v of ESP32-CAM

luc-github, 2022-01-28:
the board power and GPIO voltage are 2 differents things - some people people power the esp32 using 5v others use 3.3v - it depends of what you have and also of power supply current stability
I personnaly always power esp32 cam using 5V because camera drain a lot

dagarwal82, 2022-01-28:
Aaaahhh  I understood now. I didn't see (or choose not to see :D ) the Resistors in your link. I will add some resistors, thought it was out of the box. 
By the way just saw smoke coming out of ESP32-CAM because I connected 5v of RAMP1.4 briefly to 3.3v ESP32. :D . haha
Gotta get another one and try your solution. thanks

luc-github, 2022-01-28:
Ho - sorry - yes I told you esp32 is not 5V tolerant

dagarwal82, 2022-01-29:
Followed everythng as per the instruction, fried another one but stll no luck. I used multimeter to test the voltage. I am getting 1.6v on both tx and rx. but now not even seeing ESP32 wifi access point

dagarwal82, 2022-01-29:
Well getting wifi now. but as soon as I connect tx and rx to RAMPS 1.4 my esp32 does not work

dagarwal82, 2022-01-29:
I am going to validate this schematic error. May be thats why I am having issues.
http://notserpe.blogspot.com/2012/05/ramps-14-schematic-errors.html

dagarwal82, 2022-01-29:
No luck

dagarwal82, 2022-01-30:
I suspect the default baud rate I am using 1152000 (Just noticed RAMPS 1.4 has 2500000) . Waiting for the 5v voltage regulator now to test

DivingDuck, 2022-01-30:
A zero too much can have huge impact, hopefully only a typo:
**1152000 (Just noticed RAMPS 1.4 has 2500000)**

dagarwal82, 2022-01-30:
Yes that was typo. Everything works now ! it was the baud rate issue

closed 2022-02-28 by luc-github