luc-github/ESP3D-WEBUI #226: [FEATURE REQUEST]Notification panel

luc-github, 2022-02-02:
Add notification panel that keep all notifications displayed by toaster error and [ESP600] notifications.
allowing to scroll/clear all already displayed notifications

MitchBradley, 2022-02-02:
I don't know if this fits into your UI paradigm - but instead of a panel that takes up screen real estate, I would prefer a modal dialog that would float over the top of the UI when you ask for it, then dismiss it when you are done.  Notifications like this are not something that you interact with in normal use.  I think that panels should be for things that need to stay on the screen during normal use.

luc-github, 2022-02-03:
Panels are show/hide anytime and unlike modal dialog they are updated in live
This panel is more a log of all notifications in same way the terminal is a log of all commands received, it does need to be opened at start and even no visible if user do not need it.
A float modal dialog is modal -> blocking so I do not see the point to be float, additionaly a float dialog is not something easy to handle in mobile view, so instead of creating a new control specificaly for the notification log which will increase code size - using panel API is the solution that fit all needs currently.
If in future when UI is done and size is small as expected, we can improve the user experience by adding new kind of control / user flow, it will be definitly done - but right now I prefer to stick to current API implementation if it fit the needs.

luc-github, 2022-03-29:
Notifications panel added
[image]

it add 69,195 bytes->69,667 bytes = + 472 bytes  which is I think not too much

closed 2022-03-29 by luc-github