luc-github/ESP3D-WEBUI #239: [FEATURE REQUEST]Print status panel

luc-github, 2022-05-10:
Need to display print status with correspondig informations / commands
* status 
* progress 
* printing filename
* pause / resume / stop buttons

* information page would show only status / progress

luc-github, 2022-05-11:
here messages that need to be handled as POC with Marlin, according command or status
* Not printing:
   - M27 => Not SD printing
   - M27 C => Current file: (no file)
* Printing
  - M27 =>SD printing byte 2432/5557
  - M27 C => Current file: TESTPR~1.GCO testprint.gcode
* Printing done
  - Done printing file

* Status:
  - echo:busy: processing
  - Error:Heating failed, system stopped! Heater_ID: E0
  - echo:Homing Failed
  - Error:Printer halted. kill() called!

For status some are successively displayed - so will display the last one, for other need to refer to terminal output
e.g: `echo:Homing Failed` won't be displayed or very fast because just after `Error:Printer halted. kill() called!` will be displayed

luc-github, 2022-05-30:
Implemented

olivierbarriere, 2022-06-09:
It already works great, but would it be possible to handle M31 (time elapsed since the start of the current print job) in the print status panel?

https://reprap.org/wiki/G-code#M31:_Output_time_since_last_M109_or_SD_card_start_to_serial
Example
M31
The response looks like:
echo:54 min, 38 sec

And also maybe an estimate of the remaining time as:
remaining  = elapsed * (100 - progress) / progress

luc-github, 2022-06-09:
Hmm I am not sure if it will be very reliable to detect this output, need to add a regex to detect and extract the value for computation here :
https://github.com/luc-github/ESP3D-WEBUI/blob/3.0/src/targets/Printer3D/Marlin/filters.js#L96-L131

I am not sure the computation will be correct as it come from 2 differents outputs with possible delay between them

olivierbarriere, 2022-06-09:
I just tested on the printer, and the message appears with a "Print time: " description after "echo:"
echo:Print time: 9m 57s

That's now an explicit message, and it can be displayed as is in "Print time" in the status panel without the need to match the full regexp. The full regexp is still useful if we want to reconstruct the time in seconds, estimate the remaining time, and back transform it into y/d/h/m/s.

By the way, when there is no print job, the output is
echo:Print time: 0s 
which should be easy to ignore

luc-github, 2022-06-09:
doc says it display time if   M109 is active

luc-github, 2022-07-17:
I have implemented the stream support in GRBL webUI - it actully does not support the job progress but the stream progression which is different - so it act exactly as offline controler:
[image]

I will check if this can be improved but I am not sure as in ESP3D the streaming process only react to `ok` and does not catch/analyze any others answers as WebUI do

Next Week I will add the `M31` support for Marlin as I have missed it

luc-github, 2022-07-21:
M31 support for Marlin is implemented
[image]

closed 2022-07-21 by luc-github