luc-github/ESP3D-WEBUI #242: [FEATURE REQUEST]Configuration  Wizard

luc-github, 2022-05-10:
now all ESP3D Feature are ready and 3D printer  feature are almost ready too, it is time to define a configuration wizard to:
* Setup Network
* Setup UI according printer features

luc-github, 2022-06-05:
As now working on setup wizard here the possible steps

0.    Collect [ESP400] data if not yet available
1.    Configure network:
        * Select AP/STA 
        * Access Point SSID to connect
        * Access Point Password to connect
        * Set ESP Name
 
2. Restart System if changed Sta setting to allow internet access
3. Set Firmware target if necessary (only for ESP3D firmware)
4. Check Firmware target match the webUI 
   * Propose to update WebUI if internet access 
   * Raise error if not,  and stop Wizard asking for update
5. If internet access, propose to go to Internet to download language pack and apply it
6-a. If internet access, propose to download profile (FW settings + WebUI settings)
6-b. If no
   * Setup SD access (if needed)
   * Setup Polling commands (if needed)
   * TBD
   
   
The goal is to have a light setup for basic features to keep the footprint small, the advanced setup could be done by the online setup or download profile.

Any feedback / suggestion is welcome

atsju, 2022-06-06:
Although not directly related I think this could fit here.

Could you change the client mode SSID from `ESP3D` to `NETWORK_SSID` to avoid confusion ? This way ESP3D is only used for client SSID

luc-github, 2022-06-06:
Do you mean this one : https://github.com/luc-github/ESP3D/blob/3.0/esp3d/src/core/settings_esp3d.cpp#L150

It is in FW not in related - yes can be changed

atsju, 2022-06-06:
Yes this one.