luc-github/ESP3D-WEBUI #245: don't show heater t1[BUG]

tramnon, 2022-05-24:
i have mks tinybee board...it doesn't show heater t1 on webui...while showing fully on lcd 12864

luc-github, 2022-05-24:
did you enabled  it in settings in webui ?

tramnon, 2022-05-24:
> bạn đã bật nó trong cài đặt trong webui chưa?

it not in settin
[image]
g

luc-github, 2022-05-24:
You selected mixed extruder - which mean only one heater for 2 extruders => unselect mixed extruder

tramnon, 2022-05-24:
> Bạn đã chọn máy đùn hỗn hợp - có nghĩa là chỉ có một máy gia nhiệt cho 2 máy đùn => bỏ chọn máy đùn hỗn hợp

i unchecked...it's still the same.

luc-github, 2022-05-24:
I do not duplicate - what is your WEBUI version ?

[image]
[image]

luc-github, 2022-05-24:
can you do a page refresh and go to settings and show me the settings for extruders ?

tramnon, 2022-05-24:
> can you do a page refresh and go to settings and show me the settings for extruders ?

i just reloaded FW and used ui 2.1.1...it works.
thank you very much.

luc-github, 2022-05-24:
Ok weird b77 should be identical as the 2.1.1 
so I close issue

closed 2022-05-24 by luc-github