luc-github/ESP3D-WEBUI #246: [FEATURE REQUEST] v3.0-alpha2 configuration rendering

atsju, 2022-06-06:
**Is your feature request related to a problem? Please describe.**
There menu for client mode setup has a sliding panel that is just some pixels shorter than what it would need to have no sliding panel.
On phone with short screen it becomes difficult to slide the page because in fact you are limited by the sliding panel inside the page. To slide the page, you need to click on the title and slide it there leaving not much marging.

**Describe the solution you'd like**
Either the page could continue sliding once the panel is at max sliding (not sure if this is a browser setting or a webpage setting)
Or have no sliding panel in the page (easiest way for developper and user I suppose)
Or have shorter panels for sliding panel (for example if you aimed to have the title of section always visible)

luc-github, 2022-06-06:
Sorry could you do screenshot ?- I do not understand the problem -  all is managed automaticaly and rendering on depend resolution - so I may not reproduced your issue - see screenshot using emulator
[image]

atsju, 2022-06-06:
[image]
Let me know if this is clearer

luc-github, 2022-06-06:
What is your phone ? what resolution it has ? 
I do not reproduce your issue - it use android UX which react differently according pressure / gesture you use 
I can navigate both in panel and page

atsju, 2022-06-06:
redmi note 7 
from the internet screen has: 2340 x 1080 pixels

I use chrome on android. It's fully possible that better browser let you slide the page once the panel is at the end. For me it's not the case.

luc-github, 2022-06-06:
well I do not know - my thumb is always in middle of page, I never use the slider
* if  your thumb is inside the panel => it is slide panel =>Normal
* if you thumb is on header bar of panel => it slide page => Normal 
* if your thumb is on panel but push is soft  => it slide page => Normal

atsju, 2022-06-06:
My thumb is in middle of page also. Slider bar just shows my problem

> if your thumb is inside the panel => it is slide panel =>Normal
This is what I talk about. I find this annoying when the panel will slide only max half a line... No need for panel if you do not need to slide much.

Now with usage I understand how to slide page from within panel (not related to soft push for me):
you go down in sliding the panel, release thumb, wait for slider bar to disappear and the because you are already at the end, you can slide again and it will slide the page.
If you change direction you need to slide the panel again, wait, slide the page.

Now that usage is clear it's "simple" but it would be easier without sliding panel at all. Just have a fixed length page ?
Of course this comment is only related to my screen size but I don't see what this particular sliding panel brings (I tried phone horizontally and although screen is much "shorter" it still slides only half a line)

luc-github, 2022-06-06:
Well resolution is definitly a nightmare in web dev  -  2.1 I had to chase all user cases (and actually there are still a lot of not handled situations) , so in 3.0 I decided to rely on rest API and no put limit the panel limit are necessary for several panel with dynamic and huge content, terminal, macro, filesystem, external content, for navigation it also allow to not fully  parse all panels before reaching the targeted one.
There is not best solution - only some mitigation according situations

atsju, 2022-06-06:
I have absolutely no experience in web dev, if you tell me there is no solution then there is no solution.

However I'm really sorry if I'm too insistant but I like to understand. 

[image]

The webpage on computer has also this famous sliding panel. Exactly at same size as on phone. 
Again this sliding panel is just slightly shorter than what page would render without sliding.
Have you set no limit and all my browser are rendering in same way? or you set a global limit somewhere because it's needed for 
> panel with dynamic and huge content, terminal, macro, filesystem, external content

?
--If it's possible and doesn't complexify the code too much-- Maybe it makes sense to remove the limit on the settings page as it has a known fixed size.

Anyway I understand priorities and using 3 days on rendering of this panel would make absolutely no sense regarding your other developments in progress.

luc-github, 2022-06-06:
As mentioned earlier it is all matter of resolution:
on my screen which is considered a `standard` screen 1920x1080 => no issue
[image]

And one thing that also is important in addition to screen resolution - it is HDPI scalling which can be additional pain

luc-github, 2022-06-06:
And I got a different feedback than yours before on this UI, I got request to put panels smaller so all fit in computer screen ^_^, which means  more slider visibles on some panels ^_^
I said same => it  is all matter of resolution and mitigation

atsju, 2022-06-06:
No problem. Thank you for clarification anyway.

Side info, my screen is 1920x1080. I did not search for HDPI setting. Zooming or changing window size did not affect rendering much (which is good)

luc-github, 2022-06-10:
I close issue as it is resolution and HDPI related and there is not real solution

closed 2022-06-10 by luc-github