luc-github/ESP3D-WEBUI #251: [BUG]Marlin Mixed extruders  is incorrect

luc-github, 2022-07-19:
According https://marlinfw.org/docs/gcode/M165.html
* Number of extruders is limited to 6 but currently in webUI 3.0 there is no limit
* Name of extruders is A B C D H I, currently there is no specific name but A B C ...

ETE-Design, 2022-07-20:
@luc-github How about make it possible to rename them?

luc-github, 2022-07-20:
rename ? but only A B C D H I are allowed
what name do you want to give?

luc-github, 2022-07-20:
but you can put the color of control to reflect the color

luc-github, 2022-07-21:
Issue is fixed by https://github.com/luc-github/ESP3D-WEBUI/commit/42605a2445e5b8ca5242988308362da7e23a14d5

[image]

closed 2022-07-21 by luc-github