luc-github/ESP3D-WEBUI #252: [FEATURE REQUEST]Add GRBLHal support : https://github.com/grblHAL

luc-github, 2022-07-27:
As started to discuss here https://github.com/grblHAL/ESP32/issues/39

terjeio, 2022-07-27:
An initial update for WebUI 3.0 of the [grblHAL WebUI plugin](https://github.com/grblHAL/Plugin_WebUI) is now available. I am testing with a [Nucleo STM32F756 board](https://github.com/grblHAL/STM32F7xx) and the plugin should also work with the [iMXRT1062 (Teensy 4.x) driver](https://github.com/grblHAL/iMXRT1062). Note that the [ESP32 grblHAL driver](https://github.com/grblHAL/ESP32) does not use the plugin, I will backport the code changes to that later on.

@dresco is working on a [STM32H7xx driver](https://github.com/dresco/STM32H7xx) that has ethernet and SD card support, it will be possible to use the plugin for this driver when it is updated to use the latest [grblHAL core](https://github.com/grblHAL/core).

luc-github, 2022-07-27:
Ok understood - unfortunatly I have no STM boards but I can emulate webserver and board responses in webUI dev mode

I saw the the https://github.com/grblHAL/Plugin_WebUI but I was thinking was same code in https://github.com/grblHAL/ESP32 so I did not realize it was V3 based - I will have a look

luc-github, 2022-07-28:
@terjeio you wrote 
> my plan to contact you about some issues I have encountered

Can you tell which ones  ?I can start by these first

terjeio, 2022-07-28:
Issues encountered:

ESP701 for querying the status of a running job suddenly (randomly) is no longer sent after a running a few. Reloading resumes the querying.

If wrong password is given on login correcting it and resubmitting leads to the USER field in the form data missing.

The default file system is FLASHFS, could this be the first listed in the target `supportedFileSystems` (in _files.js_) or perhaps read from the ESP800 response?

https://github.com/luc-github/ESP3D-WEBUI/blob/bac40f99ee64e1ed310ba924acc120e5f1bcf166/src/components/Panels/Files.js#L509-L512

I've changed the code above and it seems to work but not sure if this has any side-effects (other than failure if none are specified):

```
        if (useUiContextFn.getValue("autoload") && currentFS == "") {
            currentFS = files.supported[0].value
            onSelectFS()
        }
```

> Is grblHal typo correct ? I saw also different writing like GRBLHal and GRBLHAL

"grblHAL" is the correct version. The welcome message from the controller is not correct, but I have been reluctant to change it as it may break senders.

> do you have a logo for grblHal

There are a few proposals but none has been decided upon. I have to make my mind up...

> Do you want to go to https://github.com/grblHAL/ ?

Yes, I have changed this already.

> I see on esp32 version you cannot update the webUI as do not use the maintenance page and directly flashed the webUI,
is that the same with pluggin ?

Yes. For the plugin changing this will require a flash filing system. I may add this later. If the WebUI is compiled into the binary then adding _index.html.gz_ to the SD card _www_ folder will override the one from the binary so there is an update path available. More work on this is needed on my end - perhaps adding download of a new version to the _www_ folder as a first step?

> I will check what play SD job commands you are using

I am using $F=, I can also use ESP220 (IIRC this is used by WebUI 2.1) but when I check your ESP3D source it seems to me that this (and some other commands) has been repurposed? The commands that may have been reporposed have the help text "??" in [this list](https://github.com/grblHAL/Plugin_WebUI/blob/2c7eb54f92ff2309f730ef24594d77aa9f7cd827/commands.c#L114-L177).

> I see extra real time commands: ... Is that list complete / up to date ?

I am not sure it is up to date, I'll have to check. Most can be ignored?
Perhaps the most important change is the new _Tool_ state, it would be great if WebUI could support [grblHAL style tool change handling](https://github.com/grblHAL/core/wiki/Manual-tool-change-protocol). And yes, this works even when running a job from the SD card.

Here are the preferences and target folder I created:

[preferences.zip](https://github.com/luc-github/ESP3D-WEBUI/files/9205610/preferences.zip)

[grblHAL.zip](https://github.com/luc-github/ESP3D-WEBUI/files/9205612/grblHAL.zip)

---

Should grblHAL get its own firmware [target name and id](https://github.com/grblHAL/Plugin_WebUI/blob/2c7eb54f92ff2309f730ef24594d77aa9f7cd827/commands.c#L57-L58)?

Some ideas:

Can WebUI fetch the settings labels from the controller? grblHAL has dynamic settings handling - plugins may add new ones, even user defined plugins. In [my sender](https://github.com/terjeio/ioSender) the settings UI is built from information requested from the controller. I can easily add an ESP command for fetching a json formatted list of setting ids and labels (and even descriptions) in a json formatted response.

On a longer term todo list will it be possible to use the WebUI style settings handling? It will require the support for floats and a flag type where bit 0 controls the enable status for the rest.

luc-github, 2022-07-28:
>ESP701 for querying the status of a running job 

Currently this command is for the ESP3D Gcode host not used for running job, why not use the ? like GRBL_ESP32 ? 
I need to see your implementation, originally ESP command are send by http and answer come back by http to, unlike others commands which use websocket, if polling is stopped it means there is a crash that should be visible in dev console or an error in notifications panel

>If wrong password is given on login correcting it and resubmitting leads to the USER field in the form data missing.
I need to check but I may did that on purpose to display or not the error message

> The default file system is FLASHFS, 

This is already the case, no ? Flash is the default because it is the one I know it is present for sure, so not sure what issue you meet

>I am using $F=, I can also use ESP220 

Yes these commands are used by GRBL_ESP32 which is only supported by WebUI 2.1, and since the GRBL_ESP32 is dropped, I did not maintained them in WebUI 3.0 

>I am not sure it is up to date, I'll have to check. Most can be ignored

it is just to allow these command to be used by macro using a text descriptor as they are not printable,

I also see some commands : `?`,`~`,!` need to be update by the not printable code to be handled properly in grblHal, so will do

>would be great if WebUI could support [grblHAL style tool change handling](https://github.com/grblHAL/core/wiki/Manual-tool-change-protocol).

You will have to tell me more what you expect for UI and behavior, but I see no issue to support it

>[target name and id](https://github.com/grblHAL/Plugin_WebUI/blob/2c7eb54f92ff2309f730ef24594d77aa9f7cd827/commands.c#L57-L58)?

The current webUI use grblHal as name, for the ID I did not defined it yet, but because it is not selectable for ESP3D, it is not major issue, 
here the current list : https://github.com/luc-github/ESP3D/blob/3.0/esp3d/src/include/defines.h#L28-L36
If you are ok to use ID=80 I will set it, it will be used to be sure WebUI match the target FW

>Can WebUI fetch the settings labels from the controller? 
currently it is the case, everything is dynamic, these labels are then translated by translation tool to give better look and correct language

So actually you can add any sections / items you want in settings just follow current syntax, if need more type of settings, just need to define type, type constraint, rendering form and should be ok like others types

luc-github, 2022-07-28:
Reading your version now 
* preferences content has some changes already, polling commands can have difference frequency
* your issue with  Flash FS as first is due to the order put in files.js but in latest version of WebUI order is correct

reading ongoing...

luc-github, 2022-07-28:
I have updated and build the index.html.gz according your changes https://github.com/luc-github/ESP3D-WEBUI/blob/3.0/dist/CNC/GRBLHal/index.html.gz

I still need to update the real commands parts

I have also noticed you do not support the `one session feature` so many webUI can be open at once, is that a choice ? I won't recommend it

terjeio, 2022-07-28:
>>   ESP701 for querying the status of a running job

> Currently this command is for the ESP3D Gcode host not used for running job, why not use the ? like GRBL_ESP32 ?
I need to see your implementation, originally ESP command are send by http and answer come back by http too, unlike others commands which use websocket, if polling is stopped it means there is a crash that should be visible in dev console or an error in notifications panel

Polling for the real time status continues, only the ESP701 request stops. If I remove the ESP701 request the stream status is not displayed even if it is available in the real-time report. I assume this could be changed to use the real time report instead?

>> The default file system is FLASHFS,

> This is already the case, no ? Flash is the default because it is the one I know it is present for sure, so not sure what issue you meet

It is not present in the controllers using the plugin.

>>    I am using $F=, I can also use ESP220

> Yes these commands are used by GRBL_ESP32 which is only supported by WebUI 2.1, and since the GRBL_ESP32 is dropped, I did not maintained them in WebUI 3.0

Ok, this explains it. They should not have been reassigned to a different use then? Or was this a customization made by bdring and his team?

>> [target name and id](https://github.com/grblHAL/Plugin_WebUI/blob/2c7eb54f92ff2309f730ef24594d77aa9f7cd827/commands.c#L57-L58)?

> If you are ok to use ID=80 I will set it, it will be used to be sure WebUI match the target FW

If not needed I am ok with using ID=10 as I do not know what it is used for in your code.

>>Can WebUI fetch the settings labels from the controller?

> currently it is the case, everything is dynamic, these labels are then translated by translation tool to give better look and correct language

Ok, what is the ESP command to use for returning those. And to be clear I am talking about the _grblHAL_ settings where those not known by the WebUI does not have useful labels:

[image]

> preferences content has some changes already, polling commands can have difference frequency

Ok, will update. Sun came out today after a period of rain - so it may take some time. And I will be away for a period starting Tuesday 2nd.

> I have also noticed you do not support the one session feature so many webUI can be open at once, is that a choice ?

Definitely not - or at least until an abitration protocol can be added. Only one client should be in control at the time. As a first step I might add support for additional read-only clients - but this is longer term.

luc-github, 2022-07-28:
these are in language pack - they are just translation from commands id : https://github.com/luc-github/ESP3D-WEBUI/blob/3.0/src/targets/CNC/grblHAL/translations/en.json

>the one session feature 
this feature is actually mostly handled by webUI, the FW just broadcast the last connected ID by websocket , so the old ID webUI automaticaly close - not hard to implement - I may try to do it if you want

I need to setup an esp32 system for grblHal as I am afraid I will do some blind code and may misunderstood  some commands

if I replace the plugin files to existing ESP32 code would it work ? any other changes should I do  - I can do changes manually and this will let me know also why  current version is always in reset error on my board

dresco, 2022-07-28:
> @dresco is working on a [STM32H7xx driver](https://github.com/dresco/STM32H7xx) that has ethernet and SD card support, it will be possible to use the plugin for this driver when it is updated to use the latest [grblHAL core](https://github.com/grblHAL/core).

Just fyi, have started looking at this. First issue is that I used the STMCubeIDE provided middleware for SD card access (using DMA & SDMMC peripheral) - which uses an older FatFs version. Am looking into what it's going to take to upgrade..  

@terjeio should I aim for R0.13c as per F4/F7 etc, or latest R0.14b?

terjeio, 2022-07-28:
>> the one session feature

> this feature is actually mostly handled by webUI, the FW just broadcast the last connected ID by websocket , so the old ID webUI automaticaly close - not hard to implement - I may try to do it if you want

Wait... One step at a time please. I have some support issues to clear before I can dig into the details.

> if I replace the plugin files to existing ESP32 code would it work ?

No, it is a major refactoring. Ideally I should update the ESP32 driver and the plugin so it could be used directly - I do not like to maintain duplicate code.

>  why current version is always in reset error on my board

Oops, I forgot to asnwer this, by default grblHAL uses [normally closed switches for all inputs](https://github.com/grblHAL/core/wiki/First-Run-Grbl-Settings). Connect the reset pin to ground and you should be fine.

@dresco - the SD card plugin can work with earlier versions of FatFS, I should have used the version id provided by FatFS but are using the [target MCU family](https://github.com/grblHAL/Plugin_SD_card/blob/382b81bd27e89130c06ce644ec04a2af5faa4809/sdcard.c#L30-L32) for now for detection... If you update aim for the latest version.

dresco, 2022-07-28:
> @dresco - the SD card plugin can work with earlier versions of FatFS, I should have used the version id provided by FatFS but are using the [target MCU [...]

Thanks, was actually the webui plugin that complained, may just be something minor...

luc-github, 2022-07-28:
WebUI is updated to a7

*  Additional real time commands have been added for grblHal and replaced in WebUI panels accordingly : https://github.com/luc-github/ESP3D-WEBUI/blob/3.0/Memo/realtimecmd.txt
* FlashFS  won't be listed/used  if  in ESP800 the `FileSystem` is set to `none`
```
[ESP800]
{
"cmd": "800",
"status": "ok",
"data": {
 "FWVersion": "3.0.0.a111",
 "FWTarget": "grblhal",
 "FWTargetID": "80",
 "Setup": "Enabled",
 "SDConnection": "direct",
 "SerialProtocol": "raw",
 "Authentication": "Disabled",
 "WebCommunication": "Synchronous",
 "WebSocketIP": "localhost",
 "WebSocketPort": "81",
 "Hostname": "esp3d",
 "WiFiMode": "STA",
 "WebUpdate": "Enabled",
 "FileSystem": "none",
 "Time": "none",
}
}
```

luc-github, 2022-07-28:
about : Can WebUI fetch the settings labels from the controller?
it is possible just need to define the format you want to use and I can customize the grblHal panel for settings because it is specific, currently I use the GRBL `$$` display but we can imagine another command that do same and add label and constraint for each like for ESP3D settings even section if necessary and it will dynamically build the UI accordingly

terjeio, 2022-07-29:
I have just committed an update to the [ESP32 driver](https://github.com/grblHAL/ESP32) that uses (parts of) the [WebUI plugin](https://github.com/grblHAL/Plugin_WebUI).
Add _preferences.json_ to the root of the SD card.

I left the `SerialProtocol` value to `Socket` in the [ESP800 response](https://github.com/grblHAL/Plugin_WebUI/blob/c490f79580348a9b1f26e9df3dd0cff2f74a3e49/commands.c#L1085-L1109) - not sure ift that is correct but it works for me.
There _is_ flash filesystem support for the ESP32, I will change the `Filesystem` value later to reflect that.

Note that forms based autentication is not yet supported for the ESP32 driver.
I'll put improved settings handling down the list until the basic stuff is completed - forms based login and unified SD card handling are the largest items.

And BTW the ESP32 webserver is really slow compared to e.g. the iMXRT1062 controller where WebUI loads in the blink of an eye...

luc-github, 2022-07-29:
I have just committed an update to the [ESP32 driver](https://github.com/grblHAL/ESP32) that uses (parts of) the [WebUI plugin](https://github.com/grblHAL/Plugin_WebUI).

Ok thank you I give a try tomorrow - I was out all day today and need to finish some code first

SerialProtocol value to Socket in the [ESP800 response] - it is fine - there is no use currently in WebUI but better to be consistent, if the need popup

Ho yes ESP32 sync webserver is very slow, the test I did on current grblHal is already far faster (it use the idf async http server right  ?) I am not surprised on another chip the performances are better - on reprap they claim to have wifi file transfer of 1.2MB/s but they do not use esp webserver ... 
I have also noticed performances issues related to SD https://github.com/luc-github/ESP3D/issues/576, so I use SDfat instead of ESP32 SD library in ESP3D

thank you for the porting It will be easier for me to experiment - I will feedback you tomorrow

luc-github, 2022-07-30:
Hi I have flashed my board with latest git and webui 3.0, very cool job thank you, 

so I have some questions :

* Is that a choice to not display the IP at start ? it is painful when board is in STA/Client mode but start in reset and so it is locked , cannot send any command to know the IP, I see mdns is present but no SSDP, so on windows, the only choice is to find info on DHCP server ...

* ESP800
```
[ESP800]
FW version:1.1f
FW target:grblHAL
FW ID:80
SD connection:direct
Serial protocol:Socket
Authentication:Disabled
Web Communication:Synchronous
Web Socket IP:192.168.2.108
Web Socket Port:81
Hostname:Grbl
WiFi mode:AP
File system:directsd
Hostname:Grbl
```  
Some comments :
- In STA mode - it still display `WiFi mode:AP`
- `File system:directsd`  `directsd` is not supported should be `none` or `spiffs` or `littlefs`
- Above status lead me to add a new flag : `hostFS`,  because you mentioned you do not use Flash on some system, so the index.html.gz is hosted on SD (or USB may be in future), so WebUI update, preferences.json should be read/uploaded from/to SD/www, using the new flag hostFS would tell webUI where to load/save preferences.json, what do you think ?

* Settings
the settings use old format of 2.1 instead of 3.0, so they cannot be loaded, and webUI raise this error :
`WebUI and Firmware version are not compatible`, WebUI 3.X has new output format for all ESP commands, the same as you used in about page, I enclose both format

[3.0.txt](https://github.com/luc-github/ESP3D-WEBUI/files/9224114/3.0.txt)
[2.X.txt](https://github.com/luc-github/ESP3D-WEBUI/files/9224115/2.X.txt)

* $G
give me `[GC:G0 G54 G17 G21 G90 G94 G49 G98 G50 M5 M9 T0 F0 S0.]`
what is the meaning of `.` in `S0.` ?
Form another side, is that normal to have `F0` instead of `F0.0` as mentioned : https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#grbl-response-messages
Because I need to know to adjust the regex catching patern to get informations and processing them, and there is not information on https://github.com/grblHAL/core/wiki/Report-extensions page

* New flags / states
That is for me to read https://github.com/grblHAL/core/wiki/Report-extensions and adjust each triger / rendering one by one

Sorry to ask so many questions, hope you can answer before  you will going to vacations - so I won't bother you 😸

terjeio, 2022-07-30:
> Is that a choice to not display the IP at start ?

Yes, as it cannot be displayed for all the platforms grblHAL supports, see [this wiki page](https://github.com/grblHAL/core/wiki/For-sender-developers). I may add SSDP and/or add the IP address to the full status real-time report to fix this.

This is the latest v3 json version:

> ESP800

`{"cmd":"800","status":"ok","data":{"FWVersion":"1.1f","FWTarget":"grblHAL","FWTargetID":"80","Setup":"Enabled","SDConnection":"direct","SerialProtocol":"Socket","Authentication":"Enabled","WebCommunication":"Synchronous","WebSocketIP":"10.0.0.22","WebSocketPort":"81","Hostname":"Grbl","WiFiMode":"STA","WebUpdate":"Disabled","FileSystem":"none","Time":"none"}}`

Note that the backend supports v2.1 as well as v3.0, I use the version parameter added to the ESP800 command to detect 3.0 and set the mode for all subsequent commands (until a new ESP800 is requested):
`http://10.0.0.22/command?cmd=[ESP800]json=yes%20version=3.`

> the CPU freq is 19 - I guess this is wrong

Yep, it is the stepper code clock frequency, I have to add a new HAL value for the MCU clock...

> $G ... what is the meaning of . in S0. ?

I guess it is for implying that the value is a float - inherited from legacy Grbl.

> Form another side, is that normal to have F0 instead of F0.0

Good question, I have to check my code.

> I see `$ES` so I may use it instead of `$$` to list grblHal settings, is that ok ?

No, `$ES` is for getting information about the settings, not the values. There is an internal API for querying setting details which I use to [generate the ESP400 response](https://github.com/grblHAL/Plugin_WebUI/blob/c490f79580348a9b1f26e9df3dd0cff2f74a3e49/commands.c#L593-L678), for most grblHAL settings this is relatively straight forward. The settings API is designed to allow user defined settings (in plugins or board support code), these are also possible to query and will show up in the `$ES` response etc.

> change command still seems same `$<ID>=<value>`, right ?

Correct. I try to be backwards compatible with the legacy Grbl specification, at least in spirit.

>  if I read well need to send `$EG` then $ES then `$$` to be able to display settings properly ?

I would rather like to add a ESP command for querying setting details, same format as ESP400. I can even add most to ESP400 by just looping trough them[ instead of outputting those that your backend outputs](https://github.com/grblHAL/Plugin_WebUI/blob/c490f79580348a9b1f26e9df3dd0cff2f74a3e49/commands.c#L681-L744).

> any reason not to add the value in `$ES` ? it make a lot of queries and parsing that could at least be simplified by just adding value at last position of `$ES` line format, sender knowing $ES command should not need the `$$`, because `$$` would be only for old senders for compatibility IMHO, I will handle the multiple queries anyway

The reason is that `$ES` is used once at startup (at least in my sender) to build the settings UI. `$$` is then used to fetch the values as needed.

luc-github, 2022-07-30:
>Note that the backend supports v2.1 as well as v3.0, I use the version parameter added to the ESP800 command to detect 3.0 and set the mode for all subsequent commands (until a new ESP800 is requested):

Ok so I see the issue - I have manually sent `[ESP800]` in terminal to verify settings and that generate the problem

>I would rather like to add a ESP command for querying setting details, same format as ESP400

The ESP400 management is not part of specific FW but common to all target ESP3D ESP3DLib
if you want to merge all settings in ESP400 like format - I can remove the tab `Features` for grblHal, and  handle these settings in  grblHal tab, which give you all flexibility for the additional types, I would prefer not using [ESP400] command to avoid confusion if possible but something like `$EX` for extended settings display
@terjeio  What do you think  ?

>Yes, as it cannot be displayed for all the platforms grblHAL supports, see [this wiki page](https://github.com/grblHAL/core/wiki/For-sender-developers). I may add SSDP and/or add the IP address to the full status real-time report to fix this.

My question was because you already  display 
`[MSG:WIFI STA ACTIVE]`
so adding [MSG: IP XXX.XXX.XXX.XXX] when is possible is still better than nothing by hooking the get Ip event I think IMHO

When locked the `?` seems not working so I am not sure putting in report is necessary, additionnaly it will add extra info that should not change, and this info is already in $I

Final question : what sender currently support all/most grblHal extra features ? 
I have tried https://github.com/terjeio/ioSender/releases/download/2.0.40/ioSender.2.0.40.zip but it does not work on my Win11 and My norton AV does not like it, more over it remove each part of the sofware one by one saying there is a problem

terjeio, 2022-07-30:
>  I would prefer not using [ESP400] command to avoid confusion if possible but something like $EX for extended settings display
@terjeio What do you think ?

IMO it will be better to add a new ESP command that outputs the information in json format, possibly the same as ESP400 uses as you already have parser/UI builder code for that.

> My question was because you already display
[MSG:WIFI STA ACTIVE]
so adding [MSG: IP XXX.XXX.XXX.XXX] when is possible is still better than nothing by hooking the get Ip event I think IMHO

I can add that but there will still be a problem with some controllers that uses native USB for communication as most of these loses any info sent before a client is connected. Not easy this...

> When locked the ? seems not working

It should work - are you sure about this.

> Final question : what sender currently support all/most grblHal extra features ?

Only ioSender as far as I know.

> ... but it does not work on my Win11 and My norton AV does not like it

Is it Norton blocking it? I have not signed the code as the certificate for doing so [is expensive](https://github.com/terjeio/ioSender/issues/156).

> This is mostly to have a reference to verify I interpret properly the new states in webUI

There is only one new state `Tool` in the real-time report, `Alarm` and `Run` can have substates added if enabled in settings. These substates are added in the same way as for `Hold`. The `$G` report has some missing parser states added as well as some new ones.

luc-github, 2022-07-30:
>IMO it will be better to add a new ESP command 
if you really want to use ESP command then keep ESP400, the parser need to be modified for the new types only - the command itself is not an issue if you use json format - the parsing is piece of cake, in all case I will remove the feature tab as it is duplicate with grblHal settings

>It should work - are you sure about this.
I have reflashed to test the fixes in webUI I did  and I cannot duplicate anymore ... I am stumped...

>Is it Norton blocking it?

yes and deleting all files one by one, I have paused Norton and now everything seems working - only the SD tab seems unresponsive but it is working in webuI so - it is not a problem for me

I will check everything - thank you for all feed back and enjoy your vacations, I have everything to work now I think - this will keep me busy for a couple of days ^_^

terjeio, 2022-07-30:
>> IMO it will be better to add a new ESP command

> if you really want to use ESP command then keep ESP400, the parser need to be modified for the new types only - the command itself is not an issue if you use json format - the parsing is piece of cake, in all case I will remove the feature tab as it is duplicate with grblHal settings

Yes - I guess this makes it simpler for both me and you. The [datatypes](https://github.com/grblHAL/core/wiki/Report-extensions#code-sets-and-enumerations) not directly supported by WebUI are 2 (exclusive bitfield), 4 (axismask) and 6 (decimal or float if you will, later I might add double precision support in the core as an option...).

Examples in ioSender:

$22 - homing enable (exclusive bitfield) - can be emulated imperfecly by your `B` type:

[image]
Disabled

[image]
Enabled

$5 - invert limit pins (axismask)

[image]
IMO this can be emulated by your `B` type so not really unsupported.

> Yes I need to add the news states
[GC:G0 G54 G17 G21 G90 G94 G49 G98 G50 M5 M9 T0 F0 S0. ] for GrblHal
[GC:G0 G54 G17 G21 G90 G94 _______________M5 M9 T0 F0 S0 _] forGRBL

There are many more new ones, here is the source code in [report.c](https://github.com/grblHAL/core/blob/7dac639da81f4518bb2890d17b84a06867b96b8a/report.c#L703-L824) and motion modes supported in [gcode.h](https://github.com/grblHAL/core/blob/7dac639da81f4518bb2890d17b84a06867b96b8a/gcode.h#L62-L83).
grblHAL also supports the `G59.1`, `G59.2` and `G59.3` offsets.

If you aim for perfection mapping of error and alarm codes to text should be pulled from the controller? Note that these can also be added to by third party code.

luc-github, 2022-07-31:
>. The [datatypes](https://github.com/grblHAL/core/wiki/Report-extensions#code-sets-and-enumerations) not directly supported by WebUI are 2 (exclusive bitfield), 4 (axismask) 

ho yes mask controls are a pain - I had some before and I converted all them to simple booleans, but I will see what I can do  

>If you aim for perfection mapping of error and alarm codes to text should be pulled from the controller?

No I am not - I will keep this part for then end because anyway all text need to be in translations packs so I currently just refer to code for that

>There are many more new ones, here is the source code 

Yes saw it - thank you

terjeio, 2022-07-31:
I have committed some updates, noatbly forms based login (as used by WebUI 3.0) should now work and I have added IP address to the _WIFI STA ACTIVE_ message.

luc-github, 2022-07-31:
Ok will check that - I am still working on sanity check if not Flash is enabled - it was not a case I had planned so I am following every rat holes to be sure I did not forget anything

that bring me back to my question :

[ESP800] `File system:"directsd"` directsd is not supported should be `none` or `spiffs` or `littlefs`

Above status lead me to add a new flag : hostFS, because you mentioned you do not use Flash on some system, so the index.html.gz is hosted on SD (or USB may be in future), so WebUI update, preferences.json should be read/uploaded from/to SD/www, 
So add a flag `hostFS: "SD/www"` in ESP800 answer would tell webUI where to load/save preferences.json, what do you think ?

terjeio, 2022-07-31:
> [ESP800] File system:"directsd" directsd is not supported should be none or spiffs or littlefs

`none` is a bit counterintuitive, could `virtual` be a more logical option?

> Above status lead me to add a new flag : hostFS, because you mentioned you do not use Flash on some system, so the index.html.gz is hosted on SD (or USB may be in future), so WebUI update, preferences.json should be read/uploaded from/to SD/www,

In fact it is possible to fetch the WebUI from flash, but then from a [read only file system](https://github.com/grblHAL/Plugin_WebUI/blob/main/filedata.h). I do check if the file is available in the www folder and if so it will used instead.

> So add a flag hostFS: "SD/www" in ESP800 answer would tell webUI where to load/save preferences.json, what do you think ?

That could work, even when file system virtualization is implemented. Perhaps `hostFiles` or `hostPath` is a better name? Which file system becomes irrelevant when virtualization is implemented?

Just for the fun of I added a few lines to auto push the real time status report (along the $G-parser state report if enabled and changes to the report) and it works!

Just add this code before [webui_init()](https://github.com/grblHAL/ESP32/blob/f91fe73a730cd7a15f173915c422f654a17bd5f5/main/esp_webui/server.c#L668) if you want to test it:

```
static on_execute_realtime_ptr on_execute_realtime;

static void auto_report (sys_state_t state)
{
    static uint32_t ms = 0;

    if(hal.get_elapsed_ticks() >= ms) {
        ms = hal.get_elapsed_ticks() + 3000; //ms
        if(hal.stream.state.webui_connected)
            protocol_enqueue_realtime_command(CMD_STATUS_REPORT);
    }

    on_execute_realtime(state);
}
```
and in webui_init() add:

```
    on_execute_realtime = grbl.on_execute_realtime;
    grbl.on_execute_realtime = auto_report;
```

Change $10 to enable auto-reporting of the parser state:

[image]

If this is of interest there should be a way to set the timeout?

luc-github, 2022-07-31:
> File system:"directsd"

Well the wording is confusing now I should admit, initially it was meant for Flash only, as I did not expected Falsh was not available - I may rename it to `FlashFS` instead for clarity

>In fact it is possible to fetch the WebUI from flash, but then from a [read only file system](https://github.com/grblHAL/Plugin_WebUI/blob/main/filedata.h).

For this part it does not really matter where the original webUI is located as server already streamed it and it is loaded on client side, unless you want a read only preferences which make no sense I think - if FS is not accessible it is better to say there is no Flash FS available IMHO, it avoid user confusion and limit user interaction (and unexpected situation)

>That could work, even when file system virtualization is implemented. Perhaps hostFiles or hostPath is a better name?

I think `hostPath` is fine if on SD it would be `SD/www` as done today, indeed only location is important, not FS actually

 >If this is of interest there should be a way to set the timeout?
 well if you put a time out configurable then you can use the timeout to disable the autoreporting using value of timeout to 0 and no need to put a flag in report option, just my 2 cents, 
 another solution is no timeout but send report each time a command is received executed, or status change, but that may be more complex to implement 
 
 I will give a try tomorrow to this new version thank you 😸

luc-github, 2022-08-01:
Unfortunatly I cannot compile latest git I have raise issue https://github.com/grblHAL/ESP32/issues/40
I found some sanity check are missing but still have issues 

I have also did some test on current version to test the upload of `preferences.json` and found that any SD sub folder cannot be parsed  or even have some weird behavior listing the root content sometimes,
 
Upload to these sub directories seems working as I got `[MSG:Upload ok]` and I can download the file if I write proper path in browser, but listing is not working and this is not limited to `www` subdirectory but all subdirectory
is that normal ?

terjeio, 2022-08-01:
> I notice the entry ,"mode":"direct", what is for ?

I do not know, I guess it is from the WebUI 2 reverse engineering I did way back then. To be removed? 

> also "status":"" is empty - is that normal ? should be ok if no issue, no ?

That is a bug, fixed in the latest commit.

---

I have added the auto real-time report pushing but not yet with a run-time setting. Enable [here](https://github.com/grblHAL/ESP32/blob/9b774a3ae8c0e294970627986caac3a3c9aa4468/main/CMakeLists.txt#L302). Interval is ms.

I have also [added check for "webui" subprotocol](https://github.com/grblHAL/Plugin_networking/blob/440f9e47de4bc7ee77bb2794ab5150a05be54f05/websocketd.c#L844-L869) in the websocket code, IMO this is a more robust way to telling grblHAL that the WebUI is charge than later checking for ESP commands arriving via HTTP. Is this something that you may consider to add when connecting to the websocket? It should work with other implementations as well as the supported [subprotocols](https://www.rfc-editor.org/rfc/rfc6455.html#section-1.9) has to be reported back on connection accepted. The subprotocol could also be used for later versioning to ensure backwards compatibility.

luc-github, 2022-08-01:
> To be removed?

is it up to you , webUI does not use it, I though it was used for another webUI 

>I have also [added check for "webui" subprotocol...you may consider to add when connecting to the websocket?

I need to see the impact - any change in general  protocol / API will impact 
* ESP3D 
* embedded page of ESP3D 
* WebUI  
* ESP3DLib 
* embedded page of ESP3DLib

So I need to check carefuly

terjeio, 2022-08-01:
> is it up to you , webUI does not use it, I though it was used for another webUI

ok, I'll leave it for now.

> I need to see the impact - any change in general protocol / API will impact

Should not be any in the short term (if the backend websocket implementation is compliant with RFC6455). It is a matter of just adding the string when connecting. Later it could be used for version checking, if the backend does not support the requested subprotocol then the WebUI could gracefully stop with an error message.

luc-github, 2022-08-01:
Ok I have tested latest git and now compilling !!!! Thank you
I compiled with :
```
OPTION(WiFi "Wifi + protocols" ON)
OPTION(SDcard "SD Card Streaming" ON)
OPTION(WebUI "WebUI services" ON)
OPTION(WebAuth "WebUI authentication" OFF)
OPTION(SoftAP "Enable soft AP mode" ON) # WiFi only
```
I can see IP
`[MSG:WIFI STA ACTIVE, IP=192.168.2.108]`

But I also see a bug related to path, I saw it before but I though  the fix mentioned above will fix it also so I did not want to overflow the issue list
now can list sub directory :+1:
but subdir creation failed due to missing `/`
if I am in `www` directory and create a subdirectory `subdir`
I got `wwwsubdir` at root instead of `www/subdir`

Also I can see the [ESP400] output in incorrect for password - it is set to byte/boolean and it is sent in clear, I wrote a note that should be up to date 
[ESP3D [ESP400] format.txt](https://github.com/luc-github/ESP3D-WEBUI/files/9232286/ESP3D.ESP400.format.txt) 
if you change the `network/sta` it won't display the scan network button, this is minor issue - as it will be custom ESP400 I can change the trigger - but once defined it is better not to change until major revision is released

Sorry for path issue - it looks like rat hole issue 😓

terjeio, 2022-08-01:
> But I also see a bug related to path, I saw it before but I though the fix mentioned above will fix it also so I did not want to overflow the issue list

Odd as this is working for me - I can create subdirectories no problem. Try clean the project and recompile?

> Also I can see the [ESP400] output is incorrect for password - it is set to byte/boolean and it is sent in clear, I wrote a note that should be up to date

Will fix. I have to improve password handling elsewhere too - I want to report passwords in clear when a local client is connected and not if a remote is.

> if you change the network/sta it won't display the scan network button, this is minor issue - as it will be custom ESP400 I can change the trigger - but once defined it is better not to change until major revision is released

I'll look into this later - do not change the trigger for now?

luc-github, 2022-08-01:
>Odd as this is working for me - I can create subdirectories no problem. Try clean the project and recompile? 

I use fresh download git according https://github.com/grblHAL/ESP32#how-to-build-using-esp-idf-v43
So using webUI provided no change but config
And got issue...

Now I did: 
* idf.py clean
* idf.py build
* idf.py -p COM3 flash

and seems indeed make it work ...orz ... weird - I will keep watching, thank  you for the feedback

>I'll look into this later - do not change the trigger for now?

No need - just to confirm, I wait for your [ESP400] with all settings to work on grbHal panel, there is not need to spend time on $EG $ES $$ if you change to [ESP400] because parser is different, and need to define new type syntax 

I will test and adjust webUI for grblHal - enjoy you vaccations ^_^

terjeio, 2022-08-01:
I have just committted a fix for wrong mapping of password fields and added masking of passwords. I have also fixed the CPU frequency reported by ESP420.

As for adding support for grblHAL datatypes float (decimal) is the most important - best if you can handle formatting/input masking, I know it is not easy if not supported by the framework.
I'll try emulation for the others when I am back - I will be offline for a while...

luc-github, 2022-08-02:
>best if you can handle formatting/input masking

I can support any type so it is not an issue 
I will update the docs for float / masks and exclusive masks which should be same as previous with on additionnal flag for first bit as exclusive

enjoy your offline time ^_^

luc-github, 2022-08-02:
I have updated / wrote the :
For new types : [ESP400 format.md](
https://github.com/luc-github/ESP3D-WEBUI/blob/3.0/Memo/ESP3D%20%5BESP400%5D%20format.md)    
It should cover all of your existing extra types   

For websocket [websocket.md](https://github.com/luc-github/ESP3D-WEBUI/blob/3.0/Memo/websocket.md)

For connections values [ESP800 Format.md](https://github.com/luc-github/ESP3D-WEBUI/blob/3.0/Memo/ESP3D%20%5BESP800%5D%20format.md)

Let me know what you think 😸

dresco, 2022-08-02:
> Thanks, was actually the webui plugin that complained, may just be something minor...

Quick update, was a bumpy journey, but I seem to have WebUI 3 running well on my H7 port now..

The compilation issues turned out to be minor - some different variable names etc in the older FatFS version. But I've upgraded to the latest version anyway. That turned out to be the easy bit.. 

Was then having reliability problems with sdcard access on the Nucleo. This uncovered a new DMA/cache issue that I'd missed previously, and a separate signal integrity issue (using a breakout with the Nucleo) - which was resolved with a slower SDMMC peripheral clock.

But my biggest issue was the httpd service (mostly) failing to send data - which took me some time to debug. Long story short - I needed more Ethernet TX DMA descriptors. The way the http server builds a response with the dynamic headers, it uses up a descriptor for each header + a descriptor for the data.

Then just a final hiccup with needing the `-u _printf_float` to get the sdcard files listed in WebUI, and is now happily running through imaginary jobs on the  bench.. Nice!

luc-github, 2022-08-29:
@terjeio I see you are back - I have implemented your suggestion for websocket subprotocol in webUI V3, and wrote all docs for  [websocket](https://github.com/luc-github/ESP3D-WEBUI/blob/3.0/Memo/websocket.md)/  [ESP800](https://github.com/luc-github/ESP3D-WEBUI/blob/3.0/Memo/ESP3D%20%5BESP800%5D%20format.md)/ [ESP400](
https://github.com/luc-github/ESP3D-WEBUI/blob/3.0/Memo/ESP3D%20%5BESP400%5D%20format.md)    let me know if you need me to do anything

terjeio, 2022-08-30:
@luc-github I have been busy working on adding a linux/unix style virtual filing system (VFS) and switching to that in the networking/webui/sdcard code.  This means that "drives" are mounted as directories in the root file system - I'll have to work out how that maps to the WebUI. FYI I just commtited the first version of these changes.

I'll look into the websocket/ESP800/ESP400 changes, perhaps after I get Pico W networking up and running as this will allow me to mix filing systems in a developer friendly way (I am no fan of working with MCUs, such as the ESP32, lacking real debugging support). I'll come back to you when I have questions.

terjeio, 2022-09-05:
Pi Pico with networking and WebUI is up and running so time to move on...

* Websocket subprotocol
IMO this should be hardcoded at the client side, perhaps per target, and _not_ returned in the ESP800 response from the backend.
A minor issue is that you have [webui-v3](https://github.com/luc-github/ESP3D-WEBUI/blob/3.0/Memo/websocket.md#text-mode) in the memo and [webui-V3](https://github.com/luc-github/ESP3D-WEBUI/blob/02d5f55810adb5e42c1906fefc6c56f6c18870be/config/targets/CNC/GRBLHal/index.js#L99) in the default ESP800 responses, these should match?

* Datetimes
The datetime format in the ESP800 and ESP140 requests should be in accordance with [ISO8601](https://en.wikipedia.org/wiki/ISO_8601)?

* Versioning
This is tricky, since grblHAL (and other webui backend implementations) may lag your development pace _and_ users might try to load an incompatible version it would be nice if this could be handled in a good way. Currently grblHAL can detect the client version from the [websocket connection](https://github.com/grblHAL/Plugin_WebUI/blob/08b1c7434d7d39f6ff87d2c7194019f78b16a998/server.c#L119-L125) or from the [version parameter](https://github.com/grblHAL/Plugin_WebUI/blob/08b1c7434d7d39f6ff87d2c7194019f78b16a998/server.c#L230-L239) supplied with the ESP800 request. I use this to select either the v2 or v3 command parser and I intend to add code to inform the client if the requested version is not supported. This means that either a ESP800 command or a websocket connection (with the appropriate subprotocol) is the first thing a client should attempt. Currently this is not the case for the WebUI client:

[image]

Would this be hard to change?

* File system(s) handling
I want to simplify this in the backend, will take some time...

* Settings, float etc.
I have not started looking at this yet.

* Real time command response
grblHAL returns the streaming status in the real time command response, will it be possible to use this rather than polling the status via ESP701?
`<Run|MPos:27.907,0.648,0.000|Bf:0,1023|FS:500,150|Pn:PXYZ|SD:0.1,pcb_zebra.gcode>`

luc-github, 2022-09-05:
* the [ESP800] is not handled per target - that is why I make it flexible - but if no info is available in ESP800 we can hard code a value per target - I am just not sure to see the point of it 

* about webui-V3 where do you see it ?
in ESP3D I already implemented it and is lower case :
```
[ESP800]json
{
 "cmd": "800",
 "status": "ok",
 "data": {
  "FWVersion": "2.1.1-3.0.0.a212",
  "FWTarget": "marlin",
  "FWTargetID": "30",
  "Setup": "Enabled",
  "SDConnection": "shared",
  "SerialProtocol": "Socket",
  "Authentication": "Disabled",
  "WebCommunication": "Synchronous",
  "WebSocketIP": "192.168.2.178",
  "WebSocketSubProtocol": "webui-v3",
  "WebSocketPort": "81",
  "Hostname": "Marlinesp3d",
  "WiFiMode": "STA",
  "WebUpdate": "Enabled",
  "FlashFileSystem": "LittleFS",
  "HostPath": "/",
  "Time": "none"
 }
}
``` 

* Datetimes 
About format I used the simplest one YYYY-MM-DD-HH-mm-ss  as for current usage I did not saw the need of use a formated one especially time is not keept currently when CNC/printer is of
And in WebUI V3 I did not implemented it yet to see if it was actually missing to someone - and got 0 feedback so may be this option is useless and should be removed ?

* About versioning it is your choice what you want to use, but  to be honest I do not understand why need to be compatible with 2.1 especially that there are a lot of differences in commands, output format and web handle between both version it is a 3.0 because it break compatibility with 2.1 it will be double work

* Real time status
Yes of course, this is how grbl_esp32 do, as mentioned before I did not understood why you used [ESP701] which is supposed for Flash streaming not SD in ESP3D but in case of Grbl_HAL which has the SD and is the consummer all stream should go SD stream using  CNC Gcode command for hold / resume / stop IMHO
implementing parsing of SD:0.1, pcb_zebra.gcode is not hard, 0.1 is progress on base 100 (0.1%) or base 1 (10%) ?

luc-github, 2022-09-05:
Also I have wrote more docs FYR : https://github.com/luc-github/ESP3D-WEBUI/blob/3.0/Memo/Handlers.md
https://github.com/luc-github/ESP3D-WEBUI/blob/3.0/Memo/Commands.md 
 to clarify the API used by WebUI 3.0

terjeio, 2022-09-06:
> the [ESP800] is not handled per target - that is why I make it flexible - but if no info is available in ESP800 we can hard code a value per target - I am just not sure to see the point of it

The point is that the client announces what protocol version it wants to talk. If set via ESP800 then the backend tells itself via the client - this kind of defeats the purpose of having it.

> about webui-V3 where do you see it ?

[Here](https://github.com/luc-github/ESP3D-WEBUI/blob/02d5f55810adb5e42c1906fefc6c56f6c18870be/config/targets/CNC/GRBLHal/index.js#L99) is one example.

> About format I used the simplest one YYYY-MM-DD-HH-mm-ss

You could have chosen ISO88601 formatting which is slightly different, e.g. `2022-09-06T20:17:16`. Not a big deal, I convert these to ISO8601 format by replacing the separators with the correct ones.

> About versioning it is your choice what you want to use

I try to keep backwards compatibility when possible, there is at least one more webclient that I know of and not supporting 2.1 may break that.

> implementing parsing of SD:0.1, pcb_zebra.gcode is not hard, 0.1 is progress on base 100 (0.1%) or base 1 (10%) ?

Great, the base is 100 (0.1%). grblHAL can also be configured to output the real time report automatically, currently as a compile time option. This can be used to disable polling via http requests. Is this of interest?

> Also I have wrote more docs

Nice,

luc-github, 2022-09-06:
>The point is that the client announces what protocol version it wants to talk. If set via ESP800 then the backend tells itself via the client - this kind of defeats the purpose of having it.
Yeah seems I took problem upside down - no issue I plan to do several update before the week end - I can remove the subprotocol from [ESP800] and hardcode it in webUI - it will also simplify code on FW side

>[Here](https://github.com/luc-github/ESP3D-WEBUI/blob/02d5f55810adb5e42c1906fefc6c56f6c18870be/config/targets/CNC/GRBLHal/index.js#L99) is one example.
Ho that is test server and seems ok - but I will update it as mentioned above - so it will be removed

>2022-09-06T20:17:16
To be honest I just put `-` because I though will be easier to parse...but looking at code now I do not see the benefit...
 as I will update I can follow the format you mention - to be honest did not do any research on ISO format when I implemented it
 
 >I try to keep backwards compatibility when possible
 Ok 

>This can be used to disable polling via http requests. Is this of interest?
Yes web UI parse all answers even no polling command to handle auto reporting, auto reporting is better if well managed on FW side , like if it does not pop in another running answer

terjeio, 2022-09-07:
> to be honest did not do any research on ISO format when I implemented it

Well, you were pretty close so you must have seen it before. I have created a [c function for parsing](https://github.com/grblHAL/core/blob/180f9fa9fcf258c2c09de21ea61b2ffb207a5d02/nuts_bolts.c#L268-L326) the string that you may use if you want.

> Yes web UI parse all answers even no polling command to handle auto reporting, auto reporting is better if well managed on FW side , like if it does not pop in another running answer

Yes, I saw that, even the `$G` response is handled when sent automatically (grblHAL can be configured to send it on changes).

From your [commands memo](https://github.com/luc-github/ESP3D-WEBUI/blob/3.0/Memo/Commands.md) it looks like json output is now the default, is this correct?
Also many commands has `pwd` as a parameter. How/when is this intended to be used? When not logged in (no cookie present) and authentication is enabled?

luc-github, 2022-09-07:
>I have created a [c function for parsing](https://github.com/grblHAL/core/blob/180f9fa9fcf258c2c09de21ea61b2ffb207a5d02/nuts_bolts.c#L268-L326) the string
Thank you I will have a look
 
> it looks like json output is now the default
The webUI only use json yes, in ESP3D default output is plain text because if you type command in Serial/Telnet it for human reading and add `json=yes` to the command change output to json  which is formated response for webUI, only ESP400 does not have plain text because I was lazy to do it as the plain text would not be really human readable anyway 
Yes: the `pwd=admin/user` parameter is only for serial/telnet when authentication is enabled, http use the cookie once authenticated instead
That said for serial ESP3D has 2 anothers settings :
- it can disable the serial commands totaly (it is a define) 
- it can disable serial authentication need (a setting in EEPROM) because if you have access physicaly to the board, passwords can be modified just by flash, so for convenience I got request a way to disable serial authentication, when keeping remote access authenthication

terjeio, 2022-09-11:
I've just committed an update for grblHAL with [changes](https://github.com/grblHAL/core/blob/master/changelog.md) mostly for WebUI v3 support.

* ESP400 now outputs all grblHAL settings, I emulate bitfields and axismasks via `B` type settings - this may change later when native support is available in the WebUI client.

* Flash filing system implemented by using [littlefs](https://github.com/littlefs-project/littlefs) for ESP32, RP2040 and iMXRT1020 drivers. I chose littlefs as a standard flash filing system for grblHAL since the lowlevel driver code is fairly simple if the target MCU has an API for handling flash write/read/erase, and also since it was faily easy to add last modified datestamp to files in the [VFS wrapper](https://github.com/grblHAL/Plugin_SD_card/blob/master/fs_littlefs.c).
@dresco you may consider adding this to the STM32H7xx driver - there is ample flash available for that(?) and only a smallish low-level driver is needed, here is the one used for [RP2040](https://github.com/grblHAL/RP2040/blob/master/littlefs_hal.c). 

*  Improved search across mounted file systems (or drives) for _index.html.gz_, _favicon.ico_, _preferences.json_ and, for WiFi builds, _ap_login.html_. Added support for `http://<ip>/?forcefallback=yes`, this will attempt to load [index.html.gz](https://github.com/luc-github/ESP3D/tree/3.0/embedded/server/public) from the VFS _/embedded_ folder/mount. Note that this currently fails due to the client not understanding the ESP800 data provided by the backend. I see that the ESP800 request does not add `json=yes` parameter, should the response be in plain text or json? If I return json I still get an error...

A question: is it per design that it is not possible to stream gcode files from flashfs? (flashfs is mapped to a littlefs mount directory in grblHAL).

luc-github, 2022-09-11:
>ESP400 now outputs all grblHAL settings, I emulate bitfields and axismasks via B type settings - this may change later when native support is available in the WebUI client   

bit maks is support is ready but not implemented in ESP400 I will add it soon 

>Flash filing system implemented by using [littlefs](https://github.com/littlefs-project/littlefs)

This the prefered choice also in ESP3D/ESP3DLib

>I see that the ESP800 request does not add json=yes

all ESP command used by web UI need json commands 
[image]
You may use an old version of webUI  if json no used 

>A question: is it per design that it is not possible to stream gcode files from flashfs? (flashfs is mapped to a littlefs mount directory in grblHAL).
Yes it is possible, I do not encourage this because of flash wearing and  small size of flash, I can add it if you really want but I am confused I though you hide the flash  partition

luc-github, 2022-09-12:
@terjeio 
I have do a fresh download of everything  and cannot compile :
```
no-stringop-truncation -MD -MT esp-idf/main/CMakeFiles/__idf_main.dir/webui/server.c.obj -MF esp-idf\main\CMakeFiles\__idf_main.dir\webui\server.c.obj.d -o esp-idf/main/CMakeFiles/__idf_main.dir/webui/server.c.obj   -c ../main/webui/server.c
../main/webui/server.c: In function 'wifi_scan_handler':
../main/webui/server.c:424:32: error: 'file' undeclared (first use in this function); did you mean 'pipe'?
                 vfs_puts(resp, file);
                                ^~~~
                                pipe
../main/webui/server.c:424:32: note: each undeclared identifier is reported only once for each function it appears in
../main/webui/server.c: In function 'wifi_login_receive_finished':
../main/webui/server.c:489:31: error: 'file' undeclared (first use in this function); did you mean 'pipe'?
     vfs_puts("Connecting...", file);
                               ^~~~
                               pipe
[1329/1334] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/webui/commands_v3.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

C:\Users\luc\Documents\GRBLHal\esp\esp-idf\ESP32>
```

terjeio, 2022-09-12:
>> I see that the ESP800 request does not add json=yes

> You may use an old version of webUI if json=yes is no used

The issue is not with the WebUI but with the [embedded](https://github.com/luc-github/ESP3D/tree/3.0/embedded/server/public) code which I assume is to be returned to the client if the WebUI code is not found. It was last updated 2 years ago, perhaps that is the reason?

> Yes it is possible, I do not encourage this because of flash wearing and small size of flash, I can add it if you really want but I am confused I though you hide the flash partition

I no longer do, but I may doing something wrong since it shows up on client startup:

[image]

> GrblHal can have custom Flash it is possible, currently all targets use same but I can add custom one for GRBHal like for SD

Thanks - I remember now, will look at this again. I believe I managed to hide _Flash_ and make _SD_ the default a while back.

> The webUI is flexible ^_^

Yes I know - but it takes time to understand your code, IMO it is a large and complex project.

> ... and cannot compile :

I have broken AP mode, sorry about that. A quick fix is to replace `file` with `response` in line 424 and 489.

luc-github, 2022-09-12:
>It was last updated 2 years ago, perhaps that is the reason?

ho yes the embedded page code follow the webUI one -so if it is 2 years old it is definitly outdated -  Ho boy I just realized wht you did you took the one in server/public which is a test server for upload not the embedded page itself
I have just removed from ignore the embedded dist directory - but it should be modified for your FW because the links go to esp3d repository and webui repository - I may suggest to do a copy of the embedded and change the menu.js as well as the favicon and build scripts as you do not need to convert binary to c files

>Yes I know - but it takes time to understand your code, IMO it is a large and complex project.

Ho yes that is why no one to handle it ^_^ but yours is not bad neither supporting many platforms is also a challenge

>I have broken AP mode, sorry about that. A quick fix is to replace file with response in line 424 and 489

Ok will check that thank you - no worry

terjeio, 2022-09-12:
> I have just removed from ignore the embedded dist directory 

Thanks, now it works! For testing you can replace [main/index.html.gz](https://github.com/grblHAL/ESP32/tree/master/main) with the one from dist and recompile/reflash.

> ... as you do not need to convert binary to c files

I have to for platforms other than ESP32, but I'll do it myself as I use a different format. 

> I should make it as project to allow multiple targets but until now was only 2 so I did not saw the need ...
I guess I could make a repo for it and it could become a submodule for each project

Perhaps so, I would guess this code is fairly static so not a high priority? And for the time beeing I am not going to support OTA firmware upgrades.

> ... supporting many platforms is also a challenge

Yes, abstraction and architecture are the magic words. What I need is a web frontend for building binaries for the various platforms/configurations supported. Maybe I can get some ideas from your .js code?

A minor issue, my code reports an error on the _/files_ request:

[image]

It looks like _filename_ used for filtering when action is _list_, aside from _all_ may I ask what kind of filtering can be specified and what is the default?

luc-github, 2022-09-12:
>I have to for platforms other than ESP32, but I'll do it myself as I use a different format.

The script to convert binary to c is a js so it is easy to modify - I wanted to avoid external application to be platform independant

>Perhaps so, I would guess this code is fairly static so not a high priority? And for the time beeing I am not going to support OTA firmware upgrades.

Actually it display according esp800 - so if not OTA then ota tab won't show, same for filesystem, so only menu is actually static

>Yes, abstraction and architecture are the magic words. What I need is a web frontend for building binaries for the various platforms/configurations supported. Maybe I can get some ideas from your .js code?

You may want to have a look to https://github.com/luc-github/ESP3D-Configurator it is a simple front end using json file for configuration - it work online and offline, I can imagine you can use the github-action to generate assets / binaries based on the front end configuration

>A minor issue, my code reports an error on the /files request:
 
To be honest is a fake url it is same as /files?path=..., it is not even mentionned in https://github.com/luc-github/ESP3D-WEBUI/blob/3.0/Memo/Handlers.md#files , I just did it to have consistent url when I read the devt panel, but actually it is not used - you can safely ignore filename when action is list

terjeio, 2022-09-12:
> The script to convert binary to c is a js so it is easy to modify

Found it now - yes, easy to modify for grblHAL.
But why is the generated file called _tools.html.gz_ and not _index.html.gz_? Something to do with the webserver? For grblHAL the latter name is easiest to handle.

> you can safely ignore filename when action is list

I have just commited a fix for this and the bugs you reported. I have also updated the embedded file to point to the grblHAL repo.

Thanks for the tip about the configurator, will take a look later.

luc-github, 2022-09-12:
tool.html.gz is the esp3d v2.1 version, in v3 i use same api as webpack which is index.html

the name has no meaning for the embedded part because is not a file but http answer stored on flash

dresco, 2022-09-12:
> @dresco you may consider adding this to the STM32H7xx driver - there is ample flash available for that(?) and only a smallish low-level driver is needed

Cool thanks, I'll take a look when I have time. The H743 'only' has 2MB flash, but can use QSPI flash chips (and the WeAct board already has two on it - so may be good place to start testing)..

luc-github, 2022-09-13:
@terjeio I have updated sources with latest one and so far so good compilation is fine

- I have fixed a bug in webUI for the Web Update not actually disabled even it is defined in [ESP800]
- I have renabled the feature tab and use a logo for grblHal I did myself in svg to replace the esp3d one, feel free to give me another one if it does not fit your graphical specs
- I have added a detection for the SSID button in features tab to launch the wifi scan but it is seems not working/supported.
 Do want me to disable this feature for grblhal ? or do you plan to support `[ESP410]json=yes` later (I have added the expected format in Memo FYR)

About ESP400 output:
- Do you plan to keep the emulation of bitmask to keep 2.1 compatibility support ? Just to know if it worst to add bitmask native  support in V3 if you keep emulation for 2.1

- I have also noticed you use different method than me for the `single page support`, your implementation block any new connection, when mine allow any new connection but close old one , in that case you may need to change the menu in embedded page to open webUI in same page instead of new one that will be rejected by your way, you may also rename ESP3D link name to WebUi for consistency, is it up to you

Here a screen cast  
[image]

You may want to re-arrange some sections now the rendering is visible

Let me know what you think and feedback/request any change you feel necessary ^_^

terjeio, 2022-09-13:
> I have renabled the feature tab and use a logo for grblHal I did myself in svg to replace the esp3d one, feel free to give me another one if it does not fit your graphical specs

Great - I have to decide if this is the one to use. [Here are some other proposals](https://github.com/terjeio/grblHAL/discussions/268). What do you think?

> I have added a detection for the SSID button in features tab to launch the wifi scan but it is seems not working/supported.

Bad code, will fixed in the next commit.

> Do you plan to keep the emulation of bitmask to keep 2.1 compatibility support ? Just to know if it worst to add bitmask native support in V3 if you keep emulation for 2.1

v2 and v3 is now handled in separate files so can differ. I can keep emulation for v3 if you think the rendering looks ok. What is missing is the exclusive bitmap where bit 0 controls enable of the rest, could this be handled via emulation and a new flag to control the behavoiur?

> I have also noticed you use different method than me for the single page support, your implementation block any new connection, when mine allow any new connection but close old one

This is something I will look into, easier now that the websocket subprotocol is in place. I want the [messages](https://github.com/luc-github/ESP3D-WEBUI/blob/3.0/Memo/websocket.md#text-mode) to be transparent to the core - to keep normal streaming available. I have tried to implement the PING response but I cannot get it work, it would be helpful if the bracketed elements was explained - not all are obvious. E.g. what is the datatype and expected value ranges for the `<time left>` and `<time out>` elements? To be compliant with websocket terminology perhaps the `PING` response should be `PONG`...

> You may want to re-arrange some sections now the rendering is visible

It is on my todo list to add sorting, perhaps to match[ ioSender](https://github.com/terjeio/ioSender).

> Let me know what you think and feedback/request any change you feel necessary ^_^

I believe we are getting there now, basic functionality seems robust to me - not too many rough edges left.

luc-github, 2022-09-14:
>Great - I have to decide if this is the one to use. https://github.com/terjeio/grblHAL/discussions/268. What do you think?

Thay are all nice - you may also consider the logo should be readable at lower resolution like in screenshoot, and if not, you may consider a variant for the favicon that is consistent with logo IMHO, but I am not good in design and logo is something you must feel confortable with.
[image]

>Bad code, will fixed in the next commit.

Ok so I do not change anything

>v2 and v3 is now handled in separate files so can differ. 
Ok so I can work on implement native version then - could you generate an [ESP400] with native settings for bitmask and exclusive bitmask for me to work on final implementation ?
as described in https://github.com/luc-github/ESP3D-WEBUI/blob/3.0/Memo/ESP3D%20%5BESP400%5D%20format.md the types are :
    -   M: for bits mask (only grblHal)
    -   X: for exclusive bitsfield (only grblHal)
the descrptions should be defined using -   O: is an array of {label:value} used for possible values in selection or bits labels list

> it would be helpful if the bracketed elements was explained - not all are obvious

Ok sure I will complete this part and also error messages code and messages

<time left> and <time out> are for authentication <time out> is the time of the session in millisecond, <time left> is the time left in current session each time there is an authenticated http query the value should be reset and so same as <time out>
if <time left is 0> the web UI disconnect as no authenticated anymore, if <time out is 0> there is not time out = session never end.

>I believe we are getting there now, basic functionality seems robust to me - not too many rough edges left.

Cool ^_^ I will continue to improve as described above and also add support of the SD stream from `?` report, it was not done before as I did not saw it in `https://github.com/grblHAL/core/wiki/For-sender-developers` 😉

[59 more replies]

closed 2022-09-22 by luc-github