luc-github/ESP3D-WEBUI #255: Click or Touch to move the spindle/Laser

mathojojo, 2022-08-07:
**Is your feature request related to a problem? Please describe.**
Moving the spindle or the laser at a position can require many jog operations, or to calculate the new desired position and enter it as a G0 or G1 command.
My feature request is about a graphical way to roughly move the spindle at a X and Y position (Not sure if Z is needed), and then only, adjusting it with jog commands.

**Describe the solution you'd like**
For that, it could be nice to have a 2D representation of the  bed, with : 
   - the working area as a square or rectangle,
   - the Home Position as a dot,
   - the actual spindle position as a dot or an icon,
   - the actual space origin (G54, G55, ...G59),
   - Rulers for X and Y in mm or inch (should be switchable).

You just have to click with the mouse pointer somewhere on the bed to move the spindle at this position (or touch it if touchscreen/smartphone/tablet).

It might not be so usefull for little machines like 3D printers, but for big milling CNC, it would be very appreciated.
I personnaly have a 1500mm * 1500mm milling CNC, and moving the spindle at the right position is a big part of the job...

It's integration can be done with a plugin, or added as a new tool in the V3 WebUI. Or maybe a button in the Jog space could be added, when clicked a popup appears with the 2D representation ??

**Describe alternatives you've considered**
The alternative is to do as I do actually, Jog, and jog, and jog and jog :)

**Additional context**
I hope this is not too much work, and that i'm not the only one to considere this feature as Needed :)
Thank you.

luc-github, 2022-08-08:
Interresting, so you do not want to type a command in terminal  like  `$J=G0 X500 Y500 F1000` but would prefer a  graphical way ?
Am I correct ?

O I have questions as I am still a newbie in CNC, sorry if they look obvious for you but many GCODE cnc technics are still fuzzy for me:
* the working area as a square or rectangle, 
[Question] How you get that ?

* the Home Position as a dot, 
[Question] The 0.0 of the total machine area?  - or the 0,0 of the current working area ?
* the actual spindle position as a dot or an icon,
Ok

* the actual space origin (G54, G55, ...G59),
Ok that is already displayed

* Rulers for X and Y in mm or inch (should be switchable).
[Question]This is done at start right, you do not switch when jogging ?

 * in case of displaying the full machine area, in your case 1500 X1500 according $130/$131
 May you explain to me the commands to use to switch to global space do the moving to X, Y then swich back to previous space origin (G54, G55, ...G59), I still have hard time to understand how to use  and switch between these G54, G55, ...G59

mathojojo, 2022-08-08:
Thank you for your answer,

It's not that I don't want to type commands, but it would be an interesting option that can help to win time. It can also reduce errors when typing numbers (Keyboards keys become unstable in dust environment , G0 X500 Y500 F1000 can fast become G0 X5000 Y500).
And the 2D representation is probably a way to better apprehend the CNC global state, especially for beginners.
Of course, for precise positionnement, jogs will always be the only solution.

- The working area (rectangle) is determined with : 
      * WPos X and WPos Y after Homing Cycle (maybe should integrate the $27 parameter, Homing pull-off),
      * X and Y Max travel ($130 and $131)

 [image].
[image]

 - Forget about Home Position as a dot, my mistake. I was in fact talking about  Actual Space Origins (G54 to G59). If the CNC is working in G54 space (default), then the G54 Space Origin should be drawn as a dot. If you switch your machine to G56, then, draw the G56 Space Origin as a dot..

- Actual Spindle position : OK

- Rulers for X an Y in mm or inch. You are right. If mm is used in the global interface, rulers should be in mm too, for coherence.

`in case of displaying the full machine area, in your case 1500 X1500 according $130/$131
May you explain to me the commands to use to switch to global space do the moving to X, Y then swich back to previous space origin (G54, G55, ...G59), I still have hard time to understand how to use and switch between these G54, G55, ...G59`

I'm not very advanced in CNC for now. I never used a global Space , I'm not even sure if such space exists.I'm mostly working on the default G54.

But, you can probably do it with a calculation.

In my case with 1500 x 1500mm machine :
If the Homing Pull-Off ($27) is 3mm, then $130 an $131 values are 1497mm. After Homing Cycle :
=> The global space drawn should be 1497 long, and 1497 width
=> The Top Right corner of the rectangle has position WPos X - 3 = 0 - 3 = -3mm  /  WPos Y - 3 = 0 - 3 = -3mm
=> The bottom Left corner of the rectangle has position X = -1500 Y = -1500

WPos X and WPosY are determinated with the Home cycle, it won't change whatever the space used (G54 to G59). So moves should be calculated with WPos, not with MPos.

So, if your spindle has its actual position at WPosX-300 and WPosY-200 on global space (MPosX100 MPosY100 in G54 area), and you click at the X-800 Y-700 position of the 2D representation, then the move should be : 
Xmove : WPosX Click - WPosX Spindle = (-800) - (-300) = -500
Ymove : WPosY Click - WPosY Spindle = (-700) - (-200) = -500

The easiest way would be then to do a relative move : 
G91
G0 X-500 Y-500
(of course, return to absolut mode if it was the initial state before the move by adding G91 at the end)

[image]

luc-github, 2022-08-08:
Ok thank you for the clarification, the precision would be very limited on small screen, and on phone/tablet you won't have mouse cursor to limit the click area, that said it is possible to be done, but I cannot say when it could be done, it would be linited to X,Y plan anyway, because  I do not see an easy way to do handle this in 3D

mathojojo, 2022-08-08:
Yes you are true, the precision would be very limited on small screen. But even on big screen it would be limited. It can't avoid a precise adjustement.

But with command method (G0 X-800 Y-700), you must also finalize your placement with some precision jogs.
And when you make multiple 100mm jogs, you must finalize the process with some precise jogs as well.

The goal is really to make a good enough first placement of the spindle... then adjust. On big machines, and when you have many operations to do, you can save many time.

Thank you

luc-github, 2024-06-02:
This feature is definitly a good candidate for an ESP3D-WebUI extension.

luc-github, 2024-06-13:
Working on it and improving extensions API during development
UI is already working and support of full screen allow to improve user experience 
[image]

luc-github, 2024-06-22:
The extension is done, it is currently  not specific to cnc and can be used by 3D printer too
it  is  1.0.0 - autodectection of target system coul be added easily 
The extension need user to enter the workspace information: xmin/xmax/ymin/ymax , the command for moving , and some preferences for the rendering.
Once done extensions can be run in panel : reduced or fullscreen) or in dedicated page  (normal or full screen)
[image]

The location of extension repository is here : https://github.com/luc-github/ESP3D-WEBUI/tree/3.0/extensions/clicktogo
and prebuild package is here : https://github.com/luc-github/ESP3D-WEBUI/tree/3.0/extensions/clicktogo/dist

the current package size is 4,000 bytes so 3.9KB

Feel free to test / feedback or even more push improvement PR ^_^

closed 2024-06-22 by luc-github