luc-github/ESP3D-WEBUI #256: [Question]What are the features needed for the Sand Table UI

luc-github, 2022-08-08:
Sand table may need specific UI but I do not know which one :
currently UI display
* Settings
* Jog panel
* Files panel 
* Macro panel
* Extra panels

Others CNC panels have been removed because not relevant to Sand Table per my understanding, but I may be wrong ?

Is there any specific need for Sand Table system ? I do not have any currently but I plan to have one, so feedback are more than welcome.

geekuillaume, 2022-08-28:
+1

I have a sand table and only use these panels, no need for any extruders related settings. Another potential features that could be awesome is the ability to queue gcodes files to launch new drawings after the current one is finished but this would likely be a feature request for the lib instead of the webui.

luc-github, 2022-08-28:
what FW are you using ?

geekuillaume, 2022-08-28:
I've switch back and forth between GRBL and Marlin

luc-github, 2022-08-29:
@geekuillaume yes I also though the play list feature is definitly a must for sand table