luc-github/ESP3D-WEBUI #257: Keyboard shortcuts

mathojojo, 2022-08-08:
**Is your feature request related to a problem? Please describe.**
All powerful interface, with advanced features have Keyboard shortcuts that allows better productivity. 

**Describe the solution you'd like**
It would be perfect, if you could set some shortcuts on the WebUI v3 so you can do most common operations with ease.

Example :
Y + KEY_DOWN : Lower the spindle
Y + KEY_UP : raise the spindle
Z + KEY_DOWN : move spindle ahead
Z + KEY_UP : move spindle backward
X + KEY_LEFT : move spindle to left
X + KEY_RIGHT : move spindle to right

CTRL + H : Homing Cycle

CTRL + A : Absolut mode (G90)
CTRL + R : Relative mode (G91)

CTRL + S : Switch between spaces (G54 > G55 > G56 > G57 > G58 > G59 > G54 ...)

CTRL + ESCAPE : Emergency STOP
CTRL + ALT + ESCAPE : Reset
CTRL + SPACE : Pause
CTRL + U : Unlock

CTRL + KEY+ : Increase Spindle Rotation (+1000) -> Should not be able to increase above the Max value
CTRL + KEY- : Decrease Spindle Rotation speed (-1000)-> Should not be able to decrease under the Min value

CTRL + M : Switch from M5 to M3 or from M3 to M5 or from M4 to M5
CTRL + ALT + M : Switch from M5 to M4 or from M4 to M5 or from M3 to M5

It could then even be possible to map thous shortkeys with some gamepads (PS3, XBOX, Nunchuck...) on the client station for more ergonomy and versatility.

**Describe alternatives you've considered**
Nothing to report

**Additional context**

Sorry, I'm giving you work with ergonomy stuff. Ergonomy is very important, but not vital :)

luc-github, 2022-08-08:
currently only Jog panel support shortcuts, and they are simple ones.

Having shortcut for all features is only a  nice to have  because it only usable on PC interface.

Current shortcut support is basic, adding the ones you describe needs major rewrite which is not for short time due current workload

mathojojo, 2022-08-08:
I hope I will soon have WebUI v3 on my xPro V5 to test all of this.
Yes shortcuts are for PC interface, I suppose it is most used interface. But I may be wrong.
I 'm totally aware that it must be a lot of work, and I didn't expected it for short time. As I said, it's only ergonomy, not vital, so not urgent ! :)

Thank you

luc-github, 2022-08-08:
I think grblHal already support xproV5, there is basic support of webui 3.0 already, in coming weeks it should be finalized I hope

aaronse, 2023-01-01:
Recent work in https://github.com/luc-github/ESP3D-WEBUI/tree/3.0-keyboard-remaping may help with this ask.

Given work already done by Luc to enable configurable shortcuts, maybe some additional edits enabling key shortcuts for Macros would enable most/all of the scenarios @mathojojo is looking for.

aaronse, 2023-01-02:
@luc-github  https://github.com/luc-github/ESP3D-WEBUI/commit/ba292f09c8a55b5634aab9c0ee29bb947cfbecce already, nice!

closed 2023-04-23 by luc-github

edrex, 2023-05-02:
Merged via https://github.com/luc-github/ESP3D-WEBUI/pull/288

edrex, 2023-05-02:
> currently only Jog panel support shortcuts, and they are simple ones.

docs / source for the 2.x jog shortcuts?

luc-github, 2023-05-02:
they are implemznted in 3.0 not 2.x