luc-github/ESP3D-WEBUI #261: web control MKS Tinybee

vasiliyz1979, 2022-08-29:
Hello. After successfully connecting to the access point of the motherboard, the browser does not automatically open a window to download the file that controls the printer, but opens an empty window. Can you tell me what the problem is? How to reset to factory settings.

luc-github, 2022-08-29:
type `http://192.168.0.1?forcefallback=yes` in your browser
 what do you see ?

vasiliyz1979, 2022-08-29:
Да. Я успешно подключаюсь по вай фай к плате. В мануале написано, что после подключения к плате при открытии браузера для загрузки файла который управляет принтером, браузер автоматически откроет окно для загрузки файла, но браузер открывает окно пустое.

luc-github, 2022-08-29:
type `http://192.168.0.1?forcefallback=yes` in your browser
Do a screenshot please

vasiliyz1979, 2022-08-29:
Yes, thank you. I'll do it now at work, I'll come home and do it. Thank you so much for responding.

vasiliyz1979, 2022-08-30:
What you wrote redirected to the file download screen. Everything worked out thanks.

closed 2022-08-30 by vasiliyz1979