luc-github/ESP3D-WEBUI #262: [FEATURE REQUEST] Swapping (or rotating) X and Y on the Jog panel.

ImagineerNL, 2022-09-03:
**Is your feature request related to a problem? Please describe.**
The Jog UI defines X -+ on the Horizontal axis, the Y on the vertical axis.
Since i'm always on the side of my CNC, the X is visually for me up/down (away from me / towards me) and Y is left / right.
I have often misjudged my jogging direction due to this.

**Describe the solution you'd like**
It would be great if the X and Y can be visually swapped or rotated to match the orientation of the user to the machine

**Describe alternatives you've considered**
Alternative to switch the Axis motors is unfeasible since i have Dual Y axis, which cant be translated to Dual X axis.

To be clear; the issue is not with a wrong direction where plus down and min is up; it is purely visual for X and Y to swap

luc-github, 2022-09-03:
I guess adding a setting to invert +/- for x  and for y should be welcome especially to be consistent in rendering  for https://github.com/luc-github/ESP3D-WEBUI/issues/255

What FW do you use for your cnc ? what version of webui do you use currently?

ImagineerNL, 2022-09-03:
thanks for the quick reply!

I am Using the xpro V5
With this version of the index file: 
https://github.com/Spark-Concepts/xPro-V5/blob/main/Firmware/index.html.gz

luc-github, 2022-09-03:
ho this one is based on 2.1 and unfortunatly the webui 3.0 which allow ui change is not compatible because  it need several change on fw side, 
here the cnc ui for your reference
https://github.com/luc-github/ESP3D-WEBUI/discussions/94#discussioncomment-3145882

and features list https://github.com/luc-github/ESP3D-WEBUI/blob/3.0/Features.md

so you may ask spark-concepts if they plan to support / use webui-v3.0

ImagineerNL, 2022-09-03:
Sorry, i was under the impression tbe issue/FR was created for that version. 

Would it perhaps be possible using CSS to rotate/relocate?

theres already an issue on the xPro github asking for update to WebUI v3, but they state its not final enough to support. 

Might help them if you give them a small update on that issue: https://github.com/Spark-Concepts/xPro-V5/issues/166

luc-github, 2022-09-03:
no 2.1 jog ui is an svg and not easy to update, that is why I rewrote to be the 3.0

the index.html.gz with tablet Ui is not from me but it is compatible with 2.1

I will check the issue you linked thank you

ImagineerNL, 2022-09-04:
@luc-github thank you.

Not sure if i should close this issue or not; doesnt and wont work for V2, but it remains a valid FR for V3, although not 'for me' when Spark-Concepts choses not to update to V3...

i'll leave it to you to decide on it.

luc-github, 2022-10-11:
POC of invert X and Invert Y  and swap XY has be done in HP-GL webui 
for CNC/sandtable the current UI does not need swap xy, but instead need Invert Z => To be done
for 3D Printer Jog ui is splited in 2 the printerface UI for desktop mode and CNC UI for Mobile mode => To be done

luc-github, 2023-04-23:
Implemented

closed 2023-04-23 by luc-github