luc-github/ESP3D-WEBUI #263: MKS dlc32 webui

kweilim98, 2022-09-11:
Hi Luc , may I ask does I able not to use the original MKS webui by change to the lasted webui that you have created? Because the interface lack of some features for me as CNC user

luc-github, 2022-09-11:
the webui v2.1 is compatible with MKS FW I think, the webUI V3 need several FW change and so it is not compatible sorry

kweilim98, 2022-09-11:
> the webui v2.1 is compatible with MKS FW I think, the webUI V3 need several FW change and so it is not compatible sorry

Is it have any other options?

luc-github, 2022-09-11:
use another FW and modify it to support WebUi V3 that is the only way I think

kweilim98, 2022-09-11:
> use another FW and modify it to support WebUi V3 that is the only way I think

if I want to change the webui interface from the original, how to change it or upload it?

kweilim98, 2022-09-11:
> use another FW and modify it to support WebUi V3 that is the only way I think

Can you list down what should be change to merge the v3.0 FW?

luc-github, 2022-09-11:
the WebUI v3 has almost all informations in memo directory, I have tried to describe what its expect to communicate
check https://github.com/luc-github/ESP3D-WEBUI/tree/3.0/Memo

luc-github, 2022-09-11:
to upload use embedded page 
http://your ip?forcefallback=yes

kweilim98, 2022-09-11:
Ok, thx u I will try it

kweilim98, 2022-09-11:
Can u have a screenshot for the CNC webui interface here? Wanna to see what features have in the lasted v3

luc-github, 2022-09-11:
you also can check grblhal I think it support mks dlc32 and we are working to make it having webui v3

luc-github, 2022-09-11:
https://github.com/luc-github/ESP3D-WEBUI/discussions/94#discussioncomment-3145882
all screen are in this thread

kweilim98, 2022-09-11:
> [#94 (comment)](https://github.com/luc-github/ESP3D-WEBUI/discussions/94#discussioncomment-3145882) all screens are in this thread

Thx u

kweilim98, 2022-09-11:
Does the v2.1.2 FW able to support the MKS dlc32 webui?

kweilim98, 2022-09-11:
Actually what difference between the v2.1.2 compare to v3 CNC webui? Is it miss some interface features or what?

luc-github, 2022-09-11:
>Does the v2.1.2 FW able to support the MKS dlc32 webui?

No idea sorry

>Actually what difference between the v2.1.2 compare to v3 CNC webui? Is it miss some interface features or what?

the WebUI 2.1.x for CNC is really light compare the 3.x :  https://github.com/bdring/Grbl_Esp32/wiki/ESP3D-Web-UI-for-Grbl_ESP32

luc-github, 2022-09-11:
you may search on internet, I am sure other people did other webUI  2.1 compatible

luc-github, 2022-09-11:
issue is nnow converted to Q/A in discussion pannel as it is not a bug but question