luc-github/ESP3D-WEBUI #265: [FEATURE REQUEST]Integrate the [ ESP700] command into the button

liqijian101, 2022-09-18:
Integrate the [ ESP700] command into the S button so you can start file printing directly from the button in GCODE HOST mode

luc-github, 2022-09-18:
what is the S button ?

liqijian101, 2022-09-18:
> what is the S button ?

Sorry, this is my spelling mistake. It has been corrected. It's just a button, like the one we have now, that starts the print button

luc-github, 2022-09-18:
this will be available when GCODE HOST is ready for 3D printer - this feature is currently only available for grbl 
the GCODE StreamHost feature is not ready yet https://github.com/luc-github/ESP3D/issues/243
Once supported the Flash will have stream support

luc-github, 2023-10-03:
On ESP3D-TFT used with Marlin streaming is now available, webui have been updated to display Play button for direct SD on TFT and FS on TFT