luc-github/ESP3D-WEBUI #269: [FEATURE REQUEST]grblHAL tool changer support

luc-github, 2022-09-23:
Per suggestion of @terjeio  https://github.com/luc-github/ESP3D-WEBUI/issues/252#issuecomment-1197700038
webui need to support [grblHAL style tool change handling](https://github.com/grblHAL/core/wiki/Manual-tool-change-protocol)

here the operations: https://github.com/grblHAL/core/wiki/Manual,-semi-automatic-and-automatic-tool-change

@terjeio do you have some hardware suggestion for the tool changer ? 
I need to setup one to understand the constraints

Also do you have any expectation of UI or actions you want to have in WebUI ?

As CNC newbie I need some advices 😸

terjeio, 2022-09-23:
> do you have some hardware suggestion for the tool changer ?

No, and if the machine has a tool changer and support code for it then no changes are needed for the WebUI.

The tool change protocol is for manual or semi-automatic tool changes where the tool itself has to be changed by hand, and the touch-off for setting the correct position is either manual or automatic. All modes works by the controller changing the state to `Tool` and wait for the sender sending acknowledge command character `0xA3`. After this the user changes the tool and perform a touch-off to set the correct offset. When done either a cycle start command is sent, or the cycle start input pin is asserted, to restart program execution.

When WebUI detects the 'Tool' state it should change the Pause/Resume button to Pause state so that a cycle command can be sent. This is enough to support the protocol as I belive jogging and zeroing the offsets can be done at all times from the WebUI.

For more a more advanced implementation you may consider adding status markers (similar to the _Input pins_ markers?) to indicate that the [tool length reference has been set](https://github.com/grblHAL/core/wiki/Report-extensions#realtime-report) with the `$TLR` command and one to show that a [tool length offset](https://github.com/grblHAL/core/wiki/Report-extensions#ngc-parameters-report-extensions) is active. I have such markers in ioSender in the _Work Parameters_ grouping. (Macro) buttons for the `$TLR` and `$TPW` commands would also be nice? In ioSender I leave it to the user to define macros for those.

FYI Stefan Bauer, @S2000Stefan, has made some nice [YouTube video tutorials](https://www.youtube.com/playlist?list=PLnSV6o2cRxM5mQQe4ec5cS2J8jBsEciY3) for ioSender, among them one for tool changes.

A small program with tool changes:
[Noname1.zip](https://github.com/luc-github/ESP3D-WEBUI/files/9630719/Noname1.zip)

luc-github, 2022-09-23:
>No, and if the machine has a tool changer and support code for it then no changes are needed for the WebUI.

sorry my question was more what do I need to do to  add tool changer to my cnc ? because today I do not have any 

>The tool change protocol is for manual or semi-automatic tool changes where the tool itself has to be changed by hand

Ho ok I was thinking grblHAL support same as this one https://www.youtube.com/watch?v=3ikLcC5NidU in grbl_ESP32

>For more a more advanced implementation you may consider adding status markers

Thank you I will have a look to these

>FYI Stefan Bauer, @S2000Stefan, has made some nice [YouTube video tutorials](https://www.youtube.com/playlist?list=PLnSV6o2cRxM5mQQe4ec5cS2J8jBsEciY3) for ioSender, among them one for tool changes.

Great I will check these also - thank you

terjeio, 2022-09-23:
> sorry my question was more what do I need to do to add tool changer to my cnc ? because today I do not have any

You have to make or buy one - and then write a plugin to handle the motions involved. I have a Kress spindle in my machine, and the [ATC and tool holders are very expensive](https://shop.stepcraft-systems.com/Tool-Changers) so not something I would consider for my hobby use. But sure I would like to have one...

One day I might write a configurable plugin for ATCs - but I would need a test setup for debugging/testing.

luc-github, 2022-09-23:
Thank you the [ATC and tool holders are very expensive](https://shop.stepcraft-systems.com/Tool-Changers) information what was I needed ^_^
Definitly out of my budget 😓

errolt, 2023-01-24:
Don't know if I should file a separate bug, but with V3, when grblHAL goes into TOOL state, the GUI doesn't notice that is is a form of HOLD mode, and the pause button stays a pause button instead of changing to a play/resume button. So, no way to resume.

luc-github, 2023-01-24:
as you see this is still in feature request, tool change  is not yet implemented

errolt, 2023-01-24:
That is fine. I couldn't figure out if this FR is only for semi auto/auto tool change, or for all tool change.

luc-github, 2023-01-24:
tool is specific feature of grblHAL and I need to spend time on understand it to implement ui properly

terjeio, 2023-01-27:
@luc-github I believe it should not be too difficult to implement as the WebUI runs gcode stored on the SD card - stream switching is taken care of by the SD card implementation.
You may try this:
When `Tool` state is reported send the acknowledge character (`0xA3`) and enable feed hold mode in the UI so that the cycle start character (`~` or `0x81`) can be sent. I believe jogging is possible per default so no need to change that.
If the `Tool` state is exited for some reason (could be the user signalling cycle start via a physical button) change the UI to match whatever the new state is.

luc-github, 2023-01-27:
@terjeio thank you for the explanations, I did not had time to experiment tool state yet.
* if Tool state in report 
1 - send the acknowledge character (0xA3) 
2 - show resume button (~ / 0x81)

* if Tool state disappear
 1 -  follow new state UI

I will check this - it is indeed simple to implement, I just need to track I send only one ack

luc-github, 2023-02-13:
@errolt @terjeio sorry for the delay - I was busy on others topics, here the implementation of the support of Tool state as described above
https://github.com/luc-github/ESP3D-WEBUI/blob/3.0/dist/CNC/GRBLHal/index.html.gz

Please give a try

terjeio, 2023-02-13:
I do not have a machine to test with, however a dry run looks good to me:

[image]

Even the message that can be associated with a tool change is displayed. :+1:

@errolt can test with a machine?

luc-github, 2023-02-14:
@terjeio I have a question - sorry I did not digged in grblHAL to grab the solution...
any impact of sending several 0xA3 ack ?
Like in this user case :
1-  Webui detect the Tool change state => send one 0xA3 => Ok
2 - User close the web UI
3 - User open again the web UI => UI see state is unknown then at first `?` report,see the Tool state => Send again 0xA3

Other user case:
1 - user use Sender that manage Tool change and webUI is closed => Sender send the 0XA3 =>Ok
2 -  User open the web UI => UI see state is unknown then at first `?` report, see the Tool state => Send again 0xA3

Any issue for getting a new 0xA3?

I could ignore the change state if just opening webui but if  webUI is open after any Tool state change ack is sent, webui do not send 0XA3, and so the 0xA3 will be never send  
 
 It can be an edge case but because of Murphy law...

terjeio, 2023-02-14:
@luc-github Sending more than one ack should have no ill effects since it is waited for after the state transitions to Tool. When the ack is received after the transition the wait is terminated and further acs are ignored.

luc-github, 2023-02-14:
Perfect! thank you  😸

errolt, 2023-02-14:
> @errolt can test with a machine?

Will do, but might only happen over the weekend.

errolt, 2023-02-21:
Looks good. 

Can't do a probe, but that is ok. I can use jog to touch off manually.

Thanks!

luc-github, 2023-02-21:
>Can't do a probe, but that is ok. I can use jog to touch off manually.

Do you mean probe panel is not working ? is something wrong ?

errolt, 2023-02-22:
I'm sorry. I should have been clearer. I was testing the jogging during Tool change state. In that state probing gives an Error 40, if I recall correctly. Something like "Only $ commands allowed". Jogging and zeroing works great though.

luc-github, 2023-04-23:
I guess it can be close as original topic is implemented

closed 2023-04-23 by luc-github