luc-github/ESP3D-WEBUI #273: [FEATURE REQUEST] Add support for pen plotters

ithinkido, 2022-10-08:
**Is your feature request related to a problem? Please describe.**
The ESP3D project has amazing flexibility. I would like to add support for hpgl pen plotters. These great machines are just too good to let die out. Adding the possibility of using these machines within the ESP3D framework would open up features such as wifi connectivity, SD plotting and all the other fantastic features that ESP3D has to offer

**Describe the solution you'd like**
The integration of hpgl commands into some of the panels such as JOG and status. The core features like the SD file system and terminal are already 99% usable and only need minor changes to use.

**Describe alternatives you've considered**
I had started a software that has similar features. It was not even nearly as stable as ESP3D . After looking at the ESP3D project I realized I was just trying to re-invent the wheel. The required changes have already been tried and tested so should be good to add to ESP3D. The changes to the UI would be very much within the framework of the existing ESP3D UI and would not require a total re write.

Basic notes here 
https://github.com/ithinkido/ESP3D/tree/3.0_HPGL/HPGL

luc-github, 2022-10-11:
I have added a version of WebUI for HP-GL plotter using ESP3D V3 
the Positions status , the Jog Panel  is now implemented : https://github.com/luc-github/ESP3D-WEBUI/tree/3.0/dist/Plotter/HP-GL
The UI is also a POC for https://github.com/luc-github/ESP3D-WEBUI/issues/262
It add `;` to every not [ESPXXX] that does not have it as discussed on discord
This is first try - as catching and sorting  status reports is currently challenging - the filter part will need some update according your solution in ESP3D command parser

luc-github, 2023-10-03:
@ithinkido any update on this ? I would like to know if I keep this topic open or cancel it 
There is no code yet on esp3d to actually support HP-GL plotters , do you have any time frame ?

ithinkido, 2023-10-10:
Hey Luc, I dont have a timeframe to offer. I will continue to work on this, but i am quite OK if you want to close this till i am able to add more usable code

luc-github, 2024-10-13:
@ithinkido still no visibility about time frame ?

luc-github, 2024-11-20:
Feel free to reopen when ready

closed 2024-11-20 by luc-github